nkolban/esp32-snippets #592: BLEscan error with current version of ESP32 arduino env

1technophile, 2018-07-04:
ESP32 BLE version[ v0.4.16](https://github.com/nkolban/ESP32_BLE_Arduino/tree/7951347ed68313d75c367e1f2cce763cb56d1eb2)
ESP32 arduino [env](https://github.com/espressif/arduino-esp32/tree/a835bb26c4ef80c6cba8ab493830b6c00352cfc4)
Program [BLEScan](https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/examples/BLE_scan/BLE_scan.ino)
Board: DOIT ESP32 DEVKIT V1

1. Open BLEScan.ino with Arduino IDE 1.8.5
2. Compile OK
3. Upload to the board

The serial monitor returns:
```
Scanning...
E (17) BLEDevice: esp_bt_controller_init: rc=259 Invalid state
E (18) BLEScan: esp_ble_gap_set_scan_params: err: 259, text: Invalid state
Devices found: 0
Scan done!
```

**If I do the same thing with:**
ESP32 BLE version[ v0.4.9](https://github.com/nkolban/ESP32_BLE_Arduino/tree/6bad7b42a96f0aa493323ef4821a8efb0e8815f2)
ESP32 arduino [env](https://github.com/espressif/arduino-esp32/tree/a153f6ad280e5932be9c80341556348ff505d586)
Program [BLEScan](https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/examples/BLE_scan/BLE_scan.ino)
Board: DOIT ESP32 DEVKIT V1

1. Open BLEScan.ino with Arduino IDE 1.8.5
2. Compile OK
3. Upload to the board

The serial monitor returns:
```
Scanning...
Advertised Device: Name: , Address: d4:1XXXXXXXXX:40, manufacturer data: df004c305331303330314a41, serviceUUID: 3dda0001-957f-7d4a-34a6-74696673696d 
Advertised Device: Name: MJ_HT_V1, Address: 4c:XXXXXXXXX:79 
E (42992) BT: btc_search_callback  BLE observe complete. Num Resp 2

Devices found: 2
Scan done!

```

It seems that the last version of BLE library combined with the last revision of ESP32 env doesn't enable to do scan properly.

paweb88, 2018-07-05:
I've got the same error when i tried any of examples :(

chegewara, 2018-07-06:
@1technophile @paweb88 
Sorry guys but i cant reproduce this issue. I have fresh updated arduino-esp32 and BLE 0.4.16:
`HEAD detached at 7951347`
Here is scan result from ble_scan example:
```
[D][BLEUtils.cpp:1270] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_CMPL_EVT
Devices found: 1
Scan done!
```

Nakul93, 2018-07-14:
Hi,
I am also facing this same issue

paweb88, 2018-07-15:
It works fine on: 0.4.16 :)

ps. merge correctly or be sure that you have exact version

mout551, 2018-07-18:
Hi guys,
i have the same problem with my LoPy from Pycom even if I have fresh updated arduino-esp32 and BLE 0.4.16 : 

`Scanning...
E (17) BLEDevice: esp_bt_controller_init: rc=259 Invalid state
E (18) BLEScan: esp_ble_gap_set_scan_params: err: 259, text: Invalid state
Devices found: 0
Scan done!`

Jeard, 2018-07-24:
Hi,
I've got the same error with BLE uart example. 
`E (18) BLEDevice: esp_bt_controller_init: rc=259 Invalid state
`

chegewara, 2018-07-24:
@mout551 @Jeard Could you try with this arduino-esp32 commit:
https://github.com/espressif/arduino-esp32/commit/89859f7f4c07ce3e5d487a4b689dbda78f5d5c3a

chegewara, 2018-07-24:
@Jeard  What i mean, you have to hard reset to this commit, because there is some issue with never commits (at least i had issue)

Jeard, 2018-07-25:
@mout551 : Thanks for the suggestion. It worked with `89859f7f4c07ce3e5d487a4b689dbda78f5d5c3a`

1technophile, 2018-09-02:
@chegewara thanks for this info, I confirm it works with this commit but not with the commits after.

chegewara, 2018-09-03:
I found what caused issues for me with arduino-ide latest commit. It was releasing memory from bluetooth classic.