nkolban/esp32-snippets #593: BLE scanning and  send via wifi as soon as a result is available

MoustaphaDIENG, 2018-07-04:
Hello,
I'm currently working on a BLE-Wifi gateway and I'm looking for maximum responsiveness. I would scan all the time and send by wifi as soon as I find a tag. 
Is it possible to scan and send via wifi at the same time? 
I am trying to put the scan function in the core 0 and the wifi function in the core 1. Is it possible for the antenna to listen for tags and sending at the same time data via wifi?

chegewara, 2018-07-04:
Few weeks ago i would say yes you can, its easy. That was when i just did very simple test example and all worked fine. Now i am wiser, because i had to do my own project wifi IoT + ble. 

The answer still is yes, its possible, but you have to solve many issues. First will be heap shortage. Next you have to remember that esp32 is sharing radio between wifi and ble, because of that you have to manage in your app time that is given to wifi and ble. At beginning dont try to scan before you connect wifi (or connecting wifi when you scan). Then, when you are connected you can start scanning and now is another issue. When you got results that you would like to send over wifi you HAVE to stop scan. 

There is one theoretical option you could try (i never tested it). BLE scan has 2 parameters, window and interval. You could try to play with it and see how it works.

KristianDukov, 2018-07-24:
So using an ESP to  gathered HR info by BLE and sent over Wifi won't work?

chegewara, 2018-07-24:
@KristianDukov I cant answer this question yes or no. In theory it should work, but for sure its not easy to write app ble + wifi coexistence.

MoustaphaDIENG, 2018-08-06:
Hello @chegewara,
I was the one who created this issue, but I had an unexpected event during the last month, so I could not give a return. I'm sorry about that.
So I launched the application I was talking about but with some modifications. Now I scan during 5s, then I send a list of devices that I scanned via mqtt (with PubSubClient library). There I meet another problem. The application works fine for about 10min then I see the error message: 
"Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Wifi
CPU 1: IDLE
And the error always happens during the scan period. 
Do you have an explanation for this? Thanks

chegewara, 2018-08-07:
If you are working with esp-idf then you can try to increase watchdog time or turn it off.

MoustaphaDIENG, 2018-08-07:
No i am working with the Arduino IDE. 
I already tried to modify the file sdkconfig.h but i saw on the top that it is an automatically generated file and that i should not modify it

chegewara, 2018-08-11:
I cant help you more with this, maybe @me-no-dev can help with watchdogs.

closed 2023-01-01 by MoustaphaDIENG