nkolban/esp32-snippets #596: Print Characteristic Value to Serial Monitor

kirawa6, 2018-07-06:
I'm using the BLEClient example sketch with my ESP32 in the arduino IDE, and I'm trying to read the value of my characteristic and print to the serial monitor, but I'm getting this:
[image]
this is the code I'm using:
`std::string value = pRemoteChar2->readValue();
    const char *val = value.c_str();
    Serial.print("The myo info characteristic's value is: ");
    Serial.println(val);`

any ideas as to what I am doing wrong? Thanks!

chegewara, 2018-07-07:
Your code looks good, but here is the question: what have you read from the other device? Is it string or is it binary value? With your code you can only print string messages and with very little change integer values. Im guessing its the case, peripheral you are connecting to and read characteristic value is not sending string but other data (but like always, i may be wrong).

kirawa6, 2018-07-07:
The characteristic is meant to print the firmware version of my server (the myo armband), so it's definitely possible that the device is sending that in binary/hexadecimal. I'm having a lot of trouble reading any of the read only characteristics from this device, but I appreciate your help a lot

hansmbakker, 2018-12-04:
@kirawadden this issue seems solved, could you please close it to keep the issue list clean?

cc @chegewara

closed 2018-12-09 by chegewara