nkolban/esp32-snippets #598: Error on startup with SampleNotify

markingle, 2018-07-08:
I am trying to get SampleClient_Notify and SampleNotify communicating.  SampleNotify is complaining about the adv params with error code 0x30.  The client side seems to be fine and is showing "MyDevice" from the scan but is not connecting.

Things I have tried:

1. make flash_erase
2. confirm local repo matches remote repo

I am using two SparkFun ESP32 Thing.

Thanks in advance for any suggestions on how to fix this....

**Serial output from SampleClient_Notify**

I (230) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (550) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 1, 0
I (880) SampleClient_Notify: Advertised Device: Name: , Address: 61:1f:8f:eb:da:24, manufacturer data: 4c000d04de45626210050b1052618e
I (900) SampleClient_Notify: Advertised Device: Name: , Address: 56:b3:91:5c:b1:16, manufacturer data: 4c0010020b00
**I (1200) SampleClient_Notify: Advertised Device: Name: MYDEVICE, Address: 24:0a:c4:05:70:c6, txPower: -21**

**Serial output from SampleNotify**
[0;32mI (357) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
0;32mI (677) phy: phy_version: 3910, c0c45a3, May 21 2018, 18:07:06, 1, 00m
0;32mI (927) SampleNotify: Created BLE Server 0m
[0;32mI (927) SampleNotify: Created BLE Service[0m
[0;32mI (927) SampleNotify: Created BLE Characteristic[0m
[0;32mI (927) SampleNotify: Created BLE Descriptor[0m
[0;32mI (937) SampleNotify: Started Service[0m
**[0;31mE (947) BT_HCI: hci write adv params error 0x30[0m**
[0;32mI (947) SampleNotify: Started advertising[0m

cdyrssen, 2018-07-16:
I am getting a a similar error:

Starting services...
BLE advertising UUID...
E (3518) BT_HCI: hci write adv params error 0x30

This is the code I have used for BLE server configuration:
static void run() {
	// Create the BLE Device
	BLEDevice::init("ESP32");

	printf("Initializing BLE server, services, and characteristics...\n");
	// Create the BLE Server
	BLEServer *pServer = BLEDevice::createServer();
	pServer->setCallbacks(new MyServerCallbacks());

	// Set up battery service and characteristics
	BLEService *pBatterySvc = pServer->createService(BLEUUID(BATTERY_SVC_UUID));
	pBatteryLevelChr = pBatterySvc->createCharacteristic(
		BLEUUID(BATTERY_LEVEL_UUID),
		BLECharacteristic::PROPERTY_READ
	);
	// Set up WiFi scan service and characteristics
	BLEService *pWifiSvc = pServer->createService(BLEUUID(WIFI_SVC_UUID));
	pWifiSSIDChr = pWifiSvc->createCharacteristic(
			BLEUUID(SSID_BROADCAST_UUID),
			BLECharacteristic::PROPERTY_READ
	);
	pAPSigStrChr = pWifiSvc->createCharacteristic(
			BLEUUID(AP_SIGNAL_STR_UUID),
			BLECharacteristic::PROPERTY_READ
	);
	// WILL NEED TO BE ENCRYPTED SOMEHOW!!!
	pHomePassChr = pWifiSvc->createCharacteristic(
			BLEUUID(HOME_PASS_UUID),
			//BLECharacteristic::PROPERTY_READ |
			BLECharacteristic::PROPERTY_WRITE
	);
	pClientCmdChr = pWifiSvc->createCharacteristic(
			BLEUUID(CLIENT_COMMAND_UUID),
			BLECharacteristic::PROPERTY_READ |
			BLECharacteristic::PROPERTY_WRITE
	);

	//pBatteryLevelNotifyTask = new BatteryLevelNotifyTask();
	//pBatteryLevelNotifyTask->setStackSize(8000);
	pBroadcastWifiTask = new BroadcastWifiTask();
	pBroadcastWifiTask->setStackSize(8000);

	printf("Starting services...\n");
	// Start the service
	pBatterySvc->start();
	pWifiSvc->start();

	printf("BLE advertising UUID...\n");
	// Start advertising
	pServer->getAdvertising()->start();
}

I am able to see my ESP32 in the nRF connect application, but it does not show up on any of my devices when I searching for it via native bluetooth connection means (i.e. switching bluetooth on in iOS settings and searching for ESP32). ESP32 just never shows up as an available device to connect to.

markingle, 2018-07-16:
@cdyrssen My app is running fine with the error "E (3518) BT_HCI: hci write adv params error 0x30" I think the error message may not be an issue with affecting your program.  If you are using iOS only devices there are certain timing settings and parameters in the GAP layer that iOS is expecting.  I dont have a link handy but they are referenced in the iOS developer manual.

cdyrssen, 2018-07-16:
@markingle is there something similar for Android and Windows? As I am having the same issue where the ESP32 doesn't show up on the list.

cdyrssen, 2018-07-16:
@markingle issue resolved for Android and Windows. See comment edit

markingle, 2018-07-16:
Glad you found the problem.  I get in the way of myself too!  More often than I care to admit.  Good luck with your project!

cdyrssen, 2018-07-16:
So after some testing I can see my ESP32 on Android and Windows and only on iOS through the nRF Connect. Android connects to it with now issues, but Windows tells me it didn't get a response and just sitts there. Nothing shows up on the ESP32's debug log during this whole process. I am still looking into the iOS GAP layer issue.

cdyrssen, 2018-07-16:
For anyone interested in Apple's requirements for a BLE device to be a supported accessory here is the design guidelines. BLE specifications begin on page 59.

https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf

vuanh-minh, 2018-07-19:
Hello @cdyrssen, i am currently working on a BLE project, and i have some issues with the BT_HCI: hci write adv params error 0x30 too. This appears randomly, and the advertising can't works properly. Do you know where this problem comes, and how to resolve it ?

cdyrssen, 2018-07-19:
@vuanh-minh I have not been able to completely remove the error from showing up as well. It should be noted it hasn't affected the execution of my program and everything appears to be functioning properly. 

Are you using an application such as nRF Connect or LightBlue to debug your BLE connection issues? If not, I suggest you look into those apps, as they have become invaluable in my development process (my preference is nRF Connect). It took me a bit of digging, but as referred to in this [StackOverflow question:](https://stackoverflow.com/questions/27211573/bluetooth-low-energy-advertising-to-be-discoverable-in-ios-settings), BLE devices will not show up in the iOS native bluetooth settings. Instead, the device's services UUIDs must be specifically searched for in your iOS application. 

If this is not your issue, I would refer to [this](https://github.com/nkolban/esp32-snippets/issues/122) issue thread. It has links to the above StackOverflow post as well as another github issue thread which may be your potential culprit.