nkolban/esp32-snippets #604: BLE Client Detect Device With Multiple Services

kirawa6, 2018-07-11:
I'm trying to connect my esp32 to a device (ble server) that offers multiple ble services, but I can only get it to pick up one of them. I tried using the nRF connect app to create a server with multiple services, and my esp32 still only detected one of them. On the nRF scanner I can pick up the multiple services that the device I would like to connect to has to offer, but I can only get the esp to detect one of them. Is there a way to detect multiple services with the esp32 arduino ble client sketch?

chegewara, 2018-07-11:
You can test if service UUID is advertised with:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L139-L145

But you can only getServiceUUID() for the first service UUID. This is outdated function and its left only for backward compatibility, when there was assumption that only one service UUID is advertised.

If you want to test against few services then you can do something like that:
`if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID) && advertisedDevice.isAdvertisingService(serviceUUID2) )`

kirawa6, 2018-07-11:
I'm a bit new to coding this type of program, how would I implement this test in, for example, the BLE_scan sketch?

chegewara, 2018-07-11:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L78

If you want only to scan for device then you can delete all code that is performing connection (compare with BLE_scan). But you need `pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());`

kirawa6, 2018-07-11:
I tried doing 
`if (advertisedDevice.isAdvertisingService(serviceUUID2)) Serial.println("Service UUID 2 found");` but nothing printed to my serial monitor. Does this mean that my server isn't advertising the service, or is the esp just unable to pick it up?

chegewara, 2018-07-11:
Its all depends: what is your serviceUUID2 value, what is your server advertising etc.
You can perform few intermediate steps before you end up with final code:
- `if(1)`,
- `if (advertisedDevice.haveServiceUUID())`,
- `if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID2))`

You can also do at the end something like that:
```
if (advertisedDevice.haveServiceUUID()){
   Serial.println("Service found");
   if (advertisedDevice.isAdvertisingService(serviceUUID2)){
      Serial.println("Service UUID 2 found");
   }
}
```

AlbanVl, 2018-08-20:
Hello,

I have the same problem and I tried to solve this by using the solution proposed by chegewara, but it doesn't work...
I'm sure that the server works well because I can see the 2 services and their charcteristics on my smartphone with the app "nRF Connect".
I can well see the notification "Service found" but not for the second notification... It seems that the function "advertisedDevice.isAdvertisingService(serviceUUID2)" isn't able to find more than the first service...
Is there someone who could help me?

chegewara, 2018-08-27:
@AlbanVl could you post screenshot from nRF connect with peer device advertising info?

EBZ-Krisemendt, 2018-08-27:
i just "tweaked" BLEAdvertisedDevice by adding/changing https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L131 to
```
BLEUUID BLEAdvertisedDevice::getServiceUUID(int whichOne) {
	if (m_serviceUUIDs.size() < whichOne) whichOne=0;
	return m_serviceUUIDs[whichOne];
}
```
and looped inside the advertised-callback to validate my point. tested with a device that provides 11 different services but apparently advertises only a single one... connecting each advertised device and using `BLEClient::getService(BLEUUID uuid)` seems to be the way to go. (you'll get a reference to the remote-service or a nullpointer as a return-value)

chegewara, 2018-08-27:
@EBZ-Krisemendt you are new to ble, please dont mix 2 different concepts. i just answered to you, advertised service uuid is not the same as services served after connecting to device. its open source code and you can do whatever you want with it, but dont confuse people.

AlbanVl, 2018-08-29:
Ok, I think I've resolved my problem!
First, I've to get the first service (= advertised service uuid) and when I'm connected to it, I can get the other services ( = services served).
It seems that nRF connect do directly the job and list all the services served.
Thank you for your help!

chegewara, 2018-09-25:
Hi all, sorry for this bug, i think its my fault. Most likely all issues are because of this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L140

closed 2018-12-11 by kirawadden

Sule-2002, 2020-05-30:
hello I am sorry about language, I am new on ESP32 coding,
I have same problem , I can see multiple services on nrf connect app but  when I try to find and connect ,I can connect only 1 service .
i didn't understand how you solved it. Can you share with me an example Arduino code?

kirawadden, 2020-06-01:
@Sule-2002 here is my code: https://github.com/kirawadden/ble_or_bust. It turned out that to access the other services they needed to be turned on through a service that was exposed. I had to write to a characteristic in the exposed service in order to expose the others. To find the specific packet to write, you will need to look through documentation/BLE specification of the specific sensor to use.

Sule-2002, 2020-06-01:
hello @kirawadden I really appreciate you. I realy thank you. 
I used the example that you advised me and it worked successfully and connected the other services.