nkolban/esp32-snippets #606: ibeacon repeater

KZ-IDEA, 2018-07-13:
I am trying to make an ibeacon repeater.

Basically I want to read a sensor and then send the exact same reading.

I can use the sample scan to read the sensor. I also was able to use a server to send data and see it as an ibeacon on my phone.

I do not understand objects and c++ code very well so am looking at a high level of description of how to go about doing this task.

chegewara, 2018-07-14:
To do that you have to understand beacons, how its constructed from advertising point of view and experiment with BLEBeacon class. 
Start with nRF connect to see how beacon advertising looks like, then try to copy it with esp32 code:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_iBeacon/BLE_iBeacon.ino

hansmbakker, 2018-12-15:
@KZ-IDEA there has been no activity in this thread for several months. If your issue is solved, could you please close it to keep the issue list clean?
cc @chegewara

closed 2018-12-15 by KZ-IDEA