nkolban/esp32-snippets #610: Multi characteristic in a single service.

Ananthesh015, 2018-07-19:
**Hello,
As i had used 3 characteristic in a single service and i am getting error like**

Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default, QIO, 40MHz, 4MB (32Mb), 115200, None"

E32client:107: error: 'pRemoteCharacteristicTmp' does not name a type

   pRemoteCharacteristicTmp = pRemoteService->getCharacteristic(charTmpUUID);

   ^

E32client:108: error: expected unqualified-id before 'if'

   if (pRemoteCharacteristicTmp == nullptr) {

   ^

E32client:113: error: 'Serial' does not name a type

   Serial.println(" - Found our characteristic Tmp");

   ^

E32client:116: error: 'value' does not name a type

   value = pRemoteCharacteristicTmp->readValue();

   ^

E32client:117: error: 'Serial' does not name a type

   Serial.print("The characteristic value was: ");

   ^

E32client:118: error: 'Serial' does not name a type

   Serial.println(value.c_str());

   ^

E32client:120: error: 'pRemoteCharacteristicTmp' does not name a type

   pRemoteCharacteristicTmp->registerForNotify(notifyCallbackTmp);

   ^

E32client:121: error: expected declaration before '}' token

 }

 ^

exit status 1
'pRemoteCharacteristicTmp' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

**_as i had searched that it is  need to reserve some space in RAM.
sir, plz help in resolve the problem._**

chegewara, 2018-07-19:
`BLERemoteCharacteristic *pRemoteCharacteristicTmp = pRemoteService->getCharacteristic(charTmpUUID);`

Ananthesh015, 2018-07-19:
[code]
static BLERemoteCharacteristic* pRemoteCharacteristicTmp;
static BLEUUID    charTmpUUID(BLEUUID((uint16_t)0x2A6E));
[/code]

chegewara, 2018-07-19:
`static BLEUUID charTmpUUID = new BLEUUID((uint16_t)0x2A6E);`

Ananthesh015, 2018-07-19:
Error
conversion from 'BLEUUID*' to 'BLEUUID' is ambiguous

chegewara, 2018-07-19:
Sorry,
`static BLEUUID *charTmpUUID = new BLEUUID((uint16_t)0x2A6E);`

Ananthesh015, 2018-07-19:
```
static BLEUUID    charTmpUUID(BLEUUID((uint16_t)0x2A6E));
static BLERemoteCharacteristic* pRemoteCharacteristicTmp;
```

chegewara, 2018-07-19:
maybe you are missing include?

Ananthesh015, 2018-07-19:
I had used #include "BLEDevice.h" is any another include is required......?

chegewara, 2018-07-19:
I dont know, maybe BLERemoteCharacteristic?

Ananthesh015, 2018-07-19:
I did't get you sir.

chegewara, 2018-07-19:
`#include "BLERemoteCharacteristic.h`

Ananthesh015, 2018-07-20:
`'pRemoteCharacteristicTmp' does not name a type`