nkolban/esp32-snippets #611: ble_server txPower allways -21

eltsov-da, 2018-07-25:
Hello,
I am new in BLE world so may be I ask something strange but:
I try BLE_server and BLE_scan example
and got: Advertised Device: Name: MyESP32, Address: 30:ae:a4:59:21:4e, txPower: -21 
and about 1m distance of successful scanning.
I try:

```
 BLEDevice::init("MyESP32");
BLEDevice::setPower(ESP_PWR_LVL_P7);

```but still get the same result.

Also I try 

```
BLEDevice::init("MyESP32");
esp_err_t errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P7);
errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P7);
errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN, ESP_PWR_LVL_P7); 

```
and 

Serial.println("Characteristic defined! Now you can read it in your phone!");

  Serial.printf("ESP_BLE_PWR_TYPE_DEFAULT=%d \n", ::esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_DEFAULT)); 
  Serial.printf("ESP_BLE_PWR_TYPE_ADV =%d \n", ::esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV));
  Serial.printf("ESP_BLE_PWR_TYPE_SCAN  =%d \n", ::esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_SCAN));

on server side I'v got:

Starting BLE work!
Characteristic defined! Now you can read it in your phone!
ESP_BLE_PWR_TYPE_DEFAULT=7 
ESP_BLE_PWR_TYPE_ADV =7 
ESP_BLE_PWR_TYPE_SCAN  =7 

but on scan side 
Scanning...
Advertised Device: Name: MyESP32, Address: 30:ae:a4:59:21:4e, txPower: -21 
Devices found: 1
Scan done!

and only 1m distance.

I use master espressif/arduino-esp32  master branch (update today) 
ESP32 BLE Arduino 0.4.16

chegewara, 2018-07-31:
Hi, sorry for late reply. 
Could you try with some other values from this range:
https://github.com/espressif/esp-idf/blob/master/components/bt/include/esp_bt.h#L157-L164

eltsov-da, 2018-07-31:
So I made some more investigation.
1. The fact is that using any value and both BLEDevice::setPower(ххх)
 or   
```
esp_err_t errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT,xxx);
errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, xxx);
errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN ,xxx); 

```In a BLE_scan.ino example I'v got:
Advertised Device: Name: MyESP32, Address: 30:ae:a4:59:21:4e, txPower: -21 

2. Also fact that maximum scan distance is less then 1m
3. It's seems to me that if I use ::esp_ble_tx_power_set I can change tx power
    3.1 If I try
     esp_err_t errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT,ESP_PWR_LVL_P9);
     errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
     errRc=::esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_SCAN ,ESP_PWR_LVL_P9); 
    I can got scan in about 1m range.
   3.2 if is try ESP_PWR_LVL_N12 I still can got scan but in a range !!! about 2сm 
4. it's also seems to me that if I use BLEDevice::setPower(ххх) with any values I'v got some middle range the same to esp_ble_tx_power_set and ESP_PWR_LVL_P3
5. I add 
 pAdvertising->start();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
  Serial.printf("ESP_BLE_PWR_TYPE_DEFAULT=%d \n", ::esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_DEFAULT)); 
  Serial.printf("ESP_BLE_PWR_TYPE_ADV =%d \n", ::esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_ADV));
  Serial.printf("ESP_BLE_PWR_TYPE_SCAN  =%d \n", ::esp_ble_tx_power_get(ESP_BLE_PWR_TYPE_SCAN));

and if I use BLEDevice::setPower in BLE_server.ino Serial monitor  I'v got some thing like this:
Starting BLE work!
Characteristic defined! Now you can read it in your phone!
ESP_BLE_PWR_TYPE_DEFAULT=7 
ESP_BLE_PWR_TYPE_ADV =5 
ESP_BLE_PWR_TYPE_SCAN  =5 

And I include file with some debug info (debug->verbose) etc may be it help. The main problem for me is that the longest range is 1m only
[debug_esp32_ble.txt](https://github.com/nkolban/esp32-snippets/files/2246218/debug_esp32_ble.txt)

chegewara, 2018-07-31:
Tomorrow i will try to perform the same tests to confirm it.

eltsov-da, 2018-08-06:
tomorrow never comes :)

chegewara, 2018-08-06:
Yes, sorry for that. Ive been over 10 month at work away home. Now im back and for few days i dont touch esp32. I will do this test just i need to rest.

eltsov-da, 2018-08-06:
Never mind. It's just a pun comes to my head.

eltsov-da, 2018-08-11:
And what about distance? I can got any signal only in about 1m distance

chegewara, 2018-08-11:
Its possible you have issue with esp32 board. I can scan for esp32 from about 15 m with no issue.

eltsov-da, 2018-08-13:
Yes, it's possible, but I have no problem with WiFi. And as far as I know WiFI and BLE share the same radio module.
I have 4 TTGO LoRa OLED v1 boards and all of them work the same way. 
May be someone have experience with that boards? Please tell me if BLE work’s fine or not.

mws-rmain, 2018-08-16:
Just found this thread. As @chegewara stated, the TxPower reported in the advertising data is not updated (appears it's stays at -21dBm, regardless of what is set - see  [here](https://esp32.com/viewtopic.php?f=13&t=6553&p=29201#p29201)).

Of course that still doesn't explain the poor range you're experiencing. WiFi & BLE share the same radio on the ESP32, but the WiFi stack & Bluetooth stack can have completely different settings, so operation in WiFi doesn't really affect bluetooth operation (apart from proving the antenna interface is OK).

I'm not clear on the 'LoRa' part of these boards. Do they have another radio for operation at 868/915MHz or 433/470MHz?

eltsov-da, 2018-08-17:
About LoRa radio. It's completely separate module. sx1278 or sx1276, depending of frequency  (868/915MHz or 433/470MHz). It's have it's one antenna and communicate with main chip throw SPI.
About WiFi - BLE: I of course understand that stacks are completely different. I mention it only to avoid questions about antenna or some other hardware problems.

mws-rmain, 2018-08-20:
I came across [this video (Andreas Spiess)](https://www.youtube.com/watch?v=CJNq2I_PDHQ) that does some RF Testing of some ESP32/LoRa boards (the TTGO is covered). He criticizes the on-board ESP32 antennas, and points out they are easily shorted when soldering (if you aren't very careful). Possibly something to look at?

eltsov-da, 2018-08-20:
:) once more. WiFi work perfect. And I don't do any soldering with that modules.

FedericoBusero, 2018-09-10:
In BLEAdvertising.cpp is the default value
`m_advData.include_txpower = true;`

So it looks like an issue in the the implementation of
`esp_ble_gap_config_adv_data`

chegewara, 2018-09-10:
https://github.com/espressif/esp-idf/issues/2284

closed 2018-11-13 by chegewara