nkolban/esp32-snippets #612: Service UUID Not Found

tjerge, 2018-07-27:
Hi there,

I'm using a Xiaomi Mi Band 2 as a heart rate monitor. When I scan for devices on the esp32 I can see the Mi Band but there is no advertised service SSID. Do some devices not advertise their service uuid correctly or is something else going on here?

Many thanks.

Serial output here
`load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310

ESP32 HRM BLE Client | Compiled: Jul 27 2018/08:17:03
Connecting to sensor
....
WiFi connected
IP address: 192.168.1.9
BLE Advertised Device found: Name: , Address: c3:c5:9f:98:65:f4, manufacturer data: 570100ee032524cdc732cc64f34bb4dcfe3cbf02c3c59f9865f4
`

Code here:
`/************************************************************************************************************
 *
 *  Copyright(C) 2017 Timo Sariwating
 *
 *  This program is free software: you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation, either version 3 of the License, or
 *  (at your option) any later version.
 *
 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 *  Lesser General Public License for more details.
 *
 *  You should have received a copy of the GNU Lesser General Public License
 *  along with this program; if not, If not, see <http://www.gnu.org/licenses/>.
 *
 ************************************************************************************************************
DESCRIPTION:
This sketch gets Heart Rate from an BLE HRM monitor and display it on an OLED.
- It uses an ESP32 with i2c OLED Display
/************************************************************************************************************/
#include <WiFi.h>
#include <PubSubClient.h>
#include "BLEDevice.h"
#include <ArduinoJson.h>

const String sketchName = "ESP32 HRM BLE Client";

// BLE
// The remote HRM service we wish to connect to.
static  BLEUUID serviceUUID(BLEUUID((uint16_t)0x180d));
// The HRM characteristic of the remote service we are interested in.
static  BLEUUID    charUUID(BLEUUID((uint16_t)0x2A37));

static BLEAddress *pServerAddress;
static boolean doConnect = false;
static boolean connected = false;
static boolean notification = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;

// Wifi
WiFiClient wifi_client;
const char* ssid     = "sensor";
const char* password = "z5zWw0APtI3wJm4WV4na";

// MQTT
PubSubClient mqtt(wifi_client);
[... 241 lines cut ...]
      pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)onPacket, 2, true);
      notification = true;
    }
    
  }

  if (!mqtt.connected()) {
    long now = millis();
    if (now - lastReconnectAttempt > 5000) {
      lastReconnectAttempt = now;
      // Attempt to reconnect
      if (reconnect()) {
        lastReconnectAttempt = 0;
      }
    }
  } else {
    // Client connected
    mqtt.loop();
  }

  //-------------------------------------------------------------------------------------------
  // Timed Code
  //-------------------------------------------------------------------------------------------

} // End of loop`

chegewara, 2018-07-27:
Advertising service UUID is optional, im guessing your peripheral device is advertising manufacturer data instead (of course only if this is your device, because usually windows laptop is advertising this way):
```
BLE Advertised Device found: Name: , Address: c3:c5:9f:98:65:f4, manufacturer data: 570100ee032524cdc732cc64f34bb4dcfe3cbf02c3c59f9865f4```

1807374010, 2018-08-05:
mi band 2 doesn't advertise it services'uuid in advertised data, but in scan response data. While scaning BLE devices esp32 only analyse advertised data. So it didn't find the uuid…I had tried changing the library code to let it analyse other data. And it works! But i don't know will it have side effect.

chegewara, 2018-08-07:
You can prepare PR and we will see if its good to merge or needs some work.

imranmumtaz92, 2023-02-10:
> 

@0x504B0304 , We are working on integrating WT901BLE device with ESP32 on bluetooth and facing the same issue that ESP32 does not see the service UUID of the sensor. Can you please guide which library code you updated so that it worked so that we can also your technique to make it work?