nkolban/esp32-snippets #615: How to initialise BLEAddress(advertisedDevice.getAddress())

skyng22003, 2018-08-02:
I have read your previous post about removing "new" from "MyAdvertisedDeviceCallbacks" in the BLE Device example.

I was wondering how you would replace the "new" in BLEAddress(advertisedDevice.getAddress());
in the following code:

void MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice advertisedDevice) {

		/**
		* Called for each advertising BLE server.
		*/
		Serial.print("BLE Advertised Device found: ");
		Serial.println(advertisedDevice.toString().c_str());
		// We have found a device, let us now see if it contains the service we are looking for.
		if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID)) {

			// 

			Serial.print("Found our device!  address: ");
				advertisedDevice.getScan()->stop();
				pServerAddress = new BLEAddress(advertisedDevice.getAddress());
		
			BT_conStatus = doConnect;
		} // Found our server

Thanks again for your help!

Tom5051, 2020-01-15:
Hi,
I am also hoping there is another way to compare MAC addresses from an iBeacon

chegewara, 2020-01-16:
When you create class object with new, you have to delete it when no longer required.

Tom5051, 2020-01-16:
> 
> 
> When you create class object with new, you have to delete it when no longer required.

Sorry I'm using Arduino IDE and I'm not very good. Are you able to let me know how to delete the new objects once the scan has completed? I'm not worried about duplicates in my situation.

Tom5051, 2020-01-16:
> 
> 
> When you create class object with new, you have to delete it when no longer required.

Thank you! That worked :)

JKSAR, 2021-02-06:
> > When you create class object with new, you have to delete it when no longer required.
> 
> Thank you! This worked :)
> 
> delete pServerAddress;
> pServerAddress = NULL;

Hi,

Could you share your code for us? 

Best Regards,