nkolban/esp32-snippets #616: Crash when connecting to BLE_Server with BLE_client_passkey.ino if security is enabled

skyng22003, 2018-08-03:
BLE_client_passkey.ino and BLE_client_numeric_confirmation.ino does not compile with esp_log_buffer_hex(LOG_TAG, auth_cmpl.bd_addr, sizeof(auth_cmpl.bd_addr)); uncommented

Both sketches compile once the line above is commented out however when it tries to connect to the static passkey enabled server it crashes if the security is disabled on the server end the sketches function normally and the connection is successful.

Below are the backtrace results:
0x40090c38: invoke_abort at panic.c line 649
0x40090e3b: abort at panic.c line 649
0x401811f3: __cxxabiv1::__terminate(void (*)()) at eh_terminate.cc line 112
0x4018123a: std::terminate() at eh_terminate.cc line 112
0x4016ecd3: __cxa_throw at eh_throw.cc line 87
0x400d3091: BLEClient::getService(BLEUUID) at BLEClient.cpp line 142
0x400d16e7: connectToServer(BLEAddress) at BLE_sec_test.ino line 75
0x400d19c6: loop() at BLE_sec_test.ino line 148
0x40188db0: loopTask(void*) at main.cpp line 17 (discriminator 1)

I have a BLE 4.2 server with the following security options:
1. IO CAP is set to DISPLAY ONLY
2. Authentication is set to on connect
3. The passkey has been set to a static one and pairs normally on an android phone

I think it crashes when the authentication process is initiated?

gourryinverse, 2018-08-06:
The issue is you are using an outdated library.  Do a bit pull, there is a one line fix in clearServices that will fix this.  They forgot to reset a boolean value and end up never retrieving the services

skyng22003, 2018-08-07:
Thanks for the suggestion, I updated the esp32 Arduino core and esp32_BLE and it still crashes, I think I am having the same issue as #620

chegewara, 2018-08-11:
I wish i could help, but i have no code that can reproduce this issue. All my apps work fine.

skyng22003, 2018-08-12:
@chegewara Thanks again, seems to be stupidity on my part.

The error was in this line:
pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND); 

I just had to match the security level between the host and client and everything worked.

closed 2018-08-12 by skyng22003

chegewara, 2018-08-12:
@skyng22003 I dont think its stupidity, more like its issue with esp-idf ble stack.
https://github.com/espressif/esp-idf/issues/2036