nkolban/esp32-snippets #617: ESP32 as BLE Client and Server simultaneously

Galzai, 2018-08-04:
I'm attempting to use the ESP32 as a BLE Server (HID Keyboard) connected to a device, while simultaneously running it as a BLE client  to receive notifications from another ESP32 (Basically trying to make a wireless split keyboard) .
I can get each part to work individually, but I don't really know how to approach making them work together.
I apologize if this is a simple issue, I don't have much experience with BLE.

chegewara, 2018-08-11:
It can be easy or not. You can start with merging client and server examples in one code.

BiggyBaron, 2018-08-22:
@nkolban Hello, I am trying to merge client+server code, is there way to stop and delete BLEdevice to create a new one?

Pseudocode:
`BLEDevice::init("Client");
DoSomeRandomStuff();
BLEDevice::kill("Client");
BLEDevice::init("Server");`