nkolban/esp32-snippets #618: registerForNotify BLEUUID("0x2902") ERROR: UUID value not 2, 4, 16 or 36 bytes

yuyi1005, 2018-08-04:
I am using the example of the BLE_client collecting data from a BLE server, which supports only "notify" characteristic, but it crashes in registerForNotify().

I am using the latest code in esp32-snippets, and I have found the wrong point as follow.

BLERemoteCharacteristic.cpp
Line 504: BLERemoteDescriptor *desc = getDescriptor(BLEUUID("0x2902"));

length("0x2902") is always 6, which leads to the error "UUID value not 2, 4, 16 or 36 bytes".

Here is the log:

 - Found our characteristic
[D][BLERemoteCharacteristic.cpp:484] registerForNotify(): >> registerForNotify(): Characteristic: uuid: 49535343-1e4d-4bd9-ba61-23c647249616, handle: 42 0x2a, props: broadcast: 0, read: 0, write_nr: 0, write: 0, notify: 1, indicate: 0, auth: 0
[E][BLEUUID.cpp:115] BLEUUID(): **ERROR: UUID value not 2, 4, 16 or 36 bytes**
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLERemoteCharacteristic.cpp:364] getDescriptor(): >> getDescriptor: **uuid: <NULL>**
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_FOR_NOTIFY_EVT
[D][BLERemoteCharacteristic.cpp:372] getDescriptor(): << getDescriptor: **Not found**
[Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.

chegewara, 2018-08-05:
Hi, i will check it but for now you can change it to:
`BLERemoteDescriptor *desc = getDescriptor(BLEUUID((uint16_t)0x2902));`

yuyi1005, 2018-08-06:
Thanks! Now it can work most of the time, but yet there is a problem. 

As I have five BLE servers to read data from by turns, I try to connect and disconnect for many times, but it stuck after a few minutes.

The log:

E (106920) BT_GATT: application already registered.
E (106920) BT_APPL: Register with GATT stack failed.
E (106981) BT_APPL: bta_gattc_process_api_open Failed, unknown client_if: 0

It seems that the program is stuck in: 
BLEClient.cpp
Line 99: esp_err_t errRc = ::esp_ble_gattc_app_register(0);

hetlelid, 2018-09-23:
Should the cast fix from chegewara be implementet in the codebase?

chegewara, 2018-09-23:
Could you try `BLEUUID("2902")`?

chegewara, 2018-09-23:
Its taken literally, each digit is converted to ascii, which is wrong.

What do you mean check for callbacks? No, the isn't any loop.

hetlelid, 2018-09-23:
My BLE device can both serve READ and NOTIFY on the same characteristic. So its a bit confusing. 
I mean, after i do the registerForNotify(), how to i display the value (content) coming from the notification packet?
pRemoteCharacteristic->readValue(); is for direct READ, do i use the same function for NOTIFY?
(Sorry, this is probably hijacking og the thread)

But back to the topic. Do you have any conserns about your uint16_t type cast fix?

chegewara, 2018-09-23:
You can have registerForNotify() for each remote characteristic that is serving notify or indicate, and you can in addition setCallbacks() for each characteristic that can be read/write. You can also find examples about it. 
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleRead.cpp#L25-L46

chegewara, 2018-11-06:
Fixed in here:
https://github.com/chegewara/esp32-snippets-enchancements-test/blob/multi_connect_test/components/cpp_utils/BLEUUID.cpp#L71-L126

closed 2018-11-14 by chegewara