nkolban/esp32-snippets #620: BLE abort when getCharacteristic

Caramba11, 2018-08-06:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		2018-08-01
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
Im using the "[BLE_client_numeric_confirmation.ino](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/security/BLE_client/BLE_client_numeric_confirmation.ino)" to connect to a BLE device using a six digit pin. The connection can be established, but as soon as the code runs `pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);` the ESP32 reboots.

### Sketch:
```cpp

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("6e400001-c352-11e5-953d-0002a5d5c51b");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("6e400001-c352-11e5-953d-0002a5d5c51b");

static BLEAddress *pServerAddress;
static boolean doConnect = false;
static boolean connected = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;

class MySecurity : public BLESecurityCallbacks {

  uint32_t onPassKeyRequest(){
    return 123123;
  }
  void onPassKeyNotify(uint32_t pass_key){
        ESP_LOGE(LOG_TAG, "The passkey Notify number:%d", pass_key);
  }
  bool onConfirmPIN(uint32_t pass_key){
        ESP_LOGI(LOG_TAG, "The passkey YES/NO number:%d", pass_key);
      vTaskDelay(5000);
    return true;
  }
  bool onSecurityRequest(){
    ESP_LOGI(LOG_TAG, "Security Request");
    return true;
  }
  void onAuthenticationComplete(esp_ble_auth_cmpl_t auth_cmpl){
    if(auth_cmpl.success){
      ESP_LOGI(LOG_TAG, "remote BD_ADDR:");
    //  esp_log_buffer_hex(LOG_TAG, auth_cmpl.bd_addr, sizeof(auth_cmpl.bd_addr));
      ESP_LOGI(LOG_TAG, "address type = %d", auth_cmpl.addr_type);
    }
        ESP_LOGI(LOG_TAG, "pair status = %s", auth_cmpl.success ? "success" : "fail");
  }
};

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}

bool connectToServer(BLEAddress pAddress) {
    Serial.print("Forming a connection to ");
    Serial.println(pAddress.toString().c_str());
    
    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
    BLEDevice::setSecurityCallbacks(new MySecurity());
[... 210 lines cut ...]
[D][BLEUtils.cpp:1581] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0]
[D][BLEClient.cpp:374] getServices(): << getServices
[D][BLEClient.cpp:335] getService(): << getService: found the service with uuid: 6e400001-c352-11e5-953d-0002a5d5c51b
 - Found our service
[D][BLERemoteService.cpp:167] retrieveCharacteristics(): >> getCharacteristics() for service: 6e400001-c352-11e5-953d-0002a5d5c51b
[D][BLERemoteService.cpp:198] retrieveCharacteristics(): Found a characteristic: Handle: 13, UUID: 6e400002-c352-11e5-953d-0002a5d5c51b
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 13 0x13, uuid: 6e400002-c352-11e5-953d-0002a5d5c51b
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 6e400002-c352-11e5-953d-0002a5d5c51b
[E][BLERemoteCharacteristic.cpp:308] retrieveDescriptors(): esp_ble_gattc_get_all_descr: ESP_GATT_NOT_FOUND
[D][BLERemoteCharacteristic.cpp:330] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:198] retrieveCharacteristics(): Found a characteristic: Handle: 15, UUID: 6e400003-c352-11e5-953d-0002a5d5c51b
[D][BLERemoteCharacteristic.cpp:41] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 15 0x15, uuid: 6e400003-c352-11e5-953d-0002a5d5c51b
[D][BLERemoteCharacteristic.cpp:284] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 6e400003-c352-11e5-953d-0002a5d5c51b
[E][BLERemoteCharacteristic.cpp:315] retrieveDescriptors(): 
[D][BLERemoteCharacteristic.cpp:316] retrieveDescriptors(): Found a descriptor: Handle: 16, UUID: 00002902-0000-1000-8000-00805f9b34fb
[D][BLERemoteCharacteristic.cpp:330] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[D][BLERemoteCharacteristic.cpp:49] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:214] retrieveCharacteristics(): << getCharacteristics()
abort() was called at PC 0x40184f4f on core 1

Backtrace: 0x40090c38:0x3ffd5a00 0x40090e3b:0x3ffd5a20 0x40184f4f:0x3ffd5a40 0x40184f96:0x3ffd5a60 0x4017287b:0x3ffd5a80 0x400d621d:0x3ffd5aa0 0x400d200b:0x3ffd5b00 0x400d2216:0x3ffd5b50 0x4018cb0c:0x3ffd5bb0

Rebooting..
```

chegewara, 2018-08-07:
Are you sure you have service and characteristic with the same UUID:
```
// The remote service we wish to connect to.
static BLEUUID serviceUUID("6e400001-c352-11e5-953d-0002a5d5c51b");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("6e400001-c352-11e5-953d-0002a5d5c51b");
```

skyng22003, 2018-08-07:
@Caramba11 How did you get the debug system messages above?

I think my problem is similar to your but my system messages looks slightly different 

Forming a connection to f5:e1:02:80:05:f4
 - Created client
D (1794) FreeRTOS: Semaphore taking: name: RegEvt (0x3ffe86a0), owner: <N/A> for connect
D (1802) FreeRTOS: Semaphore taken:  name: RegEvt (0x3ffe86a0), owner: connect
V (1809) FreeRTOS: >> wait: Semaphore waiting: name: RegEvt (0x3ffe86a0), owner: connect for connect
V (1810) FreeRTOS: Semaphore giving: name: RegEvt (0x3ffe86a0), owner: connect
V (1825) FreeRTOS: << wait: Semaphore released: name: RegEvt (0x3ffe86a0), owner: connect
D (1833) FreeRTOS: Semaphore taking: name: OpenEvt (0x3ffe8288), owner: <N/A> for connect
D (1841) FreeRTOS: Semaphore taken:  name: OpenEvt (0x3ffe8288), owner: connect
V (1848) FreeRTOS: >> wait: Semaphore waiting: name: OpenEvt (0x3ffe8288), owner: connect for connect
V (1899) FreeRTOS: Semaphore giving: name: OpenEvt (0x3ffe8288), owner: connect
V (1900) FreeRTOS: << wait: Semaphore released: name: OpenEvt (0x3ffe8288), owner: connect
 - Connected to server
D (1906) FreeRTOS: Semaphore taking: name: SearchCmplEvt (0x3ffe82e8), owner: <N/A> for getServices
D (1915) FreeRTOS: Semaphore taken:  name: SearchCmplEvt (0x3ffe82e8), owner: getServices
V (1922) FreeRTOS: >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffe82e8), owner: getServices for getServices
V (1938) FreeRTOS: Semaphore giving: name: RssiCmplEvt (0x3ffe7e9c), owner: <N/A>
V (1940) FreeRTOS: Semaphore giving: name: SearchCmplEvt (0x3ffe82e8), owner: getServices
V (1948) FreeRTOS: << wait: Semaphore released: name: SearchCmplEvt (0x3ffe82e8), owner: getServices
abort() was called at PC 0x401813d7 on core 1

I suspect that the authentication did not go through correctly,  as by changing the passkey in the code below doesn't change the output of the system messages, ie it still manages to connect to the server even when the passkey is wrong.

  uint32_t onPassKeyRequest(){
    return 123123;
  }

Caramba11, 2018-08-10:
Sorry for the late reply first.
@chegewara 
The issue was caused by the identical UUIDs (argh). But now the same code (just put the right UUIDs in) is not connecting to the device anymore, I will attache the new debung output.

@skyng22003 
I got these messages by switching the debug level to Verbose.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Starting Arduino BLE Client application...
[D][BLEScan.cpp:196] start(): >> start(duration=30)
[D][BLEScan.cpp:221] start(): << start()
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1229] dumpGapEvent(): [status: 0]
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_START_COMPLETE_EVT
[D][BLEUtils.cpp:1305] dumpGapEvent(): [status: 0]
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
[D][BLEUtils.cpp:1265] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 00:18:da:0a:5c:74, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -71, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 31, scan_rsp_len: 0
[D][BLEAdvertisedDevice.cpp:422] setRSSI(): - setRSSI(): rssi: -71
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x01 (ESP_BLE_AD_TYPE_FLAG), length: 1, data: 06
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x07 (ESP_BLE_AD_TYPE_128SRV_CMPL), length: 16, data: 1bc5d5a502003d95e51152c30100406e
[D][BLEAdvertisedDevice.cpp:451] setServiceUUID(): - addServiceUUID(): serviceUUID: 6e400001-c352-11e5-953d-0002a5d5c51b
[D][BLEAdvertisedDevice.cpp:251] parseAdvertisement(): Type: 0x09 (ESP_BLE_AD_TYPE_NAME_CMPL), length: 8, data: 412d304135433734
[D][BLEAdvertisedDevice.cpp:411] setName(): - setName(): name: A-0A5C74
BLE Advertised Device found: Name: A-0A5C74, Address: 00:18:da:0a:5c:74, serviceUUID: 6e400001-c352-11e5-953d-0002a5d5c51b
Found our device!  address: [D][BLEScan.cpp:244] stop(): >> stop()
[D][BLEScan.cpp:257] stop(): << stop()
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
[D][BLEUtils.cpp:1265] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 00:18:da:0a:5c:74, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -70, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 31, scan_rsp_len: 0
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
[D][BLEUtils.cpp:1265] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 00:18:da:0a:5c:74, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -68, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 31, scan_rsp_len: 0
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
[D][BLEUtils.cpp:1265] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 00:18:da:0a:5c:74, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -68, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 31, scan_rsp_len: 0
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RESULT_EVT
[D][BLEUtils.cpp:1265] dumpGapEvent(): search_evt: ESP_GAP_SEARCH_INQ_RES_EVT, bda: 00:18:da:0a:5c:74, dev_type: ESP_BT_DEVICE_TYPE_BLE, ble_addr_type: BLE_ADDR_TYPE_PUBLIC, ble_evt_type: ESP_BLE_EVT_CONN_ADV, rssi: -65, ble_adv: ??, flag: 6 ([LE General Discoverable Mode] [BR/EDR Not Supported] ), num_resps: 1, adv_data_len: 31, scan_rsp_len: 0
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT
[D][BLEUtils.cpp:1317] dumpGapEvent(): [status: 0]
Forming a connection to 00:18:da:0a:5c:74
[D][BLEDevice.cpp:54] createClient(): >> createClient
[D][BLEDevice.cpp:60] createClient(): << createClient
 - Created client
[D][BLEClient.cpp:91] connect(): >> connect(00:18:da:0a:5c:74)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_EVT
[D][BLEUtils.cpp:1550] dumpGattClientEvent(): [status: ESP_GATT_OK, app_id: 0x0]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_CONNECT_EVT
[D][BLEUtils.cpp:1407] dumpGattClientEvent(): [conn_id: 0, remote_bda: 00:18:da:0a:5c:74]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1503] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, remote_bda: 00:18:da:0a:5c:74, mtu: 23]
[... cut ...]
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_WRITE_CHAR_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_WRITE_CHAR_EVT
[D][BLEUtils.cpp:1623] dumpGattClientEvent(): [status: ESP_GATT_INSUF_AUTHENTICATION, conn_id: 0, handle: 13 0x0d, offset: 0]
[D][BLERemoteCharacteristic.cpp:579] writeValue(): << writeValue
Setting new characteristic value to "Time since boot: 3"
[D][BLERemoteCharacteristic.cpp:551] writeValue(): >> writeValue(), length: 18
`

Thanks for your help