nkolban/esp32-snippets #624: How to know that BLE notify has been sent?

tttapa, 2018-08-12:
The `BLE_notify` example just has a delay of 10 ms after `notify()`. Is there a better, more failsafe way to know when it's safe to send the next notification?

https://github.com/nkolban/esp32-snippets/blob/7aaed6d3355bd0b251bc9386b742ee0699ee5a71/cpp_utils/tests/BLETests/Arduino/BLE_notify/BLE_notify.ino#L89-L91

When sending multiple notifications without delay, only the first one shows up in WireShark.

The application is a MIDI BLE transmitter. The idea is to buffer the outgoing MIDI messages, then notify to send all messages in the buffer, and then fill the buffer again, notify again, etc.  
It's not entirely clear to me how long I have to wait before I can send the next notification.

There are other things running at the same time, and I want to keep latency as low as possible, so I'd prefer not to use delay, and check some kind of flag instead.

chegewara, 2018-08-12:
https://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv230esp_ble_gap_update_conn_paramsP28esp_ble_conn_update_params_t
https://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv228esp_ble_conn_update_params_t

tttapa, 2018-08-12:
I see. So I just wait for the maximum latency to be sure? How do I know what the maximum latency is?

> When sending multiple notifications without delay, only the first one shows up in WireShark.

I was wrong. They do all show up in WireShark.

robson31, 2018-10-31:
Is there a way to know when function `notify()` is done?

robson31, 2018-10-31:
@tttapa Did you used the connection parameters in your implementation as @chegewara suggested? Did you figured out a way to know the max latency? Thank you.

closed 2018-11-13 by chegewara