nkolban/esp32-snippets #625: BLE Client example crashes

relax81, 2018-08-13:
Hello, I'm just trying to get started with all the Bluetooth and BLE stuff but the BLE Client example (not the other ones) always result in crashes. 

Serial Monitor: 
`Found our device!  address: Forming a connection to fd:0d:cb:c8:66:51
 - Created client
 - Connected to server
 - Found our service
abort() was called at PC 0x401810d3 on core 1

Backtrace: 0x40090c38:0x3ffd5ab0 0x40090e3b:0x3ffd5ad0 0x401810d3:0x3ffd5af0 0x4018111a:0x3ffd5b10 0x4016ebb3:0x3ffd5b30 0x400d4579:0x3ffd5b50 0x400d172f:0x3ffd5bb0 0x400d192e:0x3ffd5c00 0x40188c90:0x3ffd5c60
`

Esception Decoder
`
Decoding stack results
0x40090c38: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x40090e3b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x401810d3: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x4018111a: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x4016ebb3: __cxxabiv1::__cxa_throw(void*, std::type_info*, void (*)(void*)) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc line 87
0x400d4579: BLERemoteService::getCharacteristic(BLEUUID) at C:\Users\Leif\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\BLE\src\BLERemoteService.cpp line 156
0x400d172f: connectToServer(BLEAddress) at C:\Users\Leif\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\BLE\examples\BLE_client/BLE_client.ino line 51
0x400d192e: loop() at C:\Users\Leif\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\BLE\examples\BLE_client/BLE_client.ino line 114
0x40188c90: loopTask(void*) at C:\Users\Leif\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\main.cpp line 17
`

I've installed the latest ESP32 v1.0.0 files through the Arduino IDE BoardManager

chegewara, 2018-08-13:
If you get here `Found our service` then i can assume you have issue with getting remote characteristic. Check if you have right UUIDs and/or add some assertion that will test if remote service and/or remote characteristic are not null before you try to get any actions on it.

relax81, 2018-08-14:
Thank you very much for pointing me in the right direction. 

I probably did a total beginner failure and didn't notice that the service and characteristic have a different  UUID - there was a difference in one number. Now it works perfectly fine.

closed 2018-08-14 by relax81