nkolban/esp32-snippets #626: array subscript has type 'char'

burkulesomesh43, 2018-08-15:
I am using isxdigit() function given in ctype.h librairy.
but when i am calling this function it gives an error.

error: array subscript has type 'char' 

char *urlDecode(const char *str) {

  char *dStr = (char *) malloc(strlen(str) + 1);
  char eStr[] = "00"; /* for a hex code */

  strcpy(dStr, str);
    int i; /* the counter for the string */

    for(i=0;i<strlen(dStr);++i) {

      if(dStr[i] == '%') {
        if(dStr[i+1] == 0)
          return dStr;

        if(isxdigit(dStr[i+1]) && isxdigit(dStr[i+2])) {  

          //d = 0;

          /* combine the next to numbers into one */
          eStr[0] = dStr[i+1];
          eStr[1] = dStr[i+2];

          /* convert it to decimal */
          long int x = strtol(eStr, NULL, 16);

          /* remove the hex */
          memmove(&dStr[i+1], &dStr[i+3], strlen(&dStr[i+3])+1);

          dStr[i] = x;
        }
      }
      else if(dStr[i] == '+') { dStr[i] = ' '; }
    }
  return dStr;
}

How to remove this error??

nkolban, 2018-08-15:
Howdy, is this report related to some code found in this repository or is it a general programming question?  If the later, I'd suggest posting to the ESP32 forum found here:

https://esp32.com/index.php

burkulesomesh43, 2018-08-15:
actually the code is suggested by you in esp32 forum today.

nkolban, 2018-08-15:
Ahh ... I think I'm with you.   You asked a question on the forum this morning to which I believe I responded with a link to this Github repository:

https://github.com/swarajsatvaya/UrlDecoder-C

That repository is owned by another user.  If you are having problems with the code found there, may I suggest that you raise an issue with that code on that repository?   There is no code in this repository that relates to Url decoding.

burkulesomesh43, 2018-08-15:
Ok. thank you for your response.
but if you know how to solve this problem please help me out.

problem is with isxdigit() function given in ctype.h librairy.

Matszs, 2018-08-27:
@burkulesomesh43 
Had exactly the same error with the same library, you can fix by casting the char to an int:

find: 
`if(isxdigit(dStr[i+1]) && isxdigit(dStr[i+2])) {`

replace:
`if(isxdigit((int)dStr[i+1]) && isxdigit((int)dStr[i+2])) {`

burkulesomesh43, 2018-08-27:
Yes, 
I got it.
thank you

closed 2018-08-27 by burkulesomesh43