nkolban/esp32-snippets #627: BLE security: allow bonding for first 2 minutes only

whyameye, 2018-08-16:
I've been playing with the BLE_server_authorization example at https://github.com/nkolban/esp32-snippets/pull/315 which seems to work great, bonding successfully with my Android device. I would now like to add a new feature and, being new to BLE, it's unclear to me how to do it:

2 minutes after boot, I'd like for the ESP32 server to no longer create new bonds or pairs and therefore not connect to any devices it has not previously bonded to. However I would like it to still be able to connect to devices with preexisting bonds. So if the connection has been dropped from a device the ESP32 server has previously bonded to, the server and client can reconnect. But if there is no bond, no connection would be possible after 2 minutes.

My first thought was to stop advertising but it seems to me a server cannot connect to client at all, bond or no bond, if it is not already connected and not advertising. So I'm wondering if there is a way to stop the bonding and pairing process but still allow the connections and require the security that comes with bonding/pairing.

Thanks for any suggestions you may have.

chegewara, 2018-08-17:
Maybe after 2 minutes change advertising. You can set this:
`pSecurity->setCapability(ESP_IO_CAP_OUT);` or `pSecurity->setCapability(ESP_IO_CAP_IN);`.
If esp32 does not have output to display passkey or keypad to input passkey no other device can be bonded.

whyameye, 2018-08-17:
Thanks for the suggestion. That technically works although it's not a very elegant solution as the user is still prompted to enter a pairing code on either the phone or the ESP32. I wish they were not prompted at all to do something they can't do but instead they were given an error.

chegewara, 2018-08-17:
I agree its not very elegant. I would suggest to try with advertising/connecting whitelisted devices, but there seems to be some bug in esp-idf stack or i dont understand how this should have works.

whyameye, 2018-08-17:
yes advertising to only whitelisted devices would be the ideal solution.

simfero, 2019-05-26:
I tried using the workaround suggested by @chegewara for a similar use case to @whyameye and it does work with an android smartphone, but when I try connecting with another ESP32 client instead of smartphone (i.e. ESP32 client to ESP32 server), connection and pairing succeeds regardless of the capability setting on the server. I believe this is because my ESP32 BLE Client has its capability set to ESP_IO_CAP_NONE, so ESP32 Server lets it connect without requiring a pin. This creates a security loophole which is unacceptable in my application. I know whitelist i the correct way to implement this, but apparently it doesn't work correctly on ESP32. Can anyone suggest another workaround?

Prajwal8520, 2024-10-30:
@simfero I am working on something similar to what you were working, and I am having same Issue, where ESP32 BLE Server let's ESP32 BLE client to connect and pair easily. It created huge loophole. I was wondering how you solved this problem. I am fairly new in this so, it would be great if you anyone can help.