nkolban/esp32-snippets #628: BLE registerForNotify not working (with workaround)

petereverett, 2018-08-18:
Hi

I'm using v0.4.16 of the ESP32 BLE Arduino Library.
I have created a BLE server on an ESP32 board with 2 characteristics based on the BLE_notify example.  This works as expected with the nRF Connect app on my phone:  I can read values and get notifications.

I have created a BLE client on a second ESP32 board based on the BLE_client example, to connect to the server, register for notifications and output the value (on LEDs).  This connects to the server and reads the initial value correctly.   However this line does not work:
`pRemoteCharacteristic->registerForNotify(notifyCallback);`
I don't get an error, it just doesn't receive any notifications.

If I do the following, everything works fine:
```
const uint8_t notifyOn[] = {0x1, 0x0};
pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notifyOn, 2, true);
pRemoteCharacteristic->registerForNotify(notifyCallback);
```

The following code does not work either:
```
const uint8_t notifyOn[] = {0x1, 0x0};
pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notifyOn, 2, true);
```

Is 'registerForNotify()' supposed to work when called without also writing to the descriptor manually?

If I first connect with the nRF Connect app, register for notifications, disconnect and reboot the client board, 'registerForNotify()' on its own works fine and receives the notifications.  Simply connecting and disconnecting in nRF Connect didn't make it work, I had to register for the notifications.

I think there is something wrong with the server side of things, because I built a seperate project last week to display the output from a Speed/Cadence sensor, and only used 'registerForNotify()', and it works great.

I've attached the server code as a text file - can't upload an ino file.
[BLE_server.txt](https://github.com/nkolban/esp32-snippets/files/2299546/BLE_server.txt)

chegewara, 2018-08-18:
This register for notifications only on local esp32 device only:
`pRemoteCharacteristic->registerForNotify(notifyCallback);`

On arduino you still need this workaround. Its been added to this repository, but no new arduino version has been generated for some time and changes are not included yet.

EBZ-Krisemendt, 2018-08-31:
Unfortunatly the workaround isn't working on my device...

I'm doing a manual reading of the characteristic to make sure measurement is working:
```
uint32_t value = pRemoteCharacteristic->readUInt32();
Serial.printf("The characteristic value was: %#08x\n", value);
```
which results in this output:
>[D][BLERemoteCharacteristic.cpp:434] readValue(): >> readValue(): uuid: f000aa21-0451-4000-b000-000000000000, handle: 44 0x2c
>[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_READ_CHAR_EVT
>[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_READ_CHAR_EVT
>[D][BLEUtils.cpp:1527] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, handle: 44 0x2c, value_len: 4]
>[D][BLERemoteCharacteristic.cpp:462] readValue(): << readValue(): length: 4
>The characteristic value was: 0x5cdc6720

all fine here.

now i try to activate notification (with workaround)
```
const uint8_t notifyOn[] = {0x01, 0x00};
pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notifyOn, 2, true);
pRemoteCharacteristic->registerForNotify(notifyCallback);
```
output:
>[D][BLERemoteCharacteristic.cpp:359] getDescriptor(): >> getDescriptor: uuid: 00002902-0000-1000-8000-00805f9b34fb
>[D][BLERemoteCharacteristic.cpp:363] getDescriptor(): << getDescriptor: found
>[D][BLERemoteDescriptor.cpp:138] writeValue(): >> writeValue: handle: 45, uuid: 00002902-0000-1000-8000-00805f9b34fb
>[D][BLERemoteDescriptor.cpp:157] writeValue(): << writeValue
>[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_WRITE_DESCR_EVT
>[D][BLERemoteCharacteristic.cpp:479] registerForNotify(): >> registerForNotify(): Characteristic: uuid: f000aa21-0451-4000-b000-000000000000, handle: 44 0x2c, props: broadcast: 0, read: 1, write_nr: 0, write: 0, n
otify: 1, indicate: 0, auth: 0
>[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_WRITE_DESCR_EVT
>[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_FOR_NOTIFY_EVT
>[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_REG_FOR_NOTIFY_EVT
>[D][BLEUtils.cpp:1566] dumpGattClientEvent(): [status: ESP_GATT_OK, handle: 44 0x2c]
>[D][BLERemoteCharacteristic.cpp:510] registerForNotify(): << registerForNotify()
>[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
>[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_NOTIFY_EVT
>[D][BLEUtils.cpp:1485] dumpGattClientEvent(): [conn_id: 0, remote_bda: 98:07:2d:45:cd:00, handle: 44 0x2c, value_len: 4, is_notify: 1]
>[D][BLERemoteCharacteristic.cpp:174] gattClientEventHandler(): Invoking callback for notification on characteristic Characteristic: uuid: f000aa21-0451-4000-b000-000000000000, handle: 44 0x2c, props: broadcast: 0,
 read: 1, write_nr: 0, write: 0, notify: 1, indicate: 0, auth: 0
>[D][BLERemoteCharacteristic.cpp:434] readValue(): >> readValue(): uuid: f000aa21-0451-4000-b000-000000000000, handle: 44 0x2c

apparently (according to debug message) there is one last `readValue()`-call but it's not followed by those `gattClientEventHandler()` and `dumpGattClientEvent()` calls which are occuring on the manual reading. can someone tell me what i'm missing here??

Notification work fine with nRF-Connect.

chegewara, 2018-08-31:
@EBZ-Krisemendt I have one question, because logs shows that notification is received, are you trying to read characteristic value from within notificationCallback? If yes then it wont work, because for some reason (im guessing its some issue with semaphores) any action (read/write) called from notification callback is blocking whole app. You need to set some bool variable when you receive notification and in loop if variable is true then read characteristic.

EBZ-Krisemendt, 2018-09-06:
@chegewara to anwser your question: yes, i did indeed. changed it to a volatile bool "trigger" variable and now it works like a charm. thx very much.

maybe this is worth noting in the comments within the example code to prevent others from falling in this pit.

chegewara, 2018-09-06:
To be honest it was very annoying, i lost many hours before i found it by myself.