nkolban/esp32-snippets #631: ESP_GATT_INSUF_ENCRYPTION

hetlelid, 2018-08-24:
Thanks a lot for your work Neil!

When using the BLE_Client example i get ESP_GATT_INSUF_ENCRYPTION in response from a sensor device (see below). The device is not using encryption, as i can read values from the nRF connect app.

in BLEUtils.cpp i implemented the read (similar to ESP_GATT_OK): 
`if (evtParam->read.status == ESP_GATT_INSUF_ENCRYPTION) {
  GeneralUtils::hexDump(evtParam->read.value, evtParam->read.value_len);
}`

No success. Any suggestions, how to proceed?

Debug:
`[D][BLERemoteCharacteristic.cpp:439] readValue(): >> readValue(): uuid: 00002a6e-0000-1000-8000-00805f9b34fb, handle: 25 0x19
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1527] dumpGattClientEvent(): [status: ESP_GATT_INSUF_ENCRYPTION, conn_id: 0, handle: 25 0x19, value_len: 0]
[D][BLERemoteCharacteristic.cpp:467] readValue(): << readValue(): length: 0`

hetlelid, 2018-08-26:
Thanks for reply, i agree.

Do you have experience with bonding? Is it implemented in the cpp library?

For reference, i found this document that i will try in a few days: https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/tutorial/Gatt_Security_Server_Example_Walkthrough.md

chegewara, 2018-08-27:
I cant say im expert, but know some about it (BLESecurity is my code).

First of all you have to understand how your sensor works. In my opinion best to do it is to use nRF connect on android. Its easy to use and easy to debug all messages or see how it is constructed from ble point of view (services, characteristics, descriptors etc). You will also see what kind of pairing/bonding it requires when you will try to read/write protected characteristic.

hansmbakker, 2018-12-15:
@hetlelid there has been no activity in this thread for several months. If your issue is solved, could you please close it to keep the issue list clean?
cc @chegewara

hetlelid, 2018-12-16:
Closed #631