nkolban/esp32-snippets #632: BLE HID Server - Reconnect

simon-jouet, 2018-08-25:
Hi, 

First off, thanks for the great work on the library and the example, it's surprisingly easy to get started!

So i'm trying to use the ESP32 as a Gamepad (mostly just for fun)  - with my current code the gamepad is working okay I can pair it on windows/linux and start issuing button presses. The issue i've got is that any time the ESP32 reboots the client reconnects fine but the events stop working until I remove the device and repair it.

First I tried to switch to linux for testing, as I remembered (from a while ago) that BLE devices on windows used to be a bit flaky but the issue is the same on linux. Second I tried to setup security but that wasn't a great success as I couldn't get blueman to prompt me for a passkey (I probably have done something wrong there).

So after the first pairing it works fine:
[image]

But it stops working when the ESP32 reboot and the client reconnects. Any suggestions?

```c
#include <Arduino.h>

/**
 * Create a new BLE server.
 */
#include "BLEDevice.h"
#include "BLEServer.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include "BLEHIDDevice.h"

static BLEHIDDevice* hid;
BLECharacteristic* input;
BLECharacteristic* output;

class MyCallbacks : public BLEServerCallbacks {
	void onConnect(BLEServer* pServer){
		Serial.println("connected");
	}

	void onDisconnect(BLEServer* pServer){
        Serial.println("disconnected");
	}
};

void setup() {
    //
    Serial.begin(115200);
    Serial.println("Starting");

    //
    BLEDevice::init("ESP32");

    BLEServer *pServer = BLEDevice::createServer();
    pServer->setCallbacks(new MyCallbacks());

    /*
     * Instantiate hid device
     */
    hid = new BLEHIDDevice(pServer);

    input = hid->inputReport(1); // <-- input REPORTID from report map
    output = hid->outputReport(1); // <-- output REPORTID from report map

    /*
     * Set manufacturer name (OPTIONAL)
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.manufacturer_name_string.xml
     */
    std::string name = "esp-community";
    hid->manufacturer()->setValue(name);

    /*
     * Set pnp parameters (MANDATORY)
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.pnp_id.xml
     */
    hid->pnp(0x02, 0x0810, 0xe501, 0x0106);

    /*
     * Set hid informations (MANDATORY)
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.hid_information.xml
     */
    hid->hidInfo(0x00,0x01);

    /*
     * Gamepad
     */
    const uint8_t reportMap[] = {
        0x05, 0x01,  /* USAGE_PAGE (Generic Desktop)       */
        0x09, 0x05,  /* USAGE (Game Pad)                   */
        0xa1, 0x01,  /* COLLECTION (Application)           */
        0xa1, 0x03,  /*   COLLECTION (Report)              */
        0x85, 0x01,  /*     REPORT_ID (1)                  */
        0x05, 0x09,  /*     USAGE_PAGE (Button)            */
        0x19, 0x01,  /*     USAGE_MINIMUM (Button 1)       */
        0x29, 0x0e,  /*     USAGE_MAXIMUM (Button 14)      */
        0x15, 0x00,  /*     LOGICAL_MINIMUM (0)            */
        0x25, 0x01,  /*     LOGICAL_MAXIMUM (1)            */
        0x95, 0x0e,  /*     REPORT_COUNT (14)              */
        0x75, 0x01,  /*     REPORT_SIZE (1)                */
        0x81, 0x02,  /*     INPUT (Data,Var,Abs)           */
        0x95, 0x01,  /*     REPORT_COUNT (1)               */
        0x75, 0x02,  /*     REPORT_SIZE (2)                */
        0x81, 0x03,  /*     INPUT (Cnst)                   */
        0xa1, 0x00,  /*     COLLECTION (Physical)          */
        0x05, 0x01,  /*       USAGE_PAGE (Generic Desktop) */
        0x09, 0x30,  /*       USAGE (X)                    */
        0x09, 0x31,  /*       USAGE (Y)                    */
        0x15, 0x00,  /*       LOGICAL_MINIMUM (0)          */
        0x25, 0x64,  /*       LOGICAL_MAXIMUM (100)        */
        0x75, 0x08,  /*       REPORT_SIZE (8)              */
        0x95, 0x02,  /*       REPORT_COUNT (2)             */
        0x81, 0x02,  /*       INPUT (Data,Var,Abs)         */
        0xc0,        /*     END_COLLECTION                 */
        0xc0,        /*   END_COLLECTION                   */
        0xc0         /* END_COLLECTION                     */
    };

    /*
     * Set report map (here is initialized device driver on client side) (MANDATORY)
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.report_map.xml
     */
    hid->reportMap((uint8_t*)reportMap, sizeof(reportMap));

    /*
     * We are prepared to start hid device services. Before this point we can change all values and/or set parameters we need.
     * Also before we start, if we want to provide battery info, we need to prepare battery service.
     * We can setup characteristics authorization
     */
    hid->startServices();

    /*
     * Its good to setup advertising by providing appearance and advertised service. This will let clients find our device by type
     */
    BLEAdvertising *pAdvertising = pServer->getAdvertising();
    pAdvertising->setAppearance(HID_GAMEPAD);    // <-- optional
    pAdvertising->addServiceUUID(hid->hidService()->getUUID());
    pAdvertising->start();

    BLESecurity *pSecurity = new BLESecurity();
    pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

    Serial.println("waiting 5sec");
    delay(5000);
}

void loop() {
    // put your main code here, to run repeatedly:
    uint8_t a[] = {0x03, 0x03, 0x03, 0x03};

    input->setValue(a, sizeof(a));
    input->notify();

    delay(2000);

    uint8_t b[] = {0x00, 0x00, 0x00, 0x00};

    input->setValue(b, sizeof(b));
    input->notify();

    delay(2000);
}
```

Thanks!

chegewara, 2018-08-25:
Hi, thanks for kind words. We are glad you found this library useful. You're right, there is small glitch that i have not found. In this library each time device is connecting its expected that peer device will register for notify. Seems that windows and maybe other devices too dont do it when ble device is already paired.

This is small code snippet you can use to workaround this:
```
class MyCallbacks : public BLEServerCallbacks {
  void onConnect(BLEServer* pServer){
    Serial.println("connected");
    BLEDescriptor *desc = input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
    uint8_t val[] = {0x01, 0x00};
    desc->setValue(val, 2);
  }

  void onDisconnect(BLEServer* pServer){
        Serial.println("disconnected");
  }
};
```

Thanks for posting this issue.

simon-jouet, 2018-08-25:
Great, thanks @chegewara that's working fine with this workaround I can reset the ESP32 and the gamepad inputs are registered as required. I will update the original post with the working copy so people don't try to copy paste something that's not working.

closed 2018-08-25 by simon-jouet

rupin, 2019-03-19:
@simon-jouet Have you tried using your code and tried to see what buttons show up pressed and not pressed using the HTML API? https://html5gamepad.com/

I used your code and it works like a charm. There are some small incosistencies I have found, which I am not able to find a logical reason either by experimentation or online. 

When the report is sent the first time, assume the first byte within the array has a value of 0x43. Will it show the binary representation based on which bits are turned on and which ones are not? I see that thr third LSB from the right never gets mapped to any of the buttons ( and so does the MSB). The MSB in both the nibbles is always ignored, to say it more clearly.

Couby, 2019-03-19:
Hi,
@simon-jouet @chegewara  how/where do you define/map gamepad buttons and minitsick wiring on the ESP32 ? Do we have control on that or is it something static deep in included files ?
Best regards.

chegewara, 2019-03-19:
Hi,
you have to control it in your app. All you need to map buttons and sticks is encoded in report map. For example:
```
const uint8_t reportMap[] = {
        0x05, 0x01,  /* USAGE_PAGE (Generic Desktop)       */
        0x09, 0x05,  /* USAGE (Game Pad)                   */
        0xa1, 0x01,  /* COLLECTION (Application)           */
        0xa1, 0x03,  /*   COLLECTION (Report)              */
        0x85, 0x01,  /*     REPORT_ID (1)                  */
        0x05, 0x09,  /*     USAGE_PAGE (Button)            */
        0x19, 0x01,  /*     USAGE_MINIMUM (Button 1)       */
        0x29, 0x0e,  /*     USAGE_MAXIMUM (Button 14)      */ <--- 14 bits (in 2 bytes) -- buttons
        0x15, 0x00,  /*     LOGICAL_MINIMUM (0)            */
        0x25, 0x01,  /*     LOGICAL_MAXIMUM (1)            */
        0x95, 0x0e,  /*     REPORT_COUNT (14)              */
        0x75, 0x01,  /*     REPORT_SIZE (1)                */
        0x81, 0x02,  /*     INPUT (Data,Var,Abs)           */
        0x95, 0x01,  /*     REPORT_COUNT (1)               */
        0x75, 0x02,  /*     REPORT_SIZE (2)                */ <--- + 2 bits (in the same 2 bytes) -- padding
        0x81, 0x03,  /*     INPUT (Cnst)                   */
        0xa1, 0x00,  /*     COLLECTION (Physical)          */
        0x05, 0x01,  /*       USAGE_PAGE (Generic Desktop) */
        0x09, 0x30,  /*       USAGE (X)                    */ <--- AXIS
        0x09, 0x31,  /*       USAGE (Y)                    */ <--- AXIS
        0x15, 0x00,  /*       LOGICAL_MINIMUM (0)          */
        0x25, 0x64,  /*       LOGICAL_MAXIMUM (100)        */
        0x75, 0x08,  /*       REPORT_SIZE (8)              */  
        0x95, 0x02,  /*       REPORT_COUNT (2)             */ <--- next 2 bytes for AXES
        0x81, 0x02,  /*       INPUT (Data,Var,Abs)         */
        0xc0,        /*     END_COLLECTION                 */
        0xc0,        /*   END_COLLECTION                   */
        0xc0         /* END_COLLECTION                     */
    };
```

So, you have to send 4 bytes in total. 2 bytes for buttons with 2 most significant bits not used, and 2 bytes for AXES. 

https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/

Couby, 2019-03-19:
Woho, thanks for this very quick answer.
I have to admit that I'm missing something. Will reread all this after a good night :blush: .

rupin, 2019-03-20:
@chegewara 
So If I change the number of buttons to 16 ( by changing USAGE_MAXIMUM to 0x10, report count also to 0x10), and remove the report map entries for the +2 bits, I should be able to send status of 16 buttons? Can I send these 16 buttons in two parts by changing the REPORT_SIZE and REPORT_COUNT variables?

rupin, 2019-03-20:
Here is how my loop section looks like. Notice, I am sending 0xFF for the first two bytes, so all buttons should light up. 

```

void loop() {
   
   
    uint8_t a[] = {0xff,0xff,0x55, 0x55};
    input->setValue(a, sizeof(a));
    input->notify();
    delay(2000);
    
    uint8_t b[] = {0x00, 0x00, 0x00, 0x00};
    input->setValue(b, sizeof(b));
    input->notify();

    delay(2000);
    
}

```
Here is how my report Map looks like. 

```

 const uint8_t reportMap[] = {
        0x05, 0x01,  /* USAGE_PAGE (Generic Desktop)       */
        0x09, 0x05,  /* USAGE (Game Pad)                   */
        0xa1, 0x01,  /* COLLECTION (Application)           */
        0xa1, 0x03,  /*   COLLECTION (Report)              */
        0x85, 0x01,  /*     REPORT_ID (1)                  */
        0x05, 0x09,  /*     USAGE_PAGE (Button)            */
        0x19, 0x01,  /*     USAGE_MINIMUM (Button 1)       */
        0x29, 0x10,  /*     USAGE_MAXIMUM (Button 16)      */
        0x15, 0x00,  /*     LOGICAL_MINIMUM (0)            */
        0x25, 0x01,  /*     LOGICAL_MAXIMUM (1)            */
        0x95, 0x10,  /*     REPORT_COUNT (8)              */
        0x75, 0x01,  /*     REPORT_SIZE (2)                */
        0x81, 0x02,  /*     INPUT (Data,Var,Abs)           */        
        0xa1, 0x00,  /*     COLLECTION (Physical)          */
        0x05, 0x01,  /*       USAGE_PAGE (Generic Desktop) */
        0x09, 0x30,  /*       USAGE (X)                    */
        0x09, 0x31,  /*       USAGE (Y)                    */
        0x15, 0x00,  /*       LOGICAL_MINIMUM (0)          */
        0x25, 0x64,  /*       LOGICAL_MAXIMUM (100)        */
        0x75, 0x08,  /*       REPORT_SIZE (8)              */
        0x95, 0x02,  /*       REPORT_COUNT (2)             */
        0x81, 0x02,  /*       INPUT (Data,Var,Abs)         */
        0xc0,        /*     END_COLLECTION                 */
        0xc0,        /*   END_COLLECTION                   */
        0xc0         /* END_COLLECTION                     */
    };
```

Here is how the HTML5Gamepad reports which buttons are turning on.  https://html5gamepad.com/

[image]

The mapping is also non linear, and Meaning Bit X does not always map to button X,  and sometimes there is no mapping. 

Let me try tinkering around a little bit more to understand what is going on, but if you see something amiss, let me know

Couby, 2019-03-20:
Hi @chegewara , things definitely look more crystal clear after a good night 😊  .
Reading the article your referred to, I now have a better understanding of the way all this work. And I think now I'll be able to customize the HID descriptor. Fantastic 😎 ❗️ 

I understand through this descriptor how we define the structure of data sent to the device we connect our ESP32.

The part I'm still missing, is the way to construct and update this data packet.
I understand relatively to my initial question, that the buttons/axis wiring will fuel this data packet.

In the @simon-jouet given exemple, I understand the update is done through `input->notify();`.

The refresh rate is defined by `delay(2000);`.

The data for four buttons : 
```
uint8_t a[] = {0x03, 0x03, 0x03, 0x03};
input->setValue(a, sizeof(a));
```

The data for axis : 
```
uint8_t b[] = {0x00, 0x00, 0x00, 0x00};
input->setValue(b, sizeof(b));
```

I feel so close to the goal, I'm super excited 🚀 ...
That may be obvious to ESP32 folks - and that's because I'm starting in that field by this door - how do I link my ESP32 pins to these data ❓ 
I imagine that depends on the ESP32 board I'm using...
`0x03` and `0x00' represent adresses ?
How do I identify pin adresses of my board ? Or is it something to define ?

I find this in an example you give in another topic : 
```
  pinMode(38, INPUT_PULLDOWN);
  attachInterrupt(digitalPinToInterrupt(38), clickNumLock, CHANGE);   // Num Lock
```
Wouldn't we need something like that in the present example to define the button behavior with the pin it is combined with ?

chegewara, 2019-03-20:
Data for buttons AND axis:
```
uint8_t a[] = {0x03, 0x03, 0x03, 0x03};
input->setValue(a, sizeof(a));
```
data for button and axis to tell - no buttons pressed, no axis moved:
```
uint8_t b[] = {0x00, 0x00, 0x00, 0x00};
input->setValue(b, sizeof(b));
```

chegewara, 2019-03-20:
@rupin from this website
https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/
```
USAGE_PAGE (Button)
USAGE_MINIMUM (Button 1)
USAGE_MAXIMUM (Button 16)
LOGICAL_MINIMUM (0)
LOGICAL_MAXIMUM (1)
REPORT_COUNT (16)
REPORT_SIZE (1)
INPUT (Data,Var,Abs)
```

Couby, 2019-03-20:
@chegewara oh, many thanks, I think I understand 😊 .

a[x, y, v, w] and b[x, y, v, w] arrays are mapreport updates.
So what I need to to, is to map PIN GPIO interrupt to define x, y, v, w bytes accordingly and update.
Will try that 👍 .