nkolban/esp32-snippets #633: ServiceUUID missmatch when scanning for a BLEServer

EBZ-Krisemendt, 2018-08-27:
Hi.

I'm currently trying to connect my ESP32 to a TI SensorTag CC2650 [List_of_UUIDs](http://processors.wiki.ti.com/images/9/9c/SensorTag_attribute_table.pdf)
Since i'm pretty new to BLE, i started by implementing the BLEScan.ino and BLEClient.ino in my code... but ran into a problem.

scan result says:
> Advertised Device: Name: CC2650 SensorTag, Address: 54:6c:0e:53:13:03, manufacturer data: 0d00030000, serviceUUID: 0000aa80-0000-1000-8000-00805f9b34fb

which is apparently the correct device.
unfortunatly i dont find that serviceUUID anywhere in the docs?!
as i connect this sensortag to my phone (with nordic app) i get the correct serviceUUID "0x1800" (Generic Access / Primary Service).
when i try to generate a UUID using
`static BLEUUID sensortagUUID("0x1800");`
it is `<NULL>` as soon as i try to output it
`Serial.printf("uuid: %s", sensortagUUID.toString().c_str());`

might this be an error in scan/detection of devices or is my understanding of generating UUIDs wrong??

chegewara, 2018-08-27:
Hi, as you said you are new and might not understand ble and this library. Scan result says that advertised service uuid is:
`0000aa80-0000-1000-8000-00805f9b34fb`.
You have to match this uuid with serviceUUID in this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_client/BLE_client.ino#L78

From sensor attributes table i see you will have much more questions about ble before you end your app.

EBZ-Krisemendt, 2018-08-27:
thx for your response!

if i perform a check as mentioned in BLE_client.ino it would not matter, since i still don't know if it's actually a "valid" serviceUUID. i'm kinda struggeling to explain in english to be honest, but will try anyway:
`BLEAdvertisedDevice::toString()` internally calls `BLEAdvertisedDevice::getServiceUUID()`. So if i just note/store this one and match it as done in the example, i actually match it with itself.

so i sure get the serviceUUID as mentioned before (`0000aa80-0000-1000-8000-00805f9b34fb`) BUT in https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEUUID.cpp#L358 this is just the "filled" version of a 16bit UUID. so the actual UUID is just `aa80`. but this is neither in the docs nor reproduceable by using nRF-App.
the EXPECTED uuid is in fact `0x1800` which is correctly detected using nRF-App
and if i try to generate a uuid from string `0x1800` https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEUUID.cpp#L256 (comment in source says it should accept 0xNNNN) it doesnt work at all (recieving `<NULL>`) so i'm not sure where to look for the error.

chegewara, 2018-08-27:
You have to understand how it works. Not always advertised uuid is the same as the service uuid's sensor is providing after connection. More, you can have 1 or 2 service uuids advertised but you can have many services in sensor, just like your sensor tag.

Now, like i said, before you connect you want to check if peer device is advertising any serviceUUID and if it does, then you want to know if its `0000aa80-0000-1000-8000-00805f9b34fb` or `0xaa80` in your case (they are both the same).

In most cases you are not interested in services 0x1800, 0x1801 and 0x180A because that are services required by bluetooth specification and every device got it.

When you will look at this document with sensor attributes table you will see there is no uuid 0000aa80-0000-1000-8000-00805f9b34fb at all, it is only used for advertising.

EBZ-Krisemendt, 2018-08-27:
so it's basiclly the same issue as in https://github.com/nkolban/esp32-snippets/issues/604 but since i dont find the advertised uuid in my case we ran just into some sort of language barrier?

chegewara, 2018-08-27:
Im guessing it can be language barrier until you dont understand how ble works.

From document you pasted link I can see that sensorTag does not use service uuid `0000aa80-0000-1000-8000-00805f9b34fb` at all. So, i can only assume that uuid `0000aa80-0000-1000-8000-00805f9b34fb` or `0xaa80` is only used for advertising (from logs you provided).

EBZ-Krisemendt, 2018-08-27:
guess thats right.
well, i this particular case i dont see any point in checking uuid before connecting then. checking `haveServiceUUID()` is fair, since i kinda wanna know if the device is advertising at all. but if i can't validate the uuid using official documentation i'll better check on some of the provided advertising data like "name" or "manufacturer" to be sure to connect the correct device.

still got the related/mentioned issue on creating UUIDs:
```
static BLEUUID uuid1("f000aa00-0451-4000-b000-000000000000");
Serial.println(uuid1.toString().c_str());
// expected: "f000aa00-0451-4000-b000-000000000000"
// result:      "f000aa00-0451-4000-b000-000000000000"
static BLEUUID uuid2("0x1800");
// expected: "00001800-0000-1000-8000-00805f9b34fb"
// result:      <NULL>
```
(compare: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEUUID.cpp#L373)

chegewara, 2018-08-27:
You can see what service UUID is advertised with nRF connect when you scan for devices and before you connect any.

Sure, you can check and connect device if device name you want. With this library you have just example how to use serviceUUID to connect to sensor/device but not all ble devices are advertising serviceUUID. There is many ways to find device you want to connect to. By name, by serviceUUID, by manufacturer data.

If `0000aa80-0000-1000-8000-00805f9b34fb` is not in official documentation that does not mean its some random number. If i would write that app i would use serviceUUID just like you want it.

EBZ-Krisemendt, 2018-08-27:
final result on that second issue (creating uuid):
`BLEUUID::BLEUUID(str::string _uuid)` does not accept same format as `BLEUUID::fromString(str::string _uuid)` does. my bad.

_but maybe something to include sometimes ;)_

closed 2018-08-27 by EBZ-Krisemendt

chegewara, 2018-08-27:
sorry i could not help you with that, but im glad you solved this issue.

EBZ-Krisemendt, 2018-08-27:
[last update]
according to bluetooth specification `0xaa80` stands for "incomplete list of 16bit service uuids". which means that not all available services are advertised. lesson learned.
and better don't check on this uuid before establishing connection as it is not unique.

chegewara, 2018-08-27:
`incomplete list of 16bit service uuids` means that device is advertising one or more 16-bit uuids but after connecting to device you will have access to even more services.

Because advertising packet length is limited to 31+31 bytes some device are using trick and advertising 16 bit uuid even if services are using 128 bit uuids.

chegewara, 2018-08-27:
thats fine, its your app and you do it how you like. i just tried to explain why 0xaa80 is advertised its not in documentation. maybe this will help yu with another sensor or maybe someone else will read it and find it useful.
Im not bluetooth expert, but i am good enough (i think) to help users with this library and thats what i am trying to do.