nkolban/esp32-snippets #634: How do we hide an Esp32 from other devices?

niyaziatilay, 2018-08-29:
Other devices are doing bluetooth scanning. How do we hide an Esp32 from other devices? How do we get our device not to appear?

Example; pCharacteristic = pServer->createCharacteristic( notAppearFlag ); vs...

Thank you...

chegewara, 2018-08-29:
Do not start advertising and connect from peer device with mac address. 
Read and try to use whitelisted device.
Having only one service with one characteristic and when peer device write valid value create more services/characteristics that will handle esp32 functionality.

niyaziatilay, 2018-08-29:
Thank you very much for your response and your time. I work with the Arduino IDE.

As you mentioned yesterday, I tried to ask your forum sites. I could not get a reply from Esp32 forum.

I described the issue in more detail here:

https://stackoverflow.com/questions/52068454/how-do-we-hide-communication-an-esp32-from-other-devices?noredirect=1#comment91115470_52068454

I) How do I connect directly with mac address without advertising?
II) Which function do I need to use to create a white list?

Perhaps there are examples for the espressive IDF for Eclipse. No one suggested a solution for the Arduino IDE.

I guess there is no example in the Arduino IDE. Or I did not notice. Should I set the white list via mac addresses? With the help of Bluetooth characteristic functions?

Is there a small code sample for direct connection without advertising and for whitelist configuration?

I am a teacher and I am doing this project for my school's education. Do not look at it because my English and my technical knowledge are not good. I work on voluntary basis.

Thank you for everything.

niyaziatilay, 2018-08-29:
Thank you very much for your response and your time. I work with the Arduino IDE.

As you mentioned yesterday, I tried to ask your forum sites. I could not get a reply from Esp32 forum.

I am a teacher and I am doing this project for my school's education. Do not look at it because my English and my technical knowledge are not good. I work on voluntary basis.

I described the issue in more detail here:

https://stackoverflow.com/questions/52068454/how-do-we-hide-communication-an-esp32-from-other-devices?noredirect=1#comment91115470_52068454

I) How do I connect directly with mac address without advertising?
II) Which function do I need to use to create a white list?

Perhaps there are examples for the espressive IDF for Eclipse. No one suggested a solution for the Arduino IDE.

I guess there is no example in the Arduino IDE. Or I did not notice. Should I set the white list via mac addresses? With the help of Bluetooth characteristic functions?

Is there a small code sample for direct connection without advertising and for whitelist configuration?

Thank you for everything.

niyaziatilay, 2018-08-29:
Thank you very much for your response and your time. I work with the Arduino IDE.

As you mentioned yesterday, I tried to ask your forum sites. I could not get a reply from Esp32 forum.

I am a teacher and I am doing this project for my school's education. Do not look at it because my English and my technical knowledge are not good. I work on voluntary basis.

I described the issue in more detail here:

https://stackoverflow.com/questions/52068454/how-do-we-hide-communication-an-esp32-from-other-devices

I) How do I connect directly with mac address without advertising?
II) Which function do I need to use to create a white list?

Perhaps there are examples for the espressive IDF for Eclipse. No one suggested a solution for the Arduino IDE.

I guess there is no example in the Arduino IDE. Or I did not notice. Should I set the white list via mac addresses? With the help of Bluetooth characteristic functions?

Is there a small code sample for direct connection without advertising and for whitelist configuration?

Thank you for everything.

chegewara, 2018-08-29:
Since its not library issue i want to invite you here https://github.com/chegewara/esp32-snippets/issues to not spam this library.