nkolban/esp32-snippets #637: Only able to obtain BLERemoteCharacteristics by UUID

BigJBehr, 2018-08-30:
The current implementation of the Arduino IDE BLE library only allows for retrieving BLERemoteCharacteristic objects by UUID. The BLERemoteService class contains a map that is intended to be keyed by the handle of the BLERemoteCharacteristic. However, the code to fill the map and destroy the map is not implemented. There is also a method in the .h file for retrieving a BLERemoteCharacteristic object by passing a uint16_t. I have a need to work with a BLE server that offers multiple services and one of those services has multiple characteristics with the same UUID but different handles. The current implementation of BLERemoteService keeps only the first BLERemoteCharacteristic object and drops all thers with the same UUID. This is due to the map being keyed by UUID and that maps do not support duplicate keys. I have modified the BLERemoteService class to implement the mapByHandle and the associated methods to fill the map, destroy the map and access the map. I have attached the modified files in the hope that the next release of the library will fully support the mapByHandle.

[Modified-BLERemoteService.zip](https://github.com/nkolban/esp32-snippets/files/2338244/Modified-BLERemoteService.zip)

chegewara, 2018-08-31:
Hi, thanks for reporting this issue. You can create PR or wait when we will fix it.

chegewara, 2018-11-15:
I already have it implemented in my repo, but for some reason i missed it and is not in this library yet.
https://github.com/chegewara/esp32-snippets-enchancements-test/blob/master/components/cpp_utils/BLERemoteService.cpp#L235-L245

Im only wondering if this is good solution, or maybe i should change it to find and return one characteristic or nullptr instead of whole characteristic map.

closed 2019-01-30 by chegewara