nkolban/esp32-snippets #641: multiple servers support

mangooose, 2018-09-04:
Is it possible to connect to multiple servers? if yes what's the maximum number of connections?

mangooose, 2018-09-04:
Any other alternatives that works with arduino?

chegewara, 2018-09-04:
Nothing i know, but you can use esp-idf commands.

mangooose, 2018-09-04:
How? should I just include the esp-idf headers?

chegewara, 2018-09-04:
There is SimplBLE example, start with it. im guessing esp_bt.h header will be usable at beggining.

I will try to find some time and add multi-connection support to this library, but its quite a lot code to write i think (server and client part, notify/indicate etc).

mangooose, 2018-09-04:
Thanks a lot :)

mangooose, 2018-09-04:
What about this [pull request](https://github.com/nkolban/esp32-snippets/pull/541)?(the multiple client support part)

chegewara, 2018-09-04:
Like i said, there is a lot things to consider and some code needs to be re-thinked and changed. This is just example:
https://github.com/nkolban/esp32-snippets/pull/541/files#diff-6b48b04d9f9202e7dc8114b588e968baR98

Now on each connection to server service table is cleared, but it wont work with multi-connecting. This needs to be moved into disconnect part. This is not considered in that pull request and many more very important things. I have to review whole code again and make some plan first.

chegewara, 2018-09-05:
@mangooose I have some code ready, it was much simpler than i thought, but of course it requires some more tests in real life before can be used (i tested it with 2 esp32 servers and 1 esp32 client). Here is log:
```
D (28776) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (29089) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
D (29090) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
D (29091) BLEUtils: [conn_id: 1, remote_bda: 30:ae:a4:3f:24:4e, handle: 42 0x2a, value_len: 1, is_notify: 1]
D (29101) BLERemoteCharacteristic: Invoking callback for notification on characteristic Characteristic: uuid: 00002a2b-0000-1000-8000-00805f9b34fb, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 1, auth: 0
D (29123) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (29674) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
D (29675) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
D (29676) BLEUtils: [conn_id: 1, remote_bda: 30:ae:a4:3f:24:4e, handle: 42 0x2a, value_len: 1, is_notify: 1]
D (29686) BLERemoteCharacteristic: Invoking callback for notification on characteristic Characteristic: uuid: 00002a2b-0000-1000-8000-00805f9b34fb, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 1, auth: 0
D (29708) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (29822) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_NOTIFY_EVT
D (29823) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
D (29824) BLEUtils: [conn_id: 0, remote_bda: 30:ae:a4:43:a1:72, handle: 42 0x2a, value_len: 1, is_notify: 1]
D (29834) BLERemoteCharacteristic: Invoking callback for notification on characteristic Characteristic: uuid: 00002a2b-0000-1000-8000-00805f9b34fb, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 1, auth: 0
D (29856) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (30259) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
D (30260) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
D (30261) BLEUtils: [conn_id: 1, remote_bda: 30:ae:a4:3f:24:4e, handle: 42 0x2a, value_len: 1, is_notify: 1]
D (30271) BLERemoteCharacteristic: Invoking callback for notification on characteristic Characteristic: uuid: 00002a2b-0000-1000-8000-00805f9b34fb, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 1, auth: 0
D (30293) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (30844) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
D (30845) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
D (30846) BLEUtils: [conn_id: 1, remote_bda: 30:ae:a4:3f:24:4e, handle: 42 0x2a, value_len: 1, is_notify: 1]
D (30856) BLERemoteCharacteristic: Invoking callback for notification on characteristic Characteristic: uuid: 00002a2b-0000-1000-8000-00805f9b34fb, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 1, auth: 0
D (30878) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (30917) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_NOTIFY_EVT
D (30918) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
D (30919) BLEUtils: [conn_id: 0, remote_bda: 30:ae:a4:43:a1:72, handle: 42 0x2a, value_len: 1, is_notify: 1]
D (30929) BLERemoteCharacteristic: Invoking callback for notification on characteristic Characteristic: uuid: 00002a2b-0000-1000-8000-00805f9b34fb, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 1, auth: 0
D (30951) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (31429) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
D (31430) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
D (31431) BLEUtils: [conn_id: 1, remote_bda: 30:ae:a4:3f:24:4e, handle: 42 0x2a, value_len: 1, is_notify: 1]
D (31441) BLERemoteCharacteristic: Invoking callback for notification on characteristic Characteristic: uuid: 00002a2b-0000-1000-8000-00805f9b34fb, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 1, auth: 0
D (31463) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (31997) BLEDevice: gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_NOTIFY_EVT
D (31998) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
D (31999) BLEUtils: [conn_id: 0, remote_bda: 30:ae:a4:43:a1:72, handle: 42 0x2a, value_len: 1, is_notify: 1]
D (32009) BLERemoteCharacteristic: Invoking callback for notification on characteristic Characteristic: uuid: 00002a2b-0000-1000-8000-00805f9b34fb, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 1, auth: 0
D (32031) SampleClient_Notify: Notify callback for characteristic 00002a2b-0000-1000-8000-00805f9b34fb of data length 1
D (32042) BLEDevice: gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_NOTIFY_EVT
D (32050) BLEUtils: GATT Event: ESP_GATTC_NOTIFY_EVT
[... cut ...]
 * Perform the work of a sample BLE client.
 */
void SampleClient_Notify(void) {
	ESP_LOGD(LOG_TAG, "SampleClient_Notify starting");
	BLEDevice::init("esp32");
	pBLEScan = BLEDevice::getScan();
	pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
	pBLEScan->setActiveScan(true);
	pBLEScan->start(3);
} // SampleClient
```

I will push it to my github now and if will pass tests i will create PR, if you want to use it with arduino-ide then you need to prepare zip file or copy/paste modified files.

mangooose, 2018-09-05:
@chegewara  Thanks for your quick reply :) I will try it today and give you feedback

chegewara, 2018-09-05:
But i have to warn you its not finished yet, but i have no tools/examples to perform full tests and see what is missing.

mangooose, 2018-09-07:
@chegewara I just updated the files and made a quick sketch to test it and i'm still not able to connect to multiple servers. Here's my code:
```
    pClient1 = BLEDevice::createClient(0);
    pClient1->connect(*pServerAddress[0]);
    Serial.println("1 connected");
    BLERemoteCharacteristic* pRemoteCharacteristic1 =  pClient1->getService(ServiceUUID)->getCharacteristic(charUUID);

    pClient2 = BLEDevice::createClient(1);
    pClient2->connect(*pServerAddress[1]);
    Serial.println("2 connected");
    BLERemoteCharacteristic* pRemoteCharacteristic2 =  pClient2->getService(ServiceUUID)->getCharacteristic(charUUID);
    delay(1000);

    pRemoteCharacteristic1->writeValue((uint8_t*)x, 7, true);
    Serial.println("color1 changed");
    pRemoteCharacteristic2->writeValue((uint8_t*)x, 7, true);
    Serial.println("color2 changed");
```

last thing it shows is "2 connected"

chegewara, 2018-09-07:
@mangooose This looks like code stuck somewhere on semaphore. 

You can try and change this line:
`if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(BLEUUID((uint16_t)0x1805))) {`
with:
`if (strcmp(advertisedDevice.getAddress().toString(), "aa:aa:aa:aa:aa:aa")) {`
or something like that.

mangooose, 2018-09-07:
Currently I don't have enough ESP32 to test with your example. On my example, I search for new BLE devices this a specific serviceUUID and then I try to connect with these devices. The first device can connect and can also change characteristics value. But when I connect to more than one device it can connect but hangs when I try to change the RemoteCharacteristic value.

chegewara, 2018-09-07:
Thats ok, you can change service and char uuids to match your devices.

mangooose, 2018-09-07:
unfortunately still doesn't work :(

chegewara, 2018-09-07:
Im not sure it has to be strcmp or strncmp and `aa:aa:aa:aa:aa:aa` is `pServerAddress[0].toString()`. 
Maybe there is some glitch in this library that your code found but my example is not finding it or maybe its some glitch with your code.

nsamala, 2018-09-11:
I added disconnect support on my PR. It uses the existing `disconnect` function which unregisters the application from `GATTC`. It then goes ahead and removes the entry from the map.

mangooose, 2018-10-04:
@chegewara any reason why you removed multiple servers support?

chegewara, 2018-10-04:
@mangooose just by accident, i have this on my list ToDo anyway, so it should be added shortly

chegewara, 2018-10-05:
@mangooose I have new, better support for multiple client to one esp32 server if you are interested.

chegewara, 2018-10-14:
It is on github already, but after 2 days i can say its more like sandbox or testing library. When i tested it 2 days ago it was working really nice, but since then i feel like amator. There is few bugs and few issues i cant fix.

If you want to give it a try and maybe tell me what do you think about it then you are welcome:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

Olaclemmy, 2018-10-14:
Thanks for your kind gesture @chegewara, you're really hardworking person. Thanks once again for your time and efforts to keep this forum going. 

I will check it out and give it a try this coming weekend. 
I will also let you know my outcome as regards my testing.

chegewara, 2018-10-14:
I would like to say that we are ready for multi connectivity beta tests:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

Donderda, 2018-10-17:
@chegewara : I'm trying to use your test. Which version of esp-idf did you use? When I try to compile my project, I get this compile-errors:
```
CXX build/cpp_utils/BLEClient.o
/Users/user/esp/project/components/cpp_utils/BLEClient.cpp: In member function 'void BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
/Users/user/esp/project/components/cpp_utils/BLEClient.cpp:227:27: error: 'struct esp_ble_gattc_cb_param_t::gattc_search_cmpl_evt_param' has no member named 'searched_service_source'
    if(p_data->search_cmpl.searched_service_source == ESP_GATT_SERVICE_FROM_REMOTE_DEVICE) {
                           ^
/Users/user/esp/project/components/cpp_utils/BLEClient.cpp:227:54: error: 'ESP_GATT_SERVICE_FROM_REMOTE_DEVICE' was not declared in this scope
    if(p_data->search_cmpl.searched_service_source == ESP_GATT_SERVICE_FROM_REMOTE_DEVICE) {
                                                      ^
/Users/user/esp/project/components/cpp_utils/BLEClient.cpp:229:35: error: 'struct esp_ble_gattc_cb_param_t::gattc_search_cmpl_evt_param' has no member named 'searched_service_source'
    } else if (p_data->search_cmpl.searched_service_source == ESP_GATT_SERVICE_FROM_NVS_FLASH) {
                                   ^
/Users/user/esp/project/components/cpp_utils/BLEClient.cpp:229:62: error: 'ESP_GATT_SERVICE_FROM_NVS_FLASH' was not declared in this scope
    } else if (p_data->search_cmpl.searched_service_source == ESP_GATT_SERVICE_FROM_NVS_FLASH) {
                                                              ^
/Users/user/esp/project/components/cpp_utils/BLEClient.cpp: At global scope:
/Users/user/esp/project/components/cpp_utils/BLEClient.cpp:46:20: warning: 'LOG_TAG' defined but not used [-Wunused-variable]
 static const char* LOG_TAG = "BLEClient";
                    ^
make[1]: *** [BLEClient.o] Error 1
make: *** [component-cpp_utils-build] Error 2
```

Did I do something wrong?

chegewara, 2018-10-17:
I am always on master branch, you did nothing wrong. There is few flags to work with bluetooth data in nvs cache and you are missing those flags. You can find it in BLEClient.cpp and just comment out, its only logging.
`ESP_GATT_SERVICE_FROM_REMOTE_DEVICE`
`ESP_GATT_SERVICE_FROM_NVS_FLASH`

Donderda, 2018-10-17:
Thank you sir! I'll try now and give you feedback 👍

chegewara, 2018-11-29:
Implemented in latest version.

epanta, 2019-02-17:
@chegewara Is your code (https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test) still available?

chegewara, 2019-02-17:
Sorry, code has been merged to this library and is no longer available. Since then some code has been changed too, many bugfixes that still were in my repository.

epanta, 2019-02-17:
@chegewara But is possible connect 2 servers with a client with this library? If not.. could you share again your solution?

chegewara, 2019-02-17:
As far as i know it should be possible, if you have problems let me know.

epanta, 2019-03-31:
@chegewara It worked.. but not always.. I think that I did something wrong.

schmidtb89, 2019-04-25:
@epanta , @chegewara  excuse me how do you get the code  (https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test) ?

chegewara, 2019-04-25:
hi @schmidtb89 
this repository is no longer available, but it was merged to this master