nkolban/esp32-snippets #642: issues using cpp_utils -

wegunterjr, 2018-09-07:
I am trying to use the cpp_utils library in my platformio environment (VSCode), but having some problems.
Probably user error, but figured I would ask:
https://community.platformio.org/t/vscode-cannot-include-private-library-intellisense-not-working-and-build-fails/4997

Now, I have to add cpp_utils to every include statement in every file in the cpp_utils folder ....  seems like there is anotther way but right now, still working through that.

chegewara, 2018-09-07:
Im not working with platformio but i am using VSCode with esp-idf and most of files are not recognised by intellisense. Maybe its my fault and i dont know how to configure it, but i am using msys cli to build binary file so its not big issue to me (i dont have to include any cpp_utils files except those i am using).

wegunterjr, 2018-09-07:
Ah... Intellisense does work for me...and when I added the cpp_utils
library, it totally finds those files too.

Just wondering if I am doing the includes wrong.

chegewara, 2018-09-07:
There was few questions about platformio in here, maybe some answer will help you.
https://github.com/nkolban/esp32-snippets/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+platformio

wegunterjr, 2018-09-10:
alright, getting it to recognize the cpp_utils but had to add the files I am using to another folder
cpp_utils->WiFi (just a name, doesn't have to be this) ->src -> files.
They show up in intellisense.   there is a squiggly line under the include file, but it will navigate to it and compiles just fine.

`lib_extra_dirs = lib/cpp_utils`
and it recognizes any subfolder in there, but not the individual files.

closed 2018-11-13 by chegewara