nkolban/esp32-snippets #644: Stopping WiFi seems to inhibit all future BT scans in Arduino.

michaelhwhitten, 2018-09-11:
I'm testing a BT to web gateway than scans for BT advertising data and posts it to useful places using WiFi. I'd like to test BT scanning performance when WiFi is on an off; this to seek performance tradeoffs. Thing is, once I've connected to WiFi and then attempted to stop WiFi, every subsequent BT scan returns zero objects. That doesn't seem right. Or is it? Am I doing something wrong? Code below; comments and input welcome!

`#include <esp_system.h>
#include <WiFi.h>
#include <WiFiMulti.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

WiFiMulti wifiMulti;

//sets BT scan time.
#define SCAN_TIME  5 // seconds

//This will be the BLE scan instance
BLEScan* pBLEScan;

void WiFiEvent(WiFiEvent_t event){
  switch(event) {
    case SYSTEM_EVENT_STA_GOT_IP:
      Serial.print("WiFi event: SYSTEM_EVENT_STA_GOT_IP, IP address: ");
      Serial.println(WiFi.localIP());  
      break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
      Serial.println("WiFi event: SYSTEM_EVENT_STA_DISCONNECTED");
      break;
    case SYSTEM_EVENT_STA_START:
      Serial.println("WiFi event: SYSTEM_EVENT_STA_START");
      break;
    case SYSTEM_EVENT_STA_CONNECTED:
      Serial.println("WiFi event: SYSTEM_EVENT_STA_CONNECTED");
      break;
    case SYSTEM_EVENT_WIFI_READY:
      Serial.println("WiFi event: SYSTEM_EVENT_WIFI_READY");
      break;
    case SYSTEM_EVENT_AP_START:
      Serial.println("WiFi event: SYSTEM_EVENT_AP_START");
      break;
    case SYSTEM_EVENT_SCAN_DONE:
      Serial.println("WiFi event: SYSTEM_EVENT_SCAN_DONE");
      break;
    case SYSTEM_EVENT_STA_STOP:
      Serial.println("WiFi event: SYSTEM_EVENT_STA_STOP");
      break;
    default:
      Serial.print("WiFi event, unhandled: ");
      Serial.println(event);
      break;
  }
}

void setup() {
  Serial.begin(115200);
  
  //Set up WiFi event handling and feedback
  WiFi.onEvent(WiFiEvent);

  //add networks
  wifiMulti.addAP("SomeSSID","SomePw");
  wifiMulti.addAP("AnotherSSID","anotherPw");

  //prep for bluetooth scanning
  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(300); // msec between start of BT scans
  pBLEScan->setWindow(300); // msec scan time - must be <= interval
}

void loop() {
    int deviceCount;
    int i;

    Serial.println("Starting BT scan");
    BLEScanResults foundDevices = pBLEScan->start(SCAN_TIME);
    deviceCount = foundDevices.getCount();
    Serial.print("Finished BT scan. Found devices: ");
    Serial.println(deviceCount);

    //connect to WiFi in order to post something about the bt scan just completed
    i=0;
    while (wifiMulti.run() != WL_CONNECTED) {
      if (i = 4) return; //if we can't connect, try posting next time
      i++;
      delay(500);
    }

    //this is where we would post data
    //postSomeData();

    //This code should turn off wifi to make bt scanning more reliable. We have observed better BT performance when
    //wifi was never initiated. I've tried all the stopping methods below in attempts to work around the issue.
    //It does turn off wifi - but every bt scan from that point forward returns 0 objects.
    //Why? It's an esp32 mystery to me right now.
    WiFi.mode(WIFI_MODE_NULL);
//    WiFi.disconnect(true);
//    WiFi.mode(WIFI_OFF);
    delay(1000);
    
//    btStop(); //Tried this. Doesn't help.
//    delay(1000);
//    btStart(); //Doesn't matter if used with or without btStop; doesn't help.
//    delay(1000);
}`

michaelhwhitten, 2018-09-11:
Thanks for the reply. Yes, I’ve tried and tested various scan parameters so am aware that’s possible. It’s how my gateway is running right now in fact.

In this case though I really do want to completely shut down wifi while BT scanning, turning it on only to post. The intent is to test and characterize performance tradeoffs.

It seems that should be possible yet BT scans never work again in my testing. That seems like a bug, so, I’m hoping for community insight.

chegewara, 2018-09-11:
I have arduino app with ble + wifi (aws mqtt). Its not easy because i am using secure connection and this is hard battle with heap shortage, but my app is running.
What it does is: connect to aws IoT and start polling for subscribed messages then scan for ble POS printer and connect. Ive done few tests and after disconnecting ble printer my app is starting scanning and reconnect. So in my case ble + wifi works fine.

Maybe my app structure helps with that. I have separate tasks for wifi and ble, and im doing short ble scans.

michaelhwhitten, 2018-09-12:
I can appreciate the struggle. It took some tuning, keeping up with recent library update, etc but we also have them running together. I know that can be done. I'd just like to benchmark overall performance when only running wifi while posting data.

That code I posted should compile readily. If you'd be willing, I'd appreciate it if you'd run it once and see what happens. On mine, the first BT scan works and returns the expected number of objects; all the code does is print out the number found during the scan. But once wifi is connected then WiFi.mode(WIFI_MODE_NULL) is called, BT scans forever after (until resetting the chip of course) return zero objects. It never works again.

Thanks in advance if you have time for the experiment.

chegewara, 2018-09-13:
I did some tests with your code and it seems to work. I dont know if you need this line or not, but i commented it out:
`WiFi.mode(WIFI_MODE_NULL);`
Now most important change in this code i did is:
```
pBLEScan->setInterval(500); // msec between start of BT scans
pBLEScan->setWindow(3000); // msec scan time - must be <= interval
```
Now question why? Because i am using android smarthone with nRF connect for testing and advertising interval is about 250-300ms. Now, to not miss any advertising packets i have setup to 500ms. Also windows time is different than interval to give some radio time to wifi.
To be honest i dont know which value should be greater, because for me both options works:
`setInterval > setWindow` and `setInterval < SetWindow`.

michaelhwhitten, 2018-09-13:
Ah, thank you! That last bit is what I was hoping someone could replicate. I appreciate you taking the time to test that.

I thought this was the right place; my mistake. I'll post over there. Thanks again!

chegewara, 2018-09-13:
Well i did replicate it, just i am not programmer (i just pretend to be) and wifi library is beyond my knowledge.

michaelhwhitten, 2018-09-13:
Mine too! And I once was a programmer.