nkolban/esp32-snippets #645: Issues compiling BLEServer sample test -

wegunterjr, 2018-09-12:
I am getting an error:

`error: expected class-name before '{' token`

```
 * Create a new BLE server.
 */
#include "BLEDevice.h"
#include "BLEServer.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include <esp_log.h>
#include <string>
//#include <../../freertos/include/freertos/Task.h>

#include "sdkconfig.h"

static char LOG_TAG[] = "SampleServer";

class MainBLEServer: public Task {
```

chegewara, 2018-09-12:
class Task is not `../../freertos/include/freertos/Task.h`, it is https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/Task.cpp

wegunterjr, 2018-09-12:
bah!   I was totally looking at that wrong.   Thanks for the correction.

hansmbakker, 2018-12-06:
@wegunterjr it looks like your issue is solved. If it is, could you please close it to keep the issue list clean?
If you want to share some useful hints with others, it might be an idea to write something on the [wiki](https://github.com/nkolban/esp32-snippets/wiki) instead.

cc @chegewara