nkolban/esp32-snippets #646: vtable for BLESecurityCallbacks - Including workaround

hetlelid, 2018-09-13:
Using example: SampleClient_authentication_numeric_confirmation.cpp

With: ESP-IDF and arduino-esp32

Creating exeption: 
`/home/osboxes/esp/test/m5_windsensor/build/main/libmain.a(main.o):(.literal._ZN10MySecurityD5Ev[MySecurity::~MySecurity()]+0x0): undefined reference to `vtable for BLESecurityCallbacks'
collect2: error: ld returned 1 exit status
/home/osboxes/esp/esp-idf/make/project.mk:410: recipe for target '/home/osboxes/esp/test/m5_windsensor/build/app-template.elf' failed
make: *** [/home/osboxes/esp/test/m5_windsensor/build/app-template.elf] Error 1`

Also, there is an issue with the Task.h, that is not included in arduino-esp32

My workaround:
- include cpp_utils in components
- delete arduino BLE from arduino-esp32

git clone https://github.com/nkolban/esp32-snippets.git
cp -r esp32-snippets/cpp_utils . && rm -rf esp32-snippets
rm -rf arduino-esp32/libraries/BLE/

chegewara, 2018-09-13:
Sorry, i cant reproduce this issue with esp-idf.
Could you check if you have turned on SMP in bluetooth settings in menuconfig? Also when did you update esp-idf and cpp_utils last time?

hetlelid, 2018-09-13:
SMP is enabled.
When i get this error i'm not using cpp_utils directly, rather the arduino-esp32 with BLE. But fixed it by using cpp_utils.

I'm using the IDF project template from: https://github.com/wtakuo/M5Stack-IDF
Maybe i should make the IDF project from scratch.

What is your recommendations how to create a new esp-idf project?