nkolban/esp32-snippets #649: HID Gamepad constant disconnects.

GameDragon2k, 2018-09-16:
I am using my ESP32 as a HID Gamepad and I am running into issues with it.

Connection seems fine at the beginning but after several perfectly working inputs, I get a hard lag spike followed by disconnect. The lag itself is a big issue, but after reconnecting, the ESP32 will continue to disconnect itself every 90 seconds or so, regardless if I send any inputs or not.

Here is my code: https://gist.github.com/GameDragon2k/90709fbe6b15e66c1b274dde3e1b2439

And this is debug for disconnects/reconnects:
```
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_DISCONNECT_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLEUtils.cpp:1723] dumpGattServerEvent(): [conn_id: 0, remote_bda: 5c:f3:70:74:7d:d3]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_DISCONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:342] startAdvertising(): >> startAdvertising
[D][BLEAdvertising.cpp:174] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLEAdvertising.cpp:187] start(): - advertising service: 00001812-0000-1000-8000-00805f9b34fb
[D][BLEAdvertising.cpp:231] start(): << start
[D][BLEServer.cpp:344] startAdvertising(): << startAdvertising
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEServer.cpp:342] startAdvertising(): >> startAdvertising
[D][BLEAdvertising.cpp:174] start(): >> start: customAdvData: 0, customScanResponseData: 0
[D][BLEAdvertising.cpp:187] start(): - advertising service: 00001812-0000-1000-8000-00805f9b34fb
[D][BLEAdvertising.cpp:231] start(): << start
[D][BLEServer.cpp:344] startAdvertising(): << startAdvertising
start advertising
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1099] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:135] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLEUtils.cpp:1707] dumpGattServerEvent(): [conn_id: 0, remote_bda: 5c:f3:70:74:7d:d3]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONNECT_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
Connection established!
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_MTU_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_MTU_EVT
[D][BLEUtils.cpp:1766] dumpGattServerEvent(): [conn_id: 0, mtu: 517]
[I][BLEDevice.cpp:113] gattServerEventHandler(): ESP_GATTS_MTU_EVT, MTU 517
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_MTU_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
```

chegewara, 2018-09-16:
I will check it, but my first thought is this:
```
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());
```
If i remember you dont have (and you should not) create server, it is created by BLEHID class.

GameDragon2k, 2018-09-16:
@chegewara I'm sorry, I don't understand the question. Are you asking where in the code the GPIO pins are mapped?

If so, that would be here: https://gist.github.com/GameDragon2k/90709fbe6b15e66c1b274dde3e1b2439#file-esp32_ble_arcade-ino-L153-L187

I should also mention that I have been testing this issue using only a single pushbutton connected to one GPIO pin.

chegewara, 2018-09-16:
> I should also mention that I have been testing this issue using only a single pushbutton connected to one GPIO pin.
Thats important information. I was asking about pin mapping because:
- some pins (35-39) soes not have internal pullup/pulldown option,
- some pins are reserved for communication with flash (pins 28-33 or GPIO6-GPIO11)
https://raw.githubusercontent.com/gojimmypi/ESP32/master/images/myESP32%20DevKitC%20pinout.png
I know its not the same board, but its almost 1:1 clone.
I dont remember how are those pins mapped in arduino, thats my question.

GameDragon2k, 2018-09-16:
Arduino uses GPIO number for mapping, so `pinMode(11, INPUT_PULLUP);` would be mapped to FLASH CMD.

I have tested GPIOs 34-39 using external resistors and didn't run into any issues regarding inputs. GPIO9 and GPIO10 can be removed from my mapping, but using them individually in testing I noticed no difference with the issue.

chegewara, 2018-09-16:
What is testing platform? android, windows or linux? if windows what version is it?

GameDragon2k, 2018-09-16:
Tested on three devices, all Windows 10 Pro x64.

chegewara, 2018-09-16:
@GameDragon2k I have tested your code with my devKitC on windows 10 pro and i have strange behavior.
At some point putty is displaying some crap and i am guessing it is the moment when your device is starting disconnecting. I have to do more tests with other esp32 board.

GameDragon2k, 2018-09-16:
Thank you for testing. I am at a lost myself, so I hope you find some solid results.

chegewara, 2018-09-16:
@GameDragon2k I have some issue with my windows atm and i need to reboot it, but unfortunately i cant. Could you try to change debug level to info or less and try again?

PS i meant windows version (windows 8 or 10) but you already did answer it

chegewara, 2018-09-16:
@GameDragon2k I had to install windows latest update and reboot laptop. Now your code seems to work with no issues, no disconnection, for about 10 minutes. I didnt run putty this time, but windows game controller properties shows me that esp32 works fine. 

PS you have to find one workaround because when you reconnecting with windows or other peer device it does not register for notification and esp32 will not send notifications.

GameDragon2k, 2018-09-16:
> 
> PS you have to find one workaround because when you reconnecting with windows or other peer device it does not register for notification and esp32 will not send notifications.

You're referring to your workaround here? https://github.com/nkolban/esp32-snippets/issues/632#issuecomment-415982078

Thank you, I have included that bit as well. I will attempt to update and test again.

chegewara, 2018-09-16:
You could try to prepare test case. Just simplify code, or use the one from #632 and send button on/off with some small delay between. 

I cant explain that but there is many issues reported that i cant recreate because users code works for me, in other hand i am reporting some ble issues that cant be recreated by espressif devs.

GameDragon2k, 2018-09-16:
I went ahead and tried using the code from https://github.com/nkolban/esp32-snippets/issues/632. 
I modified it to send inputs through GPIO33 with a delay of 20ms after each notification. Besides this, it is unchanged.

Unfortunately, within a few minutes of pressing the pushbutton, I get the same lag problem followed by the same disconnect.

chegewara, 2018-09-16:
Then i really dont know. Maybe its esp32 faulty, also when did you last time update espressif/arduino-esp32 library?

GameDragon2k, 2018-09-17:
I tried https://github.com/nkolban/esp32-snippets/issues/632 again but this time I left it completely intact. No code modifications, no pushbutton, no pin mapping.

I left the thing running and with 2000ms delay, notifications were being sent correctly for about an hour until the disconnection loop started.

Could there be some kind of garbage collection or heap issue?

chegewara, 2018-09-17:
To be honest i never been doing such long tests. You can add some memory debug command before or after notify() and one in onDisconnect callback.

chegewara, 2018-09-17:
I am using this:
`    ESP_LOGW(TAG, "%d: - RAM left %d", __LINE__, esp_get_free_heap_size());`

GameDragon2k, 2018-09-17:
Okay, well that idea is out the window. Ram seems just fine.

GameDragon2k, 2018-09-20:
Unfortunately no luck. Purchased a new Sparkfun ESP32 Thing and getting the exact same issue.

chegewara, 2018-09-21:
@GameDragon2k I really dont know what to say, it works for me and few other people.
You could try to uninstall and make fresh install of arduino. Last one test will be if you install binary file i have compiled and tested on my esp32.

This bin is from you code (copy/paste) and if i remember its working for me:
[gamedragon2k_hid.bin.txt](https://github.com/nkolban/esp32-snippets/files/2406621/gamedragon2k_hid.bin.txt)

You have to change extension and upload it with esp32 flash download tool.

senthilkumarh, 2019-03-03:
[BLE_Mouse.ino.txt](https://github.com/nkolban/esp32-snippets/files/2923235/BLE_Mouse.ino.txt)
@ chegewara the attached arduino sketch for ESP32 BT HID mouse is working in android , but not working in windows 10. Please can you help