nkolban/esp32-snippets #651: Does "BLEScan::start()" has issue of memory leak?

kymybk, 2018-09-21:
Hi.

We'd like to use "BLEScan::start()" for receive "Advertising Data" including sensor value permanently.
We've tried use it by below program, but ESP32 had restarted by lack HeapSize.

```
void setup() {
 Serial.begin(115200);
 BLEDevice::init("");
 pBLEScan = BLEDevice::getScan(); //create new scan
 pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
 pBLEScan->setActiveScan(true);
}

void loop() {
 Serial.print("Free HeapSize: ");
 Serial.println(esp_get_free_heap_size());
 pBLEScan->start(3);
 delay(1000);
}
```

Shouldn't "BLEScan::start()" called from "loop()" function?
Or does "BLEScan::start()" has issue of memory leak?

Please could you confirm this question?

chegewara, 2018-09-21:
And what is this line output:
`Serial.println(esp_get_free_heap_size());`

kymybk, 2018-09-25:
`esp_get_free_heap_size()` is shown free heap size.

The output values are as belows.

```
34688
34356
34024
33504
32868
32276
32240
31516
30688
30024
29508
28680
28132
27512
26912
26292
25832
.
.
.
5284
4432
3776
3104
2528
```

chegewara, 2018-10-03:
Try after each scan->start()  to call scan->stop().

kymybk, 2018-10-10:
Thanks.
I tried it.

```
class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
    void onResult(BLEAdvertisedDevice advertisedDevice)
    {
        .
        .
        .
        pBLEScan->stop();
    }
}
```

But the result does not seem to change.

```
Free HeapSize: 38136
Free HeapSize: 31844
Free HeapSize: 31796
Free HeapSize: 31964
Free HeapSize: 31456
Free HeapSize: 31100
Free HeapSize: 30996
Free HeapSize: 30220
Free HeapSize: 29800
Free HeapSize: 29372
Free HeapSize: 28600
Free HeapSize: 27212
Free HeapSize: 27376
Free HeapSize: 27712
Free HeapSize: 27164
.
.
.
Free HeapSize: 8068
Free HeapSize: 7908
Free HeapSize: 7184
Free HeapSize: 6768
Free HeapSize: 6668
Free HeapSize: 6476
Free HeapSize: 6364
Free HeapSize: 5744
Free HeapSize: 5488
Free HeapSize: 5016
Free HeapSize: 4948
Free HeapSize: 4676
Free HeapSize: 4600
Free HeapSize: 4032
Free HeapSize: 3680
```

chegewara, 2018-10-10:
I know its quite old issue, sorry for delay, but i am working on few updates and i have this on todo list.
https://github.com/chegewara/esp32-snippets/issues/7

kymybk, 2018-10-10:
OK, Thanks.
I hope the update solves this issue.

chegewara, 2018-10-12:
Hi, 
could you test those changes in BLEScan class, please?
https://github.com/chegewara/esp32-snippets-enchancements-test/blob/multi_connect_test/components/cpp_utils/BLEScan.cpp

chegewara, 2018-10-13:
I am doing tests right now and it seems that leak memory is in esp-idf bt stack, not this library.

chegewara, 2018-10-13:
@kymybk 
Ok, i found that memory leak in BLEScan. I am working now on multi connect implementation and i have to change a little bit BLEScan class so it will be fixed in it. Thanks again for reporting and help with tests.

If you want to fix it just for now then it is here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEScan.cpp#L201

Object is deleted from vector but is not destroyed.

chegewara, 2018-10-13:
Before it will be merged to library you can get it from this gist:
https://gist.github.com/chegewara/6fc38e4127a7e18bb5b0bb4367aa33d1

kymybk, 2018-10-15:
@ chegewara Thank you for your solution.

I tried the essentials you provided.
The results are as below.
```
Free HeapSize: 38124
temperature: 25.78 
relativeHumidity: 47.24 
ambientLight: 316.00 
pressure: 1011.66 
soundNoise: 55.04 
eCO2: 1981.00 
eTVOC: 714.00 

Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
Free HeapSize: 37604
.
.
.

```
There seems to be fix to memory leak.

But, temperature values, humidity values, etc. are not continuously received by call back.
Although it is output for the first time, since there is no output after the second time, it seems that the callback is not working properly.

chegewara, 2018-10-15:
Because new parameter has been introduced in start():
https://gist.github.com/chegewara/6fc38e4127a7e18bb5b0bb4367aa33d1#file-blescan-h-L58

If you want to read from sensors then you need to pass false and discovered devices will be cleared on every start.

chegewara, 2018-10-15:
There is possible it is not BLEScan related. I will do more tests. Thanks.

kymybk, 2018-10-15:
Thanks for your good job.
Best Regards.

chegewara, 2018-10-15:
Hi @kymybk 
I forgot about deleting advertised device when is no longer required in case it is duplicate and is no stored. Now should be fine. I updated the same gist.
https://gist.github.com/chegewara/6fc38e4127a7e18bb5b0bb4367aa33d1

chegewara, 2018-10-16:
Start of scan:
```
I (3848) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (3851) SampleScan test: free heap ammount: 4301252
I (4236) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (4239) SampleScan test: free heap ammount: 4301076
I (4624) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (4628) SampleScan test: free heap ammount: 4301076
I (5011) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (5014) SampleScan test: free heap ammount: 4300968
I (5022) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (5034) SampleScan test: free heap ammount: 4301072
I (5402) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (5406) SampleScan test: free heap ammount: 4301076
I (5791) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (5794) SampleScan test: free heap ammount: 4301076
I (6179) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (6183) SampleScan test: free heap ammount: 4301076
I (6568) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (6572) SampleScan test: free heap ammount: 4301076
```
about an 2 hours later:
```
E (7678138) SampleScan test: start scan
I (7678154) SampleScan test: Advertised Device: Name: , Address: 7c:5e:d9:78:da:f4, txPower: -7
I (7678155) SampleScan test: free heap ammount: 4301188
I (7678160) SampleScan test: Advertised Device: Name: , Address: 4c:89:93:a1:57:99, txPower: -7
I (7678168) SampleScan test: free heap ammount: 4301072
I (7678177) SampleScan test: Advertised Device: Name: , Address: 42:e1:c1:a6:68:9d, txPower: -7
I (7678184) SampleScan test: free heap ammount: 4300524
I (7678192) SampleScan test: Advertised Device: Name: , Address: 69:da:50:94:1a:7d, txPower: -7
I (7678199) SampleScan test: free heap ammount: 4300416
I (7678206) SampleScan test: Advertised Device: Name: , Address: 65:cd:32:13:9f:26, txPower: -7
I (7678214) SampleScan test: free heap ammount: 4300304
I (7678222) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (7678234) SampleScan test: free heap ammount: 4300144
I (7678551) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (7678555) SampleScan test: free heap ammount: 4299860
I (7678565) SampleScan test: Advertised Device: Name: , Address: 5c:10:d4:5e:bc:c8, txPower: -7
I (7678570) SampleScan test: free heap ammount: 4299728
I (7678577) SampleScan test: Advertised Device: Name: , Address: 4c:89:93:a1:57:99, txPower: -7
I (7678585) SampleScan test: free heap ammount: 4299576
I (7678593) SampleScan test: Advertised Device: Name: , Address: 7c:5e:d9:78:da:f4, txPower: -7
I (7678600) SampleScan test: free heap ammount: 4299684
I (7678607) SampleScan test: Advertised Device: Name: , Address: 42:e1:c1:a6:68:9d, txPower: -7
I (7678616) SampleScan test: free heap ammount: 4299796
I (7678954) SampleScan test: Advertised Device: Name: , Address: 66:7b:3a:e9:c2:8b, txPower: -7
I (7678954) SampleScan test: free heap ammount: 4299616
I (7678960) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (7678973) SampleScan test: free heap ammount: 4299304
I (7678980) SampleScan test: Advertised Device: Name: , Address: 5c:10:d4:5e:bc:c8, txPower: -7
I (7678988) SampleScan test: free heap ammount: 4299460
I (7678995) SampleScan test: Advertised Device: Name: , Address: 4c:89:93:a1:57:99, txPower: -7
I (7679003) SampleScan test: free heap ammount: 4299568
I (7679361) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (7679365) SampleScan test: free heap ammount: 4299524
I (7679750) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (7679754) SampleScan test: free heap ammount: 4299524
I (7680100) SampleScan test: Advertised Device: Name: , Address: 42:e1:c1:a6:68:9d, txPower: -7
I (7680101) SampleScan test: free heap ammount: 4299460
I (7680107) SampleScan test: Advertised Device: Name: , Address: 69:da:50:94:1a:7d, txPower: -7
I (7680115) SampleScan test: free heap ammount: 4299572
I (7680139) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (7680143) SampleScan test: free heap ammount: 4299524
I (7680499) SampleScan test: Advertised Device: Name: , Address: 65:cd:32:13:9f:26, txPower: -7
I (7680500) SampleScan test: free heap ammount: 4299460
[... cut ...]
I (7681338) SampleScan test: free heap ammount: 4300356
I (7681672) SampleScan test: Advertised Device: Name: ESP32, Address: fc:22:33:8e:ac:ba, serviceUUID: 00001234-0000-1000-8000-00805f9b34fb, txPower: 3
I (7681676) SampleScan test: free heap ammount: 4300072
I (7681685) SampleScan test: Advertised Device: Name: , Address: 66:7b:3a:e9:c2:8b, txPower: -7
I (7681691) SampleScan test: free heap ammount: 4300120
I (7681698) SampleScan test: Advertised Device: Name: , Address: 4c:89:93:a1:57:99, txPower: -7
```

kymybk, 2018-10-17:
Your test is using FreeRTOS.

I do not use FreeRTOS in my test.
My test code is a general Arduino Sketch consisting of "void setup ()" and "void loop ()" functions.

chegewara, 2018-10-17:
All functions i am using are available for you. For some reason, i dont know why yet, when i dont create task in which i call scan->start i am having memory leak too.

kymybk, 2018-10-17:
That means that in this case,
Is it the best way to create a task using FreeRTOS and call "scan-> start"?

Besides using FreeRTOS, is there any way to solve this issue?

chegewara, 2018-10-17:
I dont know yet, i couldnt find the reason. I will back to this issue some time later.

chegewara, 2018-11-14:
fixed by e8b5df4

hansmbakker, 2018-12-06:
@kymybk it looks like your issue is fixed by a commit from @chegewara. If it is fixed indeed, could you please close this issue to keep the issue list clean?