nkolban/esp32-snippets #653: After make clean, none of esp-idf examples are compiling

BasavarajaB, 2018-09-23:
I have setup esp-idf environment on virtual box using link http://iot-bits.com/esp32/getting-started-with-esp32-esp-idf-part-2/.

I was able to build examples by executing make menuconfig followed by make command.
Once i executed make clean, after that none of examples are compiling.i tried my level best to resolve the issue by googling,even tried new esp/toolchain download and setup. but no success.
Make command error is : 
user@ubuntu1604:~/esp/esp-idf_new/examples/get-started/hello_world$ make
make[1]: Entering directory `/home/user/esp/esp-idf_new/tools/kconfig'
cc  -D_GNU_SOURCE -DCURSES_LOC="<ncurses.h>" -DLOCALE -MD   -c -o mconf.o mconf.c
In file included from mconf.c:22:0:
lkc.h:12:22: fatal error: libintl.h: No such file or directory
compilation terminated.
cc: /opt/qnx660/host/linux/x86/usr/lib/gcc/i486-pc-nto-qnx6.6.0/4.7.3/cc1 error 1
make[1]: *** [mconf.o] Error 1
make[1]: Leaving directory `/home/user/esp/esp-idf_new/tools/kconfig'
make: *** No rule to make target `/home/user/esp/esp-idf_new/tools/kconfig/conf', needed by `/home/user/esp/esp-idf_new/examples/get-started/hello_world/build/include/config/auto.conf'.  Stop.
user@ubuntu1604:~/esp/esp-idf_new/examples/get-started/hello_world$ 

Can you please help me to resolve the issue.

chegewara, 2018-09-23:
Hi,
It looks like you may have issue with PATH or with IDF_PATH. 

Try printenv command for both.

hansmbakker, 2018-12-06:
@BasavarajaB your issue is quite old and we haven't heard back from you. Could you please your issue to keep the issue list clean?

cc @chegewara

closed 2018-12-09 by chegewara