nkolban/esp32-snippets #654: BLERemoteCharacteristic.cpp: free() target pointer is outside heap areas

hetlelid, 2018-09-23:
Maybe i'm using it wrong, bu i get the error below.

My workaround is to just remove the free() call:

                                //if(m_rawData != nullptr)
                                //      free(m_rawData);
                                ESP_LOGE(LOG_TAG, "free() DISABLED");

I know this introduce potential memory leak. Any suggestion to further investigation?

Ref: https://github.com/nkolban/esp32-snippets/blob/2b7d59040f3f6081f3694fa722692daddaf9cbb2/cpp_utils/BLERemoteCharacteristic.cpp#L208

---

`D][BLERemoteCharacteristic.cpp:527] registerForNotify(): << registerForNotify()
A [D][BLERemoteCharacteristic.cpp:439] readValue(): >> readValue(): uuid: 00002a6e-0000-1000-8000-00805f9b34fb, handle: 25 0x19
D (6271) FreeRTOS: Semaphore taking: name: ReadCharEvt (0x3ffd5f90), owner: <N/A> for readValue
D (6281) FreeRTOS: Semaphore taken:  name: ReadCharEvt (0x3ffd5f90), owner: readValue
[D][BLEDevice.cpp:143] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 1] ... ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1527] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, handle: 25 0x19, value_len: 2]
D (6331) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
D (6341) GeneralUtils:      -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
D (6341) GeneralUtils: 0000 11 09                                            ..
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/osboxes/esp/esp-idf/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x400d41db on core 0
0x400d41db: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4008f970:0x3ffceee0 0x4008fb41:0x3ffcef00 0x400d41db:0x3ffcef20 0x4008290b:0x3ffcef50 0x40082d1d:0x3ffcef70 0x4000bec7:0x3ffcef90 0x40114152:0x3ffcefb0 0x40117340:0x3ffcf000 0x40113199:0x3ffcf020 0x401156ed:0x3ffcf0d0 0x40139ba5:0x3ffcf110 0x40136a4a:0x3ffcf150 0x4008cacd:0x3ffcf180
0x4008f970: invoke_abort at /home/osboxes/esp/esp-idf/components/esp32/panic.c:660

0x4008fb41: abort at /home/osboxes/esp/esp-idf/components/esp32/panic.c:660

0x400d41db: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x4008290b: heap_caps_free at /home/osboxes/esp/esp-idf/components/heap/heap_caps.c:130

0x40082d1d: _free_r at /home/osboxes/esp/esp-idf/components/newlib/syscalls.c:42

0x40114152: BLERemoteCharacteristic::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/osboxes/esp/test/m5_test/components/cpp_utils/BLERemoteCharacteristic.cpp:344

0x40117340: BLERemoteService::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/osboxes/esp/test/m5_test/components/cpp_utils/BLERemoteService.cpp:256

0x40113199: BLEClient::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/osboxes/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/ext/new_allocator.h:110

0x401156ed: BLEDevice::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at /home/osboxes/esp/test/m5_test/components/cpp_utils/BLEDevice.cpp:170

0x40139ba5: btc_gattc_cb_to_app at /home/osboxes/esp/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c:31
 (inlined by) btc_gattc_cb_handler at /home/osboxes/esp/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c:795

0x40136a4a: btc_task at /home/osboxes/esp/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:110

0x4008cacd: vPortTaskWrapper at /home/osboxes/esp/esp-idf/components/freertos/port.c:401
`

chegewara, 2018-09-23:
I am not c++ expert and if i remember that part of code is mine, but it seems to be strange issue, never reported before. Like i said im not expert, but i can see now situation when you read remote characteristic and length od read value is 0, when you will read next time then m_rawData in not nullptr but cant be free because its been initialized with length 0. Im not sure if this is the case but its possible i guess.

h2zero, 2018-11-03:
I started having the problem today as well, I wrote some code that connects to a peripheral, reads the characteristic and then disconnects. I put a loop around it to repeat every 30 seconds and after 3 or 4 reads I get this error. 

I tested for the 0 length calloc()  and that doesn't seem to be the cause, i'll investigate this further tonight.

h2zero, 2018-11-03:
Found the problem, when reconnecting to peripheral the charaacteristics/services are released and m_rawData doesn't always = nullptr when remote charateristic is constructed. 

Adding  `m_rawData = nullptr; ` to the BLERemoteCharacteristic constructor and 
`if(m_rawData != nullptr){`
     `free(m_rawData);}`

to the deconstructor fixed the error and takes care of potential memory leaks.

hansmbakker, 2018-12-06:
@h2zero thank you for the fix!

@hetlelid could you please close this issue to keep the issue list clean?
cc @chegewara

hetlelid, 2018-12-07:
Close #654

commit 89aac0a8 (2018-12-19): fix #654
commit c48cb191 (2018-12-19): Merge pull request #761 from kerikun11/fix/issue_654

fix #654
closed 2018-12-07 by chegewara