nkolban/esp32-snippets #656: readValue() in notifyCallback of Client gives outdated result.

Sonnenschwein, 2018-09-25:
When using two Adafruit HUZZAH 32 Feathers, one as a server and one as a client, I encounter a strange problem. 

The server uses the (slightly modified) BLE_UART example, to send ascending numbers each second and notifies the connected client. The client then requests the value. My problem is, that the value I get at the client side is always one below the actually sent number, so it's always the last sent value and not the current one. I checked with my phone and there the values are right. 

This is the BLE_uart.ino:

	/*
		Video: https://www.youtube.com/watch?v=oCMOYS71NIU
		Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleNotify.cpp
		Ported to Arduino ESP32 by Evandro Copercini

	   Create a BLE server that, once we receive a connection, will send periodic notifications.
	   The service advertises itself as: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
	   Has a characteristic of: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E - used for receiving data with "WRITE" 
	   Has a characteristic of: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E - used to send data with  "NOTIFY"

	   The design of creating the BLE server is:
	   1. Create a BLE Server
	   2. Create a BLE Service
	   3. Create a BLE Characteristic on the Service
	   4. Create a BLE Descriptor on the characteristic
	   5. Start the service.
	   6. Start advertising.

	   In this example rxValue is the data received (only accessible inside that function).
	   And txValue is the data to be sent, in this example just a byte incremented every second. 
	*/
	#include <BLEDevice.h>
	#include <BLEServer.h>
	#include <BLEUtils.h>
	#include <BLE2902.h>

	//BLECharacteristic *pCharacteristic;
	bool deviceConnected = false;
	uint8_t txValue = 0;

	TaskHandle_t Task1;

	// See the following for generating UUIDs:
	// https://www.uuidgenerator.net/

	#define SERVICE_UUID           "c02763a4-0021-11e8-ba89-0ed5f89f718b" // UART service UUID
	#define CHARACTERISTIC_UUID_RX "c027661a-0021-11e8-ba89-0ed5f89f718b"
	#define CHARACTERISTIC_UUID_TX "c027698a-0021-11e8-ba89-0ed5f89f718b"

	//BLECharacteristic *pointerCharacteristic;

	class MyServerCallbacks: public BLEServerCallbacks {
		void onConnect(BLEServer* pServer) {
		  deviceConnected = true;
		};

		void onDisconnect(BLEServer* pServer) {
		  deviceConnected = false;
		}
	};

	class MyCallbacks: public BLECharacteristicCallbacks {
		void onWrite(BLECharacteristic *pCharacteristic) {
		  std::string rxValue = pCharacteristic->getValue();

		  if (rxValue.length() > 0) {
			Serial.println("*********");
			Serial.print("Received Value: ");
			for (int i = 0; i < rxValue.length(); i++)
			  Serial.print(rxValue[i]);

			Serial.println();
			Serial.println("*********");
		  }
[... 236 lines cut ...]
	configsip: 0, SPIWP:0xee
	clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
	mode:DIO, clock div:1
	load:0x3fff0018,len:4
	load:0x3fff001c,len:1324
	load:0x40078000,len:7788
	ho 0 tail 12 room 4
	load:0x40080400,len:6448
	entry 0x400806e8
	Starting Arduino BLE Client application...
	BLE Advertised Device found: Name: ECU1, Address: 30:ae:a4:26:37:82, serviceUUID: c02763a4-0021-11e8-ba89-0ed5f89f718b, txPower: 3
	Found our device!  address: Forming a connection to 30:ae:a4:26:37:82
	 - Created client
	 - Connected to server
	 - Found our service
	We are now connected to the BLE Server.
	The characteristic value was: 0
	Notify callback for characteristic c027698a-0021-11e8-ba89-0ed5f89f718b of data length 1
	The characteristic value was: 0
	Notify callback for characteristic c027698a-0021-11e8-ba89-0ed5f89f718b of data length 1
	The characteristic value was: 1
	Notify callback for characteristic c027698a-0021-11e8-ba89-0ed5f89f718b of data length 1

As you can see, the value of the second readValue() is 0 even if it should be 1.
What is the cause for this? Can this be fixed? This behaviour is problematic for me, because I need to receive notifications with information on what happens right now, and not what happened the last time a notification was sent. Thank you in advance

chegewara, 2018-09-25:
This is where you have bug:
```
if(firstTime){
		pCharacteristic->notify();
		firstTime = false;
	  }
	  pCharacteristic->notify();
```
When you are connected then you send notification two times with the same value, thats why you get two notifications with value 0.

BTW your RX characteristic does not need `pCharacteristic->addDescriptor(new BLE2902());`

Sonnenschwein, 2018-09-25:
@chegewara unfortunately that is not where it goes wong. This part is how I tried to fix the issue, but it does not change a thing. Also, if I read the value with my phone, sent and received values are in sync.

chegewara, 2018-09-25:
Then you should post original code. This code match logs you posted and it match exactly problem you described. 
If you expect help with your issue then post original logs and eventually code. I can only comment what i can read from logs/code.

chegewara, 2018-09-25:
@Sonnenschwein There is one more thing. I dont know how did you do that, but there is issue with read/write characteristic from within notificationCallback. Its because there is a lot of semaphores in library and some of them are blocking each other. From what i know your code should stuck when you try to read characteristic. There is one topic here about this issue. At the moment best approach is to set flag in notificationCallback and read characteristic in loop or in additional designated task.

chegewara, 2018-10-02:
@Sonnenschwein any progress with this issue?

chegewara, 2018-10-04:
There is some issue with semaphores. Read from within notification callback is causing that semaphore cant be taken. I have temporary solution for you if you are interested:
```
void read_task(void* data){
	BLERemoteCharacteristic* pBLERemoteCharacteristic = (BLERemoteCharacteristic*)data;
	ESP_LOGI(LOG_TAG, "%s", pBLERemoteCharacteristic->readValue().c_str());
	vTaskDelete(NULL);
}

static void notifyCallback(
	BLERemoteCharacteristic* pBLERemoteCharacteristic,
	uint8_t* pData,
	size_t length,
	bool isNotify) {
		ESP_LOGE(LOG_TAG, "Notify callback for characteristic %s of data %s length %d",
				pBLERemoteCharacteristic->getUUID().toString().c_str(), ((char*) pData), length);
	xTaskCreate(read_task, "read", 2048, (void*)pBLERemoteCharacteristic, 5, nullptr);
}
```

chegewara, 2018-10-04:
I know that commenting semaphore.wait solving that issue, but i dont know what other issues it can cause. I did it too in my tests, also i changed this function:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/FreeRTOS.cpp#L85
with:
```
uint32_t FreeRTOS::Semaphore::wait(std::string owner) {
	ESP_LOGV(LOG_TAG, ">> wait: Semaphore waiting: %s for %s", toString().c_str(), owner.c_str());

bool rc = false;
	if (m_usePthreads) {
		pthread_mutex_lock(&m_pthread_mutex);
	} else {
		rc = take(700, owner);
	}

	m_owner = owner;

	if (m_usePthreads) {
		pthread_mutex_unlock(&m_pthread_mutex);
	} else {
		if (rc) xSemaphoreGive(m_semaphore);
	}

	ESP_LOGV(LOG_TAG, "<< wait: Semaphore released: %s", toString().c_str());
	if(rc)
	m_owner = std::string("<N/A>");
	return m_value;
} // wait
```

Every separate solution gives me expected result, but as i said, i dont know what issue it can cause. Only solution with new task seems to be safe (but im not sure either).