nkolban/esp32-snippets #659: ESP32 BLE HID send crush symbols in the product description

ARPavel22, 2018-09-28:
Hi, and Thank you for your great work!

I'm trying to create a gamepad for the [UNITY](https://unity3d.com).
Everything works perfectly. The code works.

But when ESP connect to a computer, one of the parameters of HID is **NOT** recognized correctly.
Most likely it is **"PRODUCT NAME"**.

I think that Chinese people sewed their own hieroglyphics during the testing phase of the motherboard.

You can test the work of a hid with a simple program:
Joystic test: http://www.planetpointy.co.uk/joystick-test-application/

Here are screenshots of poor performance:

[image]
[image]
[image]

The problem is **not in the name of the Bluetooth device**. It is displayed correctly.

Can I **change the other settings of the HID device**?

chegewara, 2018-09-28:
You can use those function to setup hid specific parameters:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEHIDDevice.h#L43-L49

ARPavel22, 2018-09-29:
> You can use those function to setup hid specific parameters:
> https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEHIDDevice.h#L43-L49

Thank you for the quick response.
You showed me the right way to search.

Changing the values of PNP I noticed an interesting feature.
When the variable reaches the Unity editor, it is inverted:
[image]

This leads to the thought that string manufacture also comes to be broken. Therefore, you can see the hieroglyphs and strange symbols in the name of the device.

After I swapped parts of the variable
I saw that in the unit they came as they should be.
// hid-> pnp (0x02, 0x0810, 0xe501, 0x0106);
//CHANGE TO
     hid-> pnp (0x02, 0x1008, 0x01e5, 0x0601);
[image]

It remains to understand how to properly pass on the manufacture data:
    std::string name = "esp-community";
    hid->manufacturer()->setValue(name);

chegewara, 2018-09-29:
Hi, thanks for testing and explaining how you solved your issue.
It is how bluetooth hid works, its so called `little endian`. Bytes order is swapped. What is strange its manufacturer name, in device manager ive been checking and it was displayed as it should be (some time ago). Could you check it please?

Also for vendorID you can check this site, espressif is 0x02E5(you dont have to use it):
https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers

mitchjs, 2018-09-29:
Isn’t USB little endian?

 

And all descriptor strings are UNICODE (I think UTF-16)

 

I think the function, should swap endianness of the vid/pid …

void     pnp(uint8_t sig, uint16_t vid, uint16_t pid, uint16_t version);

Just to make ones code cleaner

 

 

mitch

 

https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers

chegewara, 2018-09-29:
> The fields in the above table are in the order of LSO to MSO. Where LSO = Least Significant Octet and MSO = Most Significant Octet

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.pnp_id.xml

ARPavel22, 2018-09-29:
> > The fields in the above table are in the order of LSO to MSO. Where LSO = Least Significant Octet and MSO = Most Significant Octet
> 
> https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.pnp_id.xml

I'm going to test your advice.
But first I want to clarify the sequence of variables.

**About HID INFO:**
`    hid->hidInfo(0x00,0x01);`
1. Country code: 0x00 - without country
2. Some flags (which I did not understand) but just set 0x01

===================================================================

**About PNP:**
```
void BLEHIDDevice::pnp(uint8_t sig, uint16_t vid, uint16_t pid, uint16_t version) 

hid-> pnp (0x02, 0x1008, 0x01e5, 0x0601);
```
1. SIG:
0x01 = Bluetooth SIG assigned Company Identifier value from the Assigned Numbers document
0x02 = USB Implementer’s Forum assigned Vendor ID value
**(Should I use the first? 0x01)**

2. vid - Get from list: https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers

3. **pid ??? can i use 0x01?**
4. **version ??? can i use 0x01?**

===================================================================

And last. What is passed in these two parameters? This is similar to the HID INFO?
**uint8_t val[] = {0x01, 0x00};
desc->setValue(val, 2);**
```
  void onConnect(BLEServer* pServer){
    Serial.println("connected");
                // workaround after reconnect (see comment below) 
                BLEDescriptor *desc = input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
                uint8_t val[] = {0x01, 0x00};
                desc->setValue(val, 2);

                _connected = true;
  }
```

chegewara, 2018-09-29:
1. Im guessing you should use 0x01 since you want to build bluetooth hid device;
2. To be honest i dont know if you can use espressif VID, esp32 is espressif board but final product is not; you should ask this on espressif/esp-idf github or on esp32.com forum
3. and 4. You can use whatever you want for PID
5. Yes. There is no info about value being little endian, but its the same case:
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml

ARPavel22, 2018-09-30:
> Could you intercept raw advertising packet from RN42 and from esp32?

I will try to grab packets with this soft:
https://www.hhdsoftware.com/device-monitoring-studio

If u can advice most simle soft or decision how to do this, please, send here.

chegewara, 2018-09-30:
I am using app on android, nRF connect.

ARPavel22, 2018-09-30:
> I am using app on android, nRF connect.

[image]

chegewara, 2018-10-07:
Hi,
any progress here?

chegewara, 2018-10-07:
Maybe there is some option to change name parsing to utf8? Because it looks like it is parsed with unicode standard probably.

ARPavel22, 2018-10-08:
> Maybe there is some option to change name parsing to utf8? Because it looks like it is parsed with unicode standard probably.

**Good news.** I received a response from the Unity team:
https://forum.unity.com/threads/input-system-update.508660/page-7#post-3763249

chegewara, 2018-10-08:
@ARPavel22 Are you using generic library in unity or did you have to buy ble library from asset store? I would like to play with unity hid too.

ARPavel22, 2018-10-08:
> @ARPavel22 Are you using generic library in unity or did you have to buy ble library from asset store? I would like to play with unity hid too.

There is new input system:
https://github.com/Unity-Technologies/InputSystem

For working with it repo you need Unity 2018.2.10 f 1 or upper

chegewara, 2018-10-08:
Great, i will try it. Ive been searching for ble library but before i could not find any for free.
Time to run unity3d. Thanks.

ARPavel22, 2018-10-08:
> Great, i will try it. Ive been searching for ble library but before i could not find any for free.
> Time to run unity3d. Thanks.

I do't use ble assets.
There is ESP32 HID device and Unity find this by input manager.

chegewara, 2018-10-08:
@ARPavel22 Could you share your unity3d code? At least part with hid device. I see unity3d input plugin is under development so i can see what i can do to help.

ARPavel22, 2018-10-08:
> @ARPavel22 Could you share your unity3d code? At least part with hid device. I see unity3d input plugin is under development so i can see what i can do to help.

I haven't even started a project yet.

- Just downloaded the repository manager. (This is ready to open Unity3D project)
- Then I open the Unity.
- I choose to OPEN a project.
- I choose the unzipped folder Input manager (master).
- Then open the project.
- And I look in the window of the dimen device.

chegewara, 2018-10-09:
Ok, the problem is hard to solve. Unity devs are using standard windows `HidD_GetManufacturerString`. This function suppose to return wchar string. I have no idea how it should read hid this ble hid descriptor. 
Its not unity devs fault (my bad), you should rather ask microsoft community/devs why utf8 string which is properly read and parsed by windows drivers cant be passed properly in `HidD_GetManufacturerString`.

ARPavel22, 2018-10-09:
> Ok, the problem is hard to solve. Unity devs are using standard windows `HidD_GetManufacturerString`. This function suppose to return wchar string. I have no idea how it should read hid this ble hid [...]

https://forum.unity.com/threads/input-system-update.508660/page-7#post-3764593

chegewara, 2018-10-09:
Yes, i did read that. See this:
>Pointer to a caller-allocated buffer that the routine uses to return the collection's manufacturer string. The routine returns a NULL-terminated **wide character string** in a human-readable format.

https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/content/hidsdi/nf-hidsdi-hidd_getmanufacturerstring

ARPavel22, 2018-10-09:
How do I change the BLEHIDDevice.cpp + BLEHIDDevice.h + BLECharacteristic.h files to send string in UTF16?

ARPavel22, 2018-10-09:
> try this in your unity code:
> https://docs.microsoft.com/en-us/dotnet/api/system.text.encoder?view=netframework-4.7.2

It does not work. Unity3D gives access to the string already parsed. And it is either empty or mistakenly chosen manufactory of another connected device.

chegewara, 2018-10-09:
@ARPavel22 This most likely wont solve your issue, but wont cause another. Fix your mine app code:
`BLEHIDDevice* hid;` <--- delete `static` word

ARPavel22, 2018-10-09:
> @ARPavel22 This most likely wont solve your issue, but wont cause another. Fix your mine app code:
> `BLEHIDDevice* hid;` <--- delete `static` word

I removed the static in the main program but it did not change anything.

chegewara, 2018-10-09:
@ARPavel22 There's maybe one more problem with windows. In device manager my esp32 ble hid device is only enumerated in bluetooth category. I dont see it in `sound, video and game controllers`. Also in unity `Input debug` shows me esp32 as `unsupported` device even if input buttons are recognized (short video on YT).

ARPavel22, 2018-10-09:
> @ARPavel22 There's maybe one more problem with windows. In device manager my esp32 ble hid device is only enumerated in bluetooth category. I dont see it in `sound, video and game controllers`. Also [...]

Try build this in arduino IDE:

`#include <Arduino.h>

/**
 * Create a new BLE server.
 */
#include "BLEDevice.h"
#include "BLEServer.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include "BLEHIDDevice.h"
BLEHIDDevice* hid;
BLECharacteristic* input;
BLECharacteristic* output;

bool _connected = false;
int brightness = 0;    // как ярко горит светодиод
int fadeAmount = 1;    // на какое значение будет затухать/загораться светодиод
uint8_t t = 0x0106;

class MyCallbacks : public BLEServerCallbacks {
  void onConnect(BLEServer* pServer){
    Serial.println("connected");

                // workaround after reconnect (see comment below) 
                BLEDescriptor *desc = input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
                uint8_t val[] = {0x01, 0x00};
                 //uint8_t val[] = {0x00, 0x01};
                desc->setValue(val, 2);

                _connected = true;
  }

  void onDisconnect(BLEServer* pServer){
        Serial.println("disconnected");
        _connected = false;
  }
};

void setup() {
    //
    Serial.begin(115200);
    Serial.println("Starting");

    //
    BLEDevice::init("ESP32_2");

    BLEServer *pServer = BLEDevice::createServer();
    pServer->setCallbacks(new MyCallbacks());

    /*
     * Instantiate hid device
     */
    hid = new BLEHIDDevice(pServer);

    input = hid->inputReport(1); // <-- input REPORTID from report map
    output = hid->outputReport(1); // <-- output REPORTID from report map

    /*
     * Set manufacturer name (OPTIONAL)
     * https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.manufacturer_name_string.xml
     */
   
    std::string name = "esp-community";
    hid->manufacturer()->setValue(name);
      
    /*
[... 86 lines cut ...]
      }    
      // ждем 30 миллисекунд для лучшего эффекта
      delay(5);                            
    }
}

void Send()
{
  /*
      // put your main code here, to run repeatedly:
    uint8_t a[] = {0x03, 0x03, 0x03, 0x03, 0x03, 0x03, 0x03};

    input->setValue(a, sizeof(a));
    input->notify();

    delay(500);
    */

    uint8_t b[] = {0x00, 0x00, brightness, 100 - brightness, brightness, 100 - brightness, brightness};

    input->setValue(b, sizeof(b));
    input->notify();

    //delay(500);
}`

chegewara, 2018-10-09:
Yes, i know. I am preparing fix to it.

ARPavel22, 2018-10-09:
> Yes, i know. I am preparing fix to it.

How can I find your YT channel?

chegewara, 2018-10-09:
@ARPavel22 I can see your device as `E502-1E5`.

my channel is `chegewaras`

ARPavel22, 2018-10-09:
> @ARPavel22 I can see your device as `E502-1E5`.
> 
> my channel is `chegewaras`

    hid->pnp(0x01, 0x02E5, 0xe501, 0x0106);

chegewara, 2018-10-09:
Try this code, its random generated value for buttons and axis.
https://gist.github.com/chegewara/d3d2160f6be3ad2c1621daa6e666a115
[image]

ARPavel22, 2018-10-09:
> Try this code, its random generated value for buttons and axis.
> https://gist.github.com/chegewara/d3d2160f6be3ad2c1621daa6e666a115
> [image]

Yes, the buttons and axes work, but I have successfully done this before.
The device with your code is still recognized randomly:

[image]

ARPavel22, 2018-10-12:
Look at this:
https://forum.unity.com/threads/input-system-update.508660/page-7#post-3767413

Dotby, 2021-06-05:
> Похоже, вас поддерживает команда Unity.

Hello again. I started to port my work to the browser as it became possible to work with the WEB HID API. And even here I encountered the same problem. Windows wants UTF 16. If you have time, think about this task. 
[image]