nkolban/esp32-snippets #660: Can I get descriptor's value field?

coppercele, 2018-09-29:
Hi, im trying to connect ESP32(client) and HID Gamepad(Server) with BLE_client sample.

My gamepad has multiple characteristics, and its have same CharUUID and DescripoterUUID.
but descripotrs value field is diffelent.

In attached file, top of charcteristic(Value fields are 01 00 and 04 00) is correct,
and others(value fields are null) are not correct.

Then I want to get value field to find correct characteristics, but it seems value field is not implemented in BLERemoteDescriptor.
(Multiple Characteristics that has same CharUUID is not assumpted?)

Can I get descriptor's value field?

My english is poor,sorry ;D

[image]

chegewara, 2018-09-29:
Try to change this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEHIDDevice.cpp#L19

Change `40` to `100` or something.

coppercele, 2018-10-01:
Thanks for reply.
But its not take effect.

Then I changed BLERemoteService::retrieveCharacteristics() to accept same Characteristics.

m_characteristicMap.insert(std::pair<std::string, BLERemoteCharacteristic*>(pNewRemoteCharacteristic->getUUID().toString(), pNewRemoteCharacteristic));
↓
char str[10];
sprintf(str, "%u", offset);
m_characteristicMap.insert(std::pair<std::string, BLERemoteCharacteristic*>(pNewRemoteCharacteristic->getUUID().toString() + str, pNewRemoteCharacteristic));

And I got 9 Charactirstics. (UUID+offset)
00002a4a-0000-1000-8000-00805f9b34fb0
00002a4b-0000-1000-8000-00805f9b34fb3
00002a4c-0000-1000-8000-00805f9b34fb1
00002a4d-0000-1000-8000-00805f9b34fb4
00002a4d-0000-1000-8000-00805f9b34fb5
00002a4d-0000-1000-8000-00805f9b34fb6
00002a4d-0000-1000-8000-00805f9b34fb7
00002a4d-0000-1000-8000-00805f9b34fb8
00002a4e-0000-1000-8000-00805f9b34fb2

I found correct Characteristic, but its hard coded. because I cannnot get Discriptor's value field(´・ω・`)

    int i = 0;
    auto itr = pCharacteristicMap->begin();
    for(auto itr = pCharacteristicMap->begin(); itr != pCharacteristicMap->end(); ++itr) {

        if (charUUID.toString() == itr->second->getUUID().toString()) {  // 0x00002a4d
			Serial.print("CharUUID mached: ");

			if (itr->second->canRead() && itr->second->canNotify() && !itr->second->canWrite()) { // Characteristic Property
				if (i == 1) { // 2nd of NR and not Write Characteristics. but its hard coded.
					Serial.print("Correct : ");
...

chegewara, 2018-10-01:
Ok, im not sure i understand you. You want to have esp32 hid device or you have hid device and you want to build esp32 host device?

coppercele, 2018-10-01:
I want to connect Bluetooth Gamepad(BLE Server) and ESP32(BLE Client).

and i want to control RC Car(with ESP32) by Gamepad with BLE HID.

Gamepad is this. https://www.amazon.co.jp/gp/product/B01N0V7UWY/

chegewara, 2018-10-01:
Ok, that sound cool. I will try to help you with any issue with HID class and connecting client with server. I need to prepare some test code and see whats going on.

chegewara, 2018-10-01:
@coppercele I see where is the problem. There is `getCharacteristics()` function that you found left for back compatibility reasons and there is another for multiple characteristics with the same UUID https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteService.h#L37

But the issue is it is only declared in header file. I have no idea why it is not implemented. I will try to add implementation when i find some time. Im trying to do some bugfixes and additions to this library.

coppercele, 2018-10-01:
Thank you!
I attched my test codes.
ino and cpp are not accepted, then i renamed to txt.

It can read from HID like below.
it seems can read datas length 2 and 4, but cannot 8 or up?
Notify callback for characteristic 00002a4d-0000-1000-8000-00805f9b34fb of data length 4
Data : 00000000

[BLE_client.txt](https://github.com/nkolban/esp32-snippets/files/2435780/BLE_client.txt)

[BLERemoteService.txt](https://github.com/nkolban/esp32-snippets/files/2435783/BLERemoteService.txt)

chegewara, 2018-10-02:
@coppercele Try this code, this is BLERemoteService.cpp:
```
/*
 * BLERemoteService.cpp
 *
 *  Created on: Jul 8, 2017
 *      Author: kolban
 */
#include "sdkconfig.h"
#if defined(CONFIG_BT_ENABLED)

#include <sstream>
#include "BLERemoteService.h"
#include "BLEUtils.h"
#include "GeneralUtils.h"
#include <esp_log.h>
#include <esp_err.h>
#ifdef ARDUINO_ARCH_ESP32
#include "esp32-hal-log.h"
#endif

static const char* LOG_TAG = "BLERemoteService";

BLERemoteService::BLERemoteService(
		esp_gatt_id_t srvcId,
		BLEClient*    pClient,
		uint16_t      startHandle,
		uint16_t      endHandle
	) {

	ESP_LOGD(LOG_TAG, ">> BLERemoteService()");
	m_srvcId  = srvcId;
	m_pClient = pClient;
	m_uuid    = BLEUUID(m_srvcId);
	m_haveCharacteristics = false;
	m_startHandle = startHandle;
	m_endHandle = endHandle;

	ESP_LOGD(LOG_TAG, "<< BLERemoteService()");
}

BLERemoteService::~BLERemoteService() {
	removeCharacteristics();
}

/*
static bool compareSrvcId(esp_gatt_srvc_id_t id1, esp_gatt_srvc_id_t id2) {
	if (id1.id.inst_id != id2.id.inst_id) {
		return false;
	}
	if (!BLEUUID(id1.id.uuid).equals(BLEUUID(id2.id.uuid))) {
		return false;
	}
	return true;
} // compareSrvcId
*/

/**
 * @brief Handle GATT Client events
 */
void BLERemoteService::gattClientEventHandler(
	esp_gattc_cb_event_t      event,
	esp_gatt_if_t             gattc_if,
	esp_ble_gattc_cb_param_t *evtParam) {
	switch(event) {
		//
		// ESP_GATTC_GET_CHAR_EVT
		//
		// get_char:
		// - esp_gatt_status_t    status
[... 228 lines cut ...]
 */
void BLERemoteService::setValue(BLEUUID characteristicUuid, std::string value) {
	ESP_LOGD(LOG_TAG, ">> setValue: uuid: %s", characteristicUuid.toString().c_str());
	getCharacteristic(characteristicUuid)->writeValue(value);
	ESP_LOGD(LOG_TAG, "<< setValue");
} // setValue

/**
 * @brief Create a string representation of this remote service.
 * @return A string representation of this remote service.
 */
std::string BLERemoteService::toString() {
	std::ostringstream ss;
	ss << "Service: uuid: " + m_uuid.toString();
	ss << ", start_handle: " << std::dec << m_startHandle << " 0x" << std::hex << m_startHandle <<
			", end_handle: " << std::dec << m_endHandle << " 0x" << std::hex << m_endHandle;
	for (auto &myPair : m_characteristicMap) {
		ss << "\n" << myPair.second->toString();
	   // myPair.second is the value
	}
	return ss.str();
} // toString

#endif /* CONFIG_BT_ENABLED */
```

chegewara, 2018-10-02:
Modified code in your ble_client.ino:
```
	  std::map<uint16_t, BLERemoteCharacteristic *>* pCharacteristicMap;

    pRemoteService->getCharacteristics(pCharacteristicMap);

    Serial.print("Number of Charactirstic: ");
    Serial.println(pCharacteristicMap->size());

//    int i = 0;

    auto itr = pCharacteristicMap->begin();
    for(auto itr = pCharacteristicMap->begin(); itr != pCharacteristicMap->end(); ++itr) {
        Serial.print("first: ");
        Serial.print(itr->second->getUUID().toString());
        Serial.print(" second :");
        Serial.println(itr->second->getUUID().toString().c_str());

     if (charUUID.toString() == itr->second->getUUID().toString()) {
			Serial.print("CharUUID mached: ");
			Serial.println(itr->second->getUUID().toString());

			if (itr->second->canRead() && itr->second->canNotify() && !itr->second->canWrite()) {
//				if (i == 1) {
					Serial.print("Correct : ");
//					Serial.println(i);
					pRemoteCharacteristic = itr->second;

					    // Read the value of the characteristic.
					    std::string value = pRemoteCharacteristic->readValue();
					    Serial.print("The characteristic value was: ");
					    Serial.println(value.c_str());
					    pRemoteCharacteristic->registerForNotify(notifyCallback);
//					break;
//				}
//				i++;
				Serial.println("next");

			}
			else {
				Serial.println("Incorrect");
			}
```

chegewara, 2018-10-03:
It seems that your function looks better than mine. I will change it in code then.

chegewara, 2018-10-03:
You can have notification callback that will be tied to every input report characteristic by reportID or you can have only one input report characteristic and all necessary data packed in one single message.

Of course you can have single notification callback function for all characteristics.

coppercele, 2018-10-04:
I succeed controling RC tank with Bluetooth HID.

↓I uploaded a movie gif to imgur.com
https://imgur.com/a/3yHAKKn

chegewara, 2018-10-04:
Nice job.

closed 2018-10-05 by coppercele

coppercele, 2018-10-05:
Thank you very much =D

Dario-Ciceri, 2022-08-22:
> Hi, I tryed its. BLERemoteService::getCharacteristics() not return multiple same Characteristics. then I made BLERemoteService::getCharacteristicsByHandle() in BLERemoteService.h and .cpp.
> 
> std::map<uint16_t, BLERemoteCharacteristic *> * BLERemoteService::getCharacteristicsByHandle() { ESP_LOGD(LOG_TAG, ">> getCharacteristics() for service: %s", getUUID().toString().c_str()); if (!m_haveCharacteristics) { retrieveCharacteristics(); } ESP_LOGD(LOG_TAG, "<< getCharacteristics() for service: %s", getUUID().toString().c_str()); return &m_characteristicMapByHandle; }
> 
> and i used it in BLE_Client.ino. then I can read from Gamepad =D Notify callback for characteristic 00002a4d-0000-1000-8000-00805f9b34fb of data length 4 Data : 02000000
> 
> I cound not notice that resisterForNotify() can accept all same characteristics.

Hi, I'm trying to do the same with another joystick, could you share your code for ble client? 
Thank you ಥ_ಥ

coppercele, 2022-08-22:
Hi,I wrote this about it in this URL(in Japanese)

https://qiita.com/coppercele/items/4ce0e8858a92410c81e3

Finally,I didnt use value fields.
then I registered all "notifyable" characteristics to Notify callback.

  NimBLERemoteService *pSvc = nullptr;
//  NimBLERemoteCharacteristic *pChr = nullptr;
  std::vector<NimBLERemoteCharacteristic*> *pChrs = nullptr;

  NimBLERemoteDescriptor *pDsc = nullptr;

  pSvc = pClient->getService(serviceUUID);
  if (pSvc) { /** make sure it's not null */
    pChrs = pSvc->getCharacteristics(true);
  }

  if (pChrs) { /** make sure it's not null */
    for (int i = 0; i < pChrs->size(); i++) {

      if (pChrs->at(i)->canNotify()) {
        if (!pChrs->at(i)->registerForNotify(notifyCB)) {
          /** Disconnect if subscribe failed */
          pClient->disconnect();
          return false;
        }
      }
    }
  }

Dario-Ciceri, 2022-08-22:
> Hi,I wrote this about it in this URL(in Japanese)
>
> https://qiita.com/coppercele/items/4ce0e8858a92410c81e3
>
> Finally,I didnt use value fields. then I registered all "notifyable" characteristics to [...]

I'll check it now, can I use it with arduino IDE?
Do you have some time to give me so I can explain what I would like to do?

Dario-Ciceri, 2022-08-22:
> Hi,I wrote this about it in this URL(in Japanese)
>
> https://qiita.com/coppercele/items/4ce0e8858a92410c81e3
>
> Finally,I didnt use value fields. then I registered all "notifyable" characteristics to [...]

It's a little bit confusing since I translated the tutorial, could you send me the code for esp32 not M5 stick ? 
Thank you, your help is very appreciated \(@^0^@)/

limeylemons80, 2025-05-28:
by the way can I use this code for a selfie stick HID BLE button

limeylemons80, 2025-05-28:
i am a student who is unfamiliar with coding. just looking to copy and paste hahah.