nkolban/esp32-snippets #661: Esp32 restarts when btStop() function is called.

mimansamaheshwari, 2018-09-29:
I am facing problem in shutting down bluetooth. When I use btStop() function in example code of BLE_uart for BLE library my esp32 restarts after few seconds. Below is the code i am using:
```
#include <BLEDevice.h>
#include <BLEServer.h>
#include <BLEUtils.h>
#include <BLE2902.h>

BLECharacteristic *pCharacteristic;
BLEServer *pServer;

bool deviceConnected = false;
uint8_t txValue = 0;
int counter=0;
// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID           "6E400001-B5A3-F393-E0A9-E50E24DCCA9E" // UART service UUID
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
    }
};

class MyCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string rxValue = pCharacteristic->getValue();

      if (rxValue.length() > 0) {
        Serial.println("*********");
        Serial.print("Received Value: ");
        for (int i = 0; i < rxValue.length(); i++)
          Serial.print(rxValue[i]);

        Serial.println();
        Serial.println("*********");
      }
    }
};

void setup() {
  Serial.begin(115200);

  // Create the BLE Device
  BLEDevice::init("UART Service");

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);

  // Create a BLE Characteristic
  pCharacteristic = pService->createCharacteristic(
                      CHARACTERISTIC_UUID_TX,
                      BLECharacteristic::PROPERTY_NOTIFY
                    );
                      
  pCharacteristic->addDescriptor(new BLE2902());

  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID_RX,
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setCallbacks(new MyCallbacks());

  // Start the service
  pService->start();

  // Start advertising
  pServer->getAdvertising()->start();
  Serial.println("Waiting a client connection to notify...");
}

void loop() {

while(counter < 5)
{
  if (deviceConnected) {
    Serial.printf("*** Sent Value: %d ***\n", txValue);
    pCharacteristic->setValue(&txValue, 1);
    pCharacteristic->notify();
    txValue++;
    counter++;
  }
  delay(1000);
}
 
pServer->stopAdvertising();
if(btStop())
{
  Serial.println("Bluetooth turned off Successfully.");
}
while(1); 

}
```

I also get following error on Serial monitor before restart:
```
Waiting a client connection to notify...
*** Sent Value: 0 ***
*** Sent Value: 1 ***
*** Sent Value: 2 ***
*** Sent Value: 3 ***
*** Sent Value: 4 ***
Bluetooth turned off Successfully.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4004baaa  PS      : 0x00060031  A0      : 0x80085d2d  A1      : 0x3ffc0560  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc5b04  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb00d8  
A10     : 0x00000000  A11     : 0x3ffd4fe0  A12     : 0x8008780c  A13     : 0x3ffd9db0  
A14     : 0x00000000  A15     : 0x3ffc3670  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4004baaa:0x3ffc0560 0x40085d2a:0x3ffc0580 0x400156a5:0x3ffc05b0 0x400552cd:0x3ffc05d0 0x40086db7:0x3ffc05f0 0x4008166d:0x3ffc0610 0x40138317:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
Waiting a client connection to notify...
```
The code I have mentioned above is just to know how to stop bluetooth when its not in use. I want to use this function in certain projects I will be working on.

chegewara, 2018-09-29:
Try this code instead of `btStop` (you wont be able to use BT anymore until you restart esp32):
```
esp_bluedroid_disable();
 esp_bluedroid_deinit();
 esp_bt_controller_disable();
 esp_bt_controller_deinit();
```
At the end you can try to add this to release heap:
`esp_bt_mem_release(ESP_BT_MODE_BTDM);`

chegewara, 2018-10-16:
@S-March To be honest i dont know, never been trying it. But i have also good news. Until you wont release memory you can deinit and init bt 
https://github.com/nkolban/esp32-snippets/issues/630#issuecomment-427635732