nkolban/esp32-snippets #662: device reboot

xiaodingding, 2018-09-29:
lib:
ESP32_BLE
Backtrace: 0x40090c38:0x3ffd1dc0 0x40090e3b:0x3ffd1de0 0x4018315f:0x3ffd1e00 0x401831a6:0x3ffd1e20 0x40170bd3:0x3ffd1e40 0x400d4ebd:0x3ffd1e60 0x400d1a9d:0x3ffd1ec0 0x400d1b86:0x3ffd1f10 0x400d1d5f:0x3ffd1f70

device connect to ble server ,esp32 will reboot.

chegewara, 2018-09-29:
Use this to decode backtrace:
https://github.com/me-no-dev/EspExceptionDecoder

chegewara, 2018-10-17:
Its hard to say. Could you paste logs before app crash?

chegewara, 2018-10-17:
Try to comment this line, i think its bug in esp-idf:
`WiFi.mode(WIFI_OFF);`

tatsutaigu, 2018-10-24:
My envionment changes to ESP-IDF.
This phnomenon disappeared.
[https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md](url)

chegewara, 2018-10-24:
You have to remember that arduino-esp32 is few commits behind esp-idf master.