nkolban/esp32-snippets #664: BLE Client stuck on m_semaphoreRegEvt.wait("connect");

paweb88, 2018-09-30:
Hi! I working on BLE with iPhone

So i create server add callback and then create client. Everything work fine until client connect:

`[D][ANCSService.cpp:16] startServer(): Starting BLE work!
[D][BLEDevice.cpp:71] createServer(): >> createServer
[D][BLEServer.cpp:282] registerApp(): >> registerApp - 0
[D][BLEDevice.cpp:97] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
[D][BLEUtils.cpp:1791] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
[D][BLEServer.cpp:154] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
[D][BLEServer.cpp:286] registerApp(): << registerApp
[D][BLEServer.cpp:272] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:78] createServer(): << createServer
[D][BLEAdvertising.cpp:140] setAdvertisementData(): >> setAdvertisementData
[D][BLEAdvertising.cpp:148] setAdvertisementData(): << setAdvertisementData
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_RAW_SET_COMPLETE_EVT
[D][BLEAdvertising.cpp:174] start(): >> start: customAdvData: 1, customScanResponseData: 0
[D][BLEAdvertising.cpp:194] start(): - no services a[D][BLEAdvertising.cpp:194] start(): - no services advertised
[D][BLEServer.cpp:112] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEAdvertising.cpp:231] start(): << start
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
[D][BLEUtils.cpp:1285] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:112] handleGAPEvent(): BLEServer ... handling GAP event!
[D][BLEUtils.cpp:1091] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
[D][BLEUtils.cpp:1123] dumpGapEvent(): [status: 0]
[D][BLEServer.cpp:112] handleGAPEvent(): BLEServer ... handling GAP event!
[D][ANCSService.cpp:5] onConnect():
**Device  connected 77:b7:8f:7f:bb:d5**

[D][BLEDevice.cpp:55] createClient(): >> createClient
[D][BLEDevice.cpp:61] createClient(): << createClient
[D][BLEClient.cpp:91] connect(): >> connect(77:b7:8f:7f:bb:d5)
[D][BLEClient.cpp:74] clearServices(): >> clearServices
[D][BLEClient.cpp:81] clearServices(): << clearServices`

And app freez and witiing for `m_semaphoreRegEvt.wait("connect");` in BLEClient

This thing works fine when i put my client code into the Task :( 

Here is my scratch of code:

`void ANCSService::ServerCallback::onConnect(BLEServer *pServer)
{
    ESP_LOGD(ancsService.LOG_TAG, "\n**Device  connected %s**\n", BLEAddress(BLEDevice::m_remoteBda).toString().c_str());
    ancsService->startClient(new BLEAddress(BLEDevice::m_remoteBda));
};

void ANCSService::ServerCallback::onDisconnect(BLEServer *pServer)
{
    ESP_LOGD(ancsService.LOG_TAG, "\n**Device  disconnected**\n");
};

void ANCSService::startServer(std::string appName, uint16_t visibleTime)
{
    ESP_LOGD(LOG_TAG, "Starting BLE work!");
    esp_log_buffer_char(LOG_TAG, LOG_TAG, sizeof(LOG_TAG));
    esp_log_buffer_hex(LOG_TAG, LOG_TAG, sizeof(LOG_TAG));

    // Initialize device
    BLEDevice::init(appName);
    BLEServer *pServer = BLEDevice::createServer();
    pServer->setCallbacks(new ServerCallback(this));
    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
    BLEDevice::setSecurityCallbacks(new Security());

    // Advertising parameters:
    // Soliciting ANCS
    BLEAdvertising *pAdvertising = pServer->getAdvertising();
    BLEAdvertisementData oAdvertisementData = BLEAdvertisementData();
    oAdvertisementData.setFlags(0x01);
    oAdvertisementData.setServiceSolicitation(BLEUUID("7905F431-B5CE-4E99-A40F-4B1E122D00D0"));
    pAdvertising->setAdvertisementData(oAdvertisementData);

    // Set security
    BLESecurity *pSecurity = new BLESecurity();
    pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
    pSecurity->setCapability(ESP_IO_CAP_OUT);
    pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

    //Start advertising
    pAdvertising->start();
};

void ANCSService::startClient(BLEAddress *bleAddress)
{
    BLEClient *pClient = BLEDevice::createClient();
    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
    BLEDevice::setSecurityCallbacks(new Security());

    BLESecurity *pSecurity = new BLESecurity();
    pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
    pSecurity->setCapability(ESP_IO_CAP_IO);
    pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
    pClient->connect(*bleAddress);
    
    BLERemoteService *pAncsService = pClient->getService(BLEUUID("7905F431-B5CE-4E99-A40F-4B1E122D00D0"));
    if (pAncsService == nullptr)
    {
        Serial.println("Nie ma service");
        ESP_LOGD(LOG_TAG, "Failed to find our service UUID: %s", ancsServiceUUID.toString().c_str());
        return;
    }
    _notificationSourceCharacteristic = pAncsService->getCharacteristic(notificationSourceCharacteristicUUID);
    if (_notificationSourceCharacteristic == nullptr)
    {
        Serial.println("Nie ma notify");
        ESP_LOGD(LOG_TAG, "Failed to find our characteristic UUID: %s", notificationSourceCharacteristicUUID.toString().c_str());
        return;
    }
    _controlPointCharacteristic = pAncsService->getCharacteristic(controlPointCharacteristicUUID);
    if (_controlPointCharacteristic == nullptr)
    {
        Serial.println("Nie ma control");
        ESP_LOGD(LOG_TAG, "Failed to find our characteristic UUID: %s", controlPointCharacteristicUUID.toString().c_str());
        return;
    }
    _dataSourceCharacteristic = pAncsService->getCharacteristic(dataSourceCharacteristicUUID);
    if (_dataSourceCharacteristic == nullptr)
    {
        Serial.println("Nie ma data");
        ESP_LOGD(LOG_TAG, "Failed to find our characteristic UUID: %s", dataSourceCharacteristicUUID.toString().c_str());
        return;
    }
    const uint8_t v[] = {0x1, 0x0};
    _notificationSourceCharacteristic->registerForNotify(_notificationSourceCallback);
    _notificationSourceCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t *)v, 2, true);
    _dataSourceCharacteristic->registerForNotify(_dataSourceCallback);
    _dataSourceCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t *)v, 2, true);
    Serial.println("Client aktywny");
}`

chegewara, 2018-09-30:
There is a lot of semaphores in this library. Why you dont want to put it into task if it works for you?
Poważnie

paweb88, 2018-09-30:
I don't want using Task because it makes me some memory leak :(
Ok i try one more time but tell me, when i create client in task i must keep alive this task ? ( infinity loop ? ) 

for example: 
````c
::xTaskCreatePinnedToCore(&ANCSService::startClient, "ServerTask", 10000, new clientParameter(new BLEAddress(BLEDevice::m_remoteBda), ancsService), 5, &ancsService->clientTaskHandle, 0);

void ANCSService::startClient(void *data)
{
    BLEClient *pClient = BLEDevice::createClient();
    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
    BLEDevice::setSecurityCallbacks(new Security());

    clientParameter *clientParam = (clientParameter *)data;
    BLESecurity *pSecurity = new BLESecurity();
    pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
    pSecurity->setCapability(ESP_IO_CAP_IO);
    pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
    pClient->setClientCallbacks(new ClientCallback(clientParam->ancsService));
    pClient->connect(*clientParam->bleAddress);
    while(1){ delay(1000)};
}
```

chegewara, 2018-09-30:
I cant answer on this question. I have one application where i am deleting task in which i am creating some variables, but i am suspecting its wrong approach. If there is leak in task we can try to fix it with some help.

ps tak

chegewara, 2018-10-02:
@paweb88 here is the issue:
```
void ANCSService::ServerCallback::onConnect(BLEServer *pServer)
{
    ESP_LOGD(ancsService.LOG_TAG, "\n**Device  connected %s**\n", BLEAddress(BLEDevice::m_remoteBda).toString().c_str());
    ancsService->startClient(new BLEAddress(BLEDevice::m_remoteBda));
};
```
When you call `ancsService->startClient` from inside onConnect then we have 2 semaphores conflict.

paweb88, 2018-10-02:
@chegewara thx for your time! I resolved this issue and move the Client to other task now it's works fine.

I don't know why it's sometimes don't worked after connect to phone, subscribe to specify characteristic, phone dont notify the new notification but after i "Forgot this device" and recconect and pair it again it works :(

Now i have other issue how to notify from the Task my main app :( but it's not the this libary :P

ps maybe you have an idea :)

```
Task watchdog got triggered. The following tasks did not feed the watchdog in time:

 - IDLE (CPU 1)

Tasks currently running:

CPU 0: IDLE

CPU 1: loopTask
```

On Server connect:
```c
void ANCSService::ServerCallback::onConnect(BLEServer *pServer)
{
    ESP_LOGI(ancsService.LOG_TAG, "\n**Device  connected %s**\n", BLEAddress(BLEDevice::m_remoteBda).toString().c_str());
    if (ancsService->clientTaskHandle != nullptr) {
        ::xTaskCreatePinnedToCore(&ANCSService::startClient, "ClientTask", 10000, new clientParameter(new BLEAddress(BLEDevice::m_remoteBda), ancsService), 5, &ancsService->clientTaskHandle, 0);
    }
}
```

Client Task:
```c
void ANCSService::startClient(void *data)
{
    ESP_LOGI(ancsService.LOG_TAG, "\n**Client start**\n");
    BLEClient *pClient = BLEDevice::createClient();
    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
    BLEDevice::setSecurityCallbacks(new Security());

    clientParameter *clientParam = (clientParameter *)data;
    BLESecurity *pSecurity = new BLESecurity();
    pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
    pSecurity->setCapability(ESP_IO_CAP_IO);
    pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
    pClient->setClientCallbacks(new ClientCallback(clientParam->ancsService)); // <---- HERE register client callback its typedef function set in setup() main app code 
    pClient->connect(*clientParam->bleAddress);

    BLERemoteService *pAncsService = pClient->getService(BLEUUID("7905F431-B5CE-4E99-A40F-4B1E122D00D0"));
    if (pAncsService == nullptr)
    {
        ESP_LOGD(LOG_TAG, "Failed to find our service UUID: %s", ancsServiceUUID.toString().c_str());
        return;
    }
 /** code **/
}
```

in main setup:
```c
class MyClientCallback : public ANCSServiceClientCallback {
    public:
        MyClientCallback(TFTDisplay *_tft) : tft(_tft) {};
        void onConnect() {
            tft->setBLEConnectionState(BLE_CONNECTED);
        }
        void onDisconnect() {
            tft->setBLEConnectionState(BLE_DISCONNECTED);
        }
    private:
        TFTDisplay *tft;
};
void setup()
{
    ancsService->startServer("CarWatch", 30000);
    ancsService->setClientCallback(new MyClientCallback(display));
}
```

chegewara, 2018-10-02:
Try to add small delay in loop. 
You can always open new issue on my github and we can talk Polish or on priv.

paweb88, 2018-10-02:
@chegewara THX here is ma whole application code: 

https://github.com/paweb88/CarWatch/tree/master/src

maybe in your free time you can look at that and tell me what i doing wrong becouse i'm not a pro C/C++ developer :P

chegewara, 2018-10-04:
@paweb88 Try to change this line (0->1):
https://github.com/nkolban/esp32-snippets/blob/ade8bcd38376b8095d3fee80fec454834ffd232d/cpp_utils/BLEClient.cpp#L99

paweb88, 2018-10-04:
@chegewara after chenage  esp_err_t errRc = ::esp_ble_gattc_app_register(0);  to 1 i have got:

```
abort() was called at PC 0x4017c2c3 on core 0

Backtrace: 0x40090c38:0x3fffac20 0x40090e3b:0x3fffac40 0x4017c2c3:0x3fffac60 0x4017c30a:0x3fffac80 0x4017c3b7:0x3fffaca0 0x400d62ad:0x3fffacc0 0x400d1b7b:0x3fffad2
```

```
0x4017c2c3: __cxa_call_unexpected at ??:?
0x40090c38: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:649
0x40090e3b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c:649
0x4017c2c3: __cxa_call_unexpected at ??:?
0x4017c30a: std::terminate() at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
0x4017c3b7: __cxa_throw at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:86
0x400d62ad: BLEClient::getService(BLEUUID) at ??:?
0x400d1b7b: ANCSService::startClient(void*) at ??:?
```

chegewara, 2018-10-04:
If i remember you cant pass member function as pointer argument:
`::xTaskCreatePinnedToCore(&ANCSService::startClient,`
You have to create plain C function that will be your task and inside it call `ANCSService::startClient()`.

chegewara, 2018-10-05:
> In the Apple Notification Center Service, the GAP peripheral (often a smartwatch) is the GATT client, because the data lies on the iOS device (GAP central and GATT server).