nkolban/esp32-snippets #666: Reading sensor values from notifycallback

Olaclemmy, 2018-10-01:
Hi, please I am new here. I am working on BLE project using two ESP32, one as a client and the other as a server. My issue is how to read sensor values sent to the client inside notifycallback. I have been trying to work it out but with no success. 

All I want is to read sensor reading and display it on Oled LCD.

Please a quick research will be appreciated. 
Thanks.

chegewara, 2018-10-01:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/SampleClient_Notify.cpp#L28-L35

hetlelid, 2018-10-03:
Depending on your datatype, you can access the data inside the notifyCallback() function that @chegewara is referring to in the following way:

`	uint16_t wd = ((uint16_t)pData[2] << 8) | pData[1];
	float temp = wd / 100.0;`

Keep in mind this is a static function. I have tried to move all BLE code into a library for a specific device, but i struggle to update the class variables from this static function. Let me know if anyone have a clue for how to do this.

Olaclemmy, 2018-10-05:
Thanks for your prompt feedback @chegewara, I managed to retrieve it using this method below. It might not be the best way to do it but it works. 

char rxValue[5];
For(int i = 0; i < length; i++)
{
  rxValue[i] = pData[i];
}
rxReading = (((String)rxValue).toFloat());
Serial.println(rxReading);

//Global variable 
float rxReading;

Please, I am not expert in this, but I could receive the same values I was sending from BLE server. 

If there's any improvement on this, I will really appreciate it.

Olaclemmy, 2018-10-05:
I really appreciate your time and effort putting into your explanation, it really helps @hetlelid.

But any improvement on my method would be appreciated.

chegewara, 2018-10-08:
Try to use this macro in notificationCallback() to output log value and paste here results:
`ESP_LOG_BUFFER_HEX("my value", pData, length);`

Olaclemmy, 2018-10-08:
Thanks for your prompt feedback @chegewara, I will definitely do that. My appreciation.

Olaclemmy, 2018-10-08:
Thanks for your help @chegewara, though I couldn't understand how to use what you suggested, but just putting it in my code does the job. Now I can receive true values from notificationCallback().

…..here is my code:

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData, size_t length, bool isNotify) 
  {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    ESP_LOG_BUFFER_HEX("my value", pData, length);
    Serial.print(" of data length ");
    Serial.println(length);
    Serial.print("Received Value: ");
    for (int i = 0; i <length; i++) {
       rxValue[i]= pData[i];
    Serial.print(pData[i]);
    Serial.print(" ");
   }
   fsrValue = (((String)rxValue).toFloat());
}

…..here is serial monitor output.

`BLE Advertised Device found: Name: FSR0, Address: b4:e6:2d:8d:78:b7, serviceUUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, txPower: -21
00:41:51.490 -> haveServiceUUID: 1
00:41:51.490 -> device's UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b
00:41:51.490 -> Found our device!  address: We are in loop.
00:41:51.490 -> We are not connected to the BLE Server yet.
00:41:51.490 -> Forming a connection to b4:e6:2d:8d:78:b7
00:41:51.490 ->  - Created client
 - Connected to server
 - Found our service
00:41:54.536 ->  - Found our characteristic
00:41:54.536 -> The characteristic value was: 0.00
00:41:54.536 -> We are now connected to the BLE Server.
Force: 0.00Kg
00:41:54.583 -> Notifications turned off
We are in loop.
00:41:56.589 -> Force: 0.00Kg
00:41:56.589 -> Notifications turned on
Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data length 4
00:41:57.692 -> Received Value: 48 46 48 48  
We are in loop.
00:41:58.593 -> Force: 0.00Kg
00:41:58.593 -> Notifications turned off
We are in loop.
00:42:00.599 -> Force: 0.00Kg
00:42:00.599 -> Notifications turned on`

chegewara, 2018-10-08:
Looks like you have it working.

Olaclemmy, 2018-10-09:
Yes, thanks for your help @chegewara. I will have to close this issue now,  since my problem is solved. 

Thanks once again for your time. 
Regards.

closed 2018-10-09 by Olaclemmy