nkolban/esp32-snippets #669: Issues sending characteristic notify - stuck on SEMAPHOR -  What am I missing?

wegunterjr, 2018-10-01:
```
D (23428) FreeRTOS: Semaphore taking: name: ConfEvt (0x3ffd4698), owner: <N/A> for notify␛[0m
D (23438) FreeRTOS: Semaphore taken:  name: ConfEvt (0x3ffd4698), owner: notify␛[0m
V (23438) FreeRTOS: >> wait: Semaphore waiting: name: ConfEvt (0x3ffd4698), owner: notify for notify␛[0m
```

Again, seems that I am having issues with the SEMAPHOR?  Am i missing something?

chegewara, 2018-10-02:
There is not much info. More logs would be helpful.
Maybe this is your case:
https://github.com/nkolban/esp32-snippets/issues/628#issuecomment-417798479

wegunterjr, 2018-10-02:
Hmmm.. I am setup as a server, a client has connected to me and is sending Write messages over BLE.
I receive the WiFi messages in my BLE callback message, and I am successfully able to connect to WiFi, but it is getting stuck on a semaphor. 

```
D (71908) BLECharacteristic: >> setValue: length=11, data=20574520474f5420495421, characteristic UUID=0d563a58-196a-48ce-ace2-dfec78acc814␛[0m
D (71928) BLECharacteristic: << setValue␛[0m
D (71928) BLECharacteristic: >> notify: length: 11␛[0m
D (71938) GeneralUtils:      00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f  ----------------␛[0m
D (71938) GeneralUtils: 0000 20 57 45 20 47 4f 54 20 49 54 21                  WE GOT IT!␛[0m
D (71948) FreeRTOS: Semaphore taking: name: ConfEvt (0x3ffd4698), owner: <N/A> for notify␛[0m
D (71958) FreeRTOS: Semaphore taken:  name: ConfEvt (0x3ffd4698), owner: notify␛[0m
V (71968) FreeRTOS: >> wait: Semaphore waiting: name: ConfEvt (0x3ffd4698), owner: notify for notify␛[0m
```

wegunterjr, 2018-10-02:
This is how I am setting up my Task:

```
class bleCharacteristicCallback: public BLECharacteristicCallbacks 
{
	void onWrite(BLECharacteristic* pCharacteristic) 
	{    
		std::string msg = pCharacteristic->getValue();
		ESP_LOGI(LOG_TAG, "BLE received: %s", msg.c_str());
		esp_log_buffer_char(LOG_TAG, msg.c_str(), msg.length());
		esp_log_buffer_hex(LOG_TAG, msg.c_str(), msg.length());
		pCharacteristic->setNotifyProperty(true);
		processMessage(msg, pCharacteristic);
		BLEUUID sentUUID = pCharacteristic->getUUID();
		ESP_LOGI(LOG_TAG, " BLEUUID Sent: %s", sentUUID.toString().c_str());

		pCharacteristic->setValue(" WE GOT IT!");
		pCharacteristic->notify();
		
	}

	void onRead(BLECharacteristic* pCharacteristic) 
	{
		std::string msg = pCharacteristic->getValue();
		ESP_LOGI(LOG_TAG, "BLE received: %s, %i", msg.c_str(), msg.length());
		esp_log_buffer_char(LOG_TAG, msg.c_str(), msg.length());
		esp_log_buffer_hex(LOG_TAG, msg.c_str(), msg.length());
	}
};
 

class MainBLEServer: public Task {
	void run(void *data) {
		ESP_LOGI(LOG_TAG, "Starting BLE work!");

		BLEDevice::init(DEVICE_NAME);
		BLEServer* pServer = BLEDevice::createServer();
        BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT_MITM);

		BLEDevice::setSecurityCallbacks(new MySecurity());
		BLEService* pService = pServer->createService("91bad492-b950-4226-aa2b-4ede9fa42f59");
        
		BLECharacteristic* pCharacteristic = pService->createCharacteristic(
			BLEUUID("0d563a58-196a-48ce-ace2-dfec78acc814"),
			BLECharacteristic::PROPERTY_BROADCAST |
			BLECharacteristic::PROPERTY_READ  |
			BLECharacteristic::PROPERTY_NOTIFY    | 
			BLECharacteristic::PROPERTY_WRITE //|
			//BLECharacteristic::PROPERTY_INDICATE
		);
	
		pCharacteristic->setValue("Hello World!");

		BLE2902* p2902Descriptor = new BLE2902();
		p2902Descriptor->setNotifications(true);
		pCharacteristic->addDescriptor(p2902Descriptor);
        p2902Descriptor->setAccessPermissions(ESP_GATT_PERM_READ_ENCRYPTED | ESP_GATT_PERM_WRITE_ENCRYPTED);
		pCharacteristic->setCallbacks(new bleCharacteristicCallback());
		pService->start();

		BLEAdvertising* pAdvertising = pServer->getAdvertising();
		pAdvertising->addServiceUUID(BLEUUID(pService->getUUID()));

		BLESecurity* pSecurity = new BLESecurity();

		pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_MITM_BOND);
		pSecurity->setCapability(ESP_IO_CAP_OUT);
		pSecurity->setKeySize();
		pSecurity->setInitEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);

		pAdvertising->start();
		ESP_LOGI(LOG_TAG, "Advertising started!");
	}
};

```
In the main, I am starting up the BLE task:

```
extern "C" 
{
	void app_main();
}

void app_main(void)
{
	ESP_LOGI(LOG_TAG, "Starting!");
	//esp_log_level_set("*", ESP_LOG_DEBUG);
	MainBLEServer* pMainBleServer = new MainBLEServer();
	pMainBleServer->setStackSize(65535);
	pMainBleServer->start();

	ESP_LOGI(LOG_TAG, "After BLE RAM left %d", esp_get_free_heap_size());

} // app_main
```

wegunterjr, 2018-10-02:
Whenever I have the notify set, it gets stuck in the SEMAPHOR like above:
```

pCharacteristic->setValue(" WE GOT IT!");
pCharacteristic->notify();
```

wegunterjr, 2018-10-04:
@nkolban any suggestions?

wegunterjr, 2018-10-04:
alright. Looking there now.
While I read that, can you consider these questions:
Am I doing something wrong with memory?
I create the BLE Task, and give it a very large stack size, but then I am calling it from a BLE Callback EventHandler -  Am i crossing memory somewhere?  Should I do another global for access in other parts?

chegewara, 2018-10-05:
Before i fix it you can try this code:
https://github.com/nkolban/esp32-snippets/issues/656#issuecomment-427073012

wegunterjr, 2018-10-07:
trying it now....

wegunterjr, 2018-10-07:
so far, so good.  I can set a value, and notify - doesn't get stuck and disconnect.

chegewara, 2018-10-07:
I tested it over 20 hours with 1-3 clients connected, every client sent write request every 50 ms and server on on write sent notification to each client. Ive got no issue, clients could connect and disconnect in mean time. 

From my observation i can say its only reliable solution and its not semaphore fault.

wegunterjr, 2018-10-07:
oh wait - I was testing with commenting out the wait in the notify function.   was that the wrong test?

chegewara, 2018-10-07:
I think it does not matter. I did tests with normal, not commented semaphore.wait. Here is my current repository i am working on with few enhancements. It will be merged here when i decide that passed all my tests.
https://github.com/chegewara/esp32-snippets-enchancements-test

wegunterjr, 2018-10-07:
so, it worked without commenting the 
	// m_semaphoreConfEvt.take("notify");
I am only able to succeed if this is commented out.  I am using v3 of the library and not master, so that might make a difference also.

chegewara, 2018-10-07:
here you can see latest logs before my esp32 lost power from usb:
```
I (80333672) SampleClient: 8--> 7392
E (80333710) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot com3: 80333.615981 length 34
I (80333726) SampleClient: 8--> 7392
E (80333770) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot com3: 80333.668192 length 34
I (80333778) SampleClient: 8--> 7392
E (80333800) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot com3: 80333.720985 length 34
I (80333831) SampleClient: 8--> 7392
E (80333860) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot com3: 80333.773192 length 34
```
over 80000 seconds runtime.

chegewara, 2018-11-15:
Should be fixed in features branch.

wegunterjr, 2018-11-18:
I just tried the feature branch, but some of the updates include esp-idf library updates that I am not using. I am in 3.30101.0 (3.1.1) esp-idf.
Does the features update that fixes this require the newest library?  or can i just bring in the changes in BLECharacteristic.cpp?

chegewara, 2018-11-18:
I am working on master esp-idf, but if there is some issue with this branch against esp-idf v3.1 then let me know and i will see what i can do.