nkolban/esp32-snippets #673: semaphore wait is causing stall in app

chegewara, 2018-10-04:
There is few possible configurations that will cause app stall because of semaphore.wait. Most often is read/write characteristic from within notificationCallback. This is caused because notification/indication procedure is waiting for confirmation, but in mean time we want to request read/write from peer device and 2 semaphores block each other.

Workaround to this situation is to create new task from within notificationCallback that will request read/write. This task will be executed right after notificationCallback ends. We have to remember that we cant in this case to call vTaskDelay/delay after creating such task, because this will cause semaphores blocking. 
I am working now on few upgrades and bugfixes but this is also on my list TODO.

wegunterjr, 2018-10-05:
> Its hard to say what kind of confirmation it is, it can be confirmation
> from bt stack that notification has been send. The fact is that event CONF
> is triggered.

Yeah... There is a take and a wait for the notify.  I commented this out
but having stack overflow issues so an not sure if that helped.

chegewara, 2018-10-06:
Ok, with some more tests i can for sure say that this semaphore. wait is not suppose to be in here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L572

It not only causes issues with stal, it also causing other issues. I am testing now enhancement with one esp32 server and 3 esp32 clients connected. Each client writes to server characteristic and when characteristic is write then server sends notification to every connected client. I have 2xesp32 that are writing every 250ms and one esp32 writing every 500ms and without that semaphore.wait i have no issue, but with that line some notifications are dropped. 

```
I (738524) SampleClient: 8--> 7460
E (738524) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot: 430.216617 length 27
E (738614) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot com3: 738.438042 length 32
E (738614) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot com9: 513.238381 length 32
I (738774) SampleClient: 8--> 7460
E (738854) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot com3: 738.688038 length 32
E (738854) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot com9: 513.488417 length 32
E (739004) SampleClient: Notify callback for characteristic beb5483e-36e1-4688-b7f5-ea07361b26a8 of data Time since boot: 430.71851788417 length 27
I`

toxuin, 2018-10-08:
Implementing a keyboard with a notification going out on every key press, I've noticed the .notify() takes quite a long time. In fact so long that it introduces a noticeable delay after a key has been pressed. Rapidly hitting a key 3 times in a row only registers 2 keypresses with bluetooth host. Getting rid of that semaphore wait seems to have fixed the problem (discovered that on my own by trial and error, then found this thread – means I am not crazy?). This is happening on a single-core device with lots of other tasks running.

chegewara, 2018-10-08:
@toxuin I have 2 questions, do you have input->notify() in separate task? and what is your Tick rate in menuconfig->FreeRTOS?

toxuin, 2018-10-08:
@chegewara yes, it's a task that does very little apart from
```
inputCharacteristic->setValue(report, sizeof(report));
inputCharacteristic->notify();```
Tick rate is default(?) 100hz.

chegewara, 2018-10-08:
@toxuin Now with tick rate 100, do you have vTaskDelay() in it and if yes do you have port_TICK_PERIOD_MS to convert ticks to ms? Also try to create notify task with higher priority than other tasks if its possible. I can send message from my example code without any delays between characters.

toxuin, 2018-10-08:
Bluetooth task in my code waits on a queue receive, so no task delays.
Thanks, Che! I will set the priority and see if that helps. Just to confirm: is it ok to remove that semaphore wait on notify?..

chegewara, 2018-10-09:
You can try, i didnt encounter any issues when i commented it out.

chegewara, 2018-10-13:
discussion moved here
https://github.com/nkolban/esp32-snippets/issues/681

closed 2018-10-13 by chegewara