nkolban/esp32-snippets #676: Issues with SampleServer_authentication_passkey

JasXSL, 2018-10-09:
### Issue 1

The example is missing onConfirmPIN. I copied it over from another example:
<pre>
bool onConfirmPIN(uint32_t pass_key){
        ESP_LOGI(LOG_TAG, "The passkey YES/NO number:%d", pass_key);
	vTaskDelay(5000);
	return true;
}
</pre>

### Issue 2 

After compiling with the change above it works. But after restarting the ESP and disconnecting my phone bluetooth for a while, it refuses to pair again and I have to manually delete the pairing from my phone and enter the key again.

### Issue 3

The example has the following:
<pre>
uint32_t onPassKeyRequest(){
        ESP_LOGI(LOG_TAG, "PassKeyRequest");
	return 123456;
}
</pre>

Am I right in assuming this is meant to allow me to specify my own passkey? This method is never run however, and the code is always randomly generated.

I modified the example slightly to compile with arduino and a single script. Here's a gist of the complete code I'm using:
https://gist.github.com/JasXSL/1dda802a24b4f6cfe742c222e9958142

I'm compiling with an up to date library from here:
https://github.com/espressif/arduino-esp32

chegewara, 2018-10-09:
This example is a little bit outdated, thats why is missing onConfirmPIN(),

This method is called when you are requested to input pin on esp32:
```
uint32_t onPassKeyRequest(){
        ESP_LOGI(LOG_TAG, "PassKeyRequest");
	return 123456;
}
```
In that callback you need to read digits from attached numeric pad or if you are connecting to peer device with hardcoded pin return it (less likely).

If you want to display pin code on esp32 and input it on peer device then you need to use this callback:
```
void onPassKeyNotify(uint32_t pass_key){
          ESP_LOGE(LOG_TAG, "The passkey Notify number:%d", pass_key);
          passKey = pass_key;
 }
```

JasXSL, 2018-10-09:
That clears up 1 & 3. Any idea why #2 might be happening? I don't mind plugging the ESP into my computer and getting the pin for pairing that way, but I'd like the pairing to stick even if the ESP is powered down for a while.

chegewara, 2018-10-09:
`pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_ONLY);`
change to:
`ESP_LE_AUTH_REQ_SC_BOND`

JasXSL, 2018-10-09:
Changed it. It successfully reconnects now, but all read/write operations time out. I'll reset everything and see if that resolves it.

JasXSL, 2018-10-09:
Same issue, after being disconnected a couple of minutes, I can connect to it. But all read/write operations time out. Some further specs if they can be of help:

|Item|Value|
|---|---|
|Boards|Sparkfun ESP32 Thing & NodeMCU|
|Arduino|Hourly build 2018-08-20|
|Software|nRF Connect on Android 6|

JasXSL, 2018-10-09:
Looks like it might have been an issue with my partition tables. Gonna investigate tomorrow and see if I can reproduce it.

JasXSL, 2018-10-10:
Yep, looks like it was an issue with my partition tables. If anyone runs into similar issues:

1. Overwrote my C:\Users\<username>\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0 data with https://github.com/espressif/arduino-esp32
2. Set my board in arduino IDE to ESP32 Dev Module.
3. Set my partition scheme to No OTA to fit the sketch.

I've updated the gist and it seems to be fully functioning now. Thanks!

closed 2018-10-10 by JasXSL