nkolban/esp32-snippets #678: BLE Scan (continuous), RSSI

m-menon, 2018-10-10:
hi, thanks for the BLE scan program. A couple of quick questions if I may

1) How do I keep the scan going continuously? I can loop, call the scan again etc but I was thinking if there is some value that can be passed to the scan to keep it scanning continuously?
2) Is is possible to get the RSSI of the beacons as those are scanned? 

I am referring to this example
```
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 30; //In seconds

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

chegewara, 2018-10-10:
1. scan(0),
2. https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L95

m-menon, 2018-10-10:
@chegewara, fantastic ... thanks! :100:

closed 2019-04-27 by m-menon

scploeger, 2019-10-30:
> 1. scan(0),
> 2. https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertisedDevice.cpp#L95

@chegewara can you clarify your response regarding 1. from above?
Maybe I am just not understanding what you are saying, but what do you mean by scan(0)? How would the originally quaoted code be changed to scan continously?

chegewara, 2019-10-31:
What you dont understand?
```
int scanTime = 30; //In seconds
BLEScanResults foundDevices = pBLEScan->start(scanTime);
```
Scan until esp32 wont crash, die or whatever.

scploeger, 2019-10-31:
@chegewara Firstly I apologize if these are dumb questions, I am trying my best to learn from this.
Will setting scantime to 0 just repeadtly loop my callback?

After have just tried this it appears this is what is happening. But then my scans still seems very slow. I have set my scan interval to 100 and my scan window to 95, but I am not acheiving scans nearly that fast. Any ideas of anything else I can try to acheive the orginally mentioned "continous" scanning?
Thanks

chegewara, 2019-10-31:
Its not dumb question, about 2 years ago when i started with esp32 i didnt know what is button debouncing.

There is many factors that impacts how fast you discover devices. Advertising setting on the other devices for example.
If you dont worry about power consumption and dont use wifi then you can setup interval and window with the same very short or longer value(just do some tests).

scploeger, 2019-10-31:
@chegewara thank you for your reply. I guess I have one more conceptual question then, now that I have got the continuous scanning to work! :-).

While continuously scanning, new beacons are discovered and come up as scan results which is the intended purpose. When will these beacons get scanned again?

For example, if I am continuously scanning and scan 5 beacons, their info is printed as they are discovered. Now what? Will the ESP just continue to print the results of scanned beacons as it comes across them, in a sort of unpredictable, random order?

chegewara, 2019-10-31:
Here is the answer:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig#L199-L204
I didnt test it by myself, but as far as i understand esp-idf will discard duplicates by address up to 20 devices.

Here is API you may want for your next question:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEScan.cpp#L204-L252

scploeger, 2019-11-02:
Thank you for the reply @chegewara 

I am trying to scan bluetooth beacons, and continuously get the rssi from nearby beacons. So I have decided that I would like to turn on duplicate scanning in hope that this will help me with this goal. I will set "CONFIG_BTDM_SCAN_DUPL_TYPE" off:
(https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/sdkconfig#L199)

I am currently using the Arduino framework, but I know that I need to use the esp-idf framework. In my IDE (PlatformIO) I have set my framework to esp-idf. I am getting some errors about missing .h files from the BLE library, but I will try to figure that out.
Once I figure this issue out, should I just be able to go into the menuconfig.h file and change the CONFIG_BTDM_SCAN_DUPL_TYPE" parameter?

Thanks again for all the help.

chegewara, 2019-11-02:
This will work
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEScan.cpp#L173

alviyandi, 2020-03-12:
ble scan maximum can scan how much?

Le-Ze, 2022-01-01:
Hello.
I'm solving the same problem.
When I set int scantime = 0 void loop doesn't work.

https://github.com/nkolban/esp32_ble_arduino/blob/master/examples/ble_scan/ble_scan.ino
int scantime = 30;
Void Loop waits for 30 seconds to the scanning.
How do they do to work alone?
Maybe through the thread?