nkolban/esp32-snippets #680: ble example requests

chegewara, 2018-10-11:
I am having some spare time now and changes in ble code i am doing are looking good. If some of you could have some special but simple example code request i can think about write one. I also want to write some wiki about ble, but if someone would like to help me with that that would be nice:
https://github.com/chegewara/esp32-snippets/wiki

Milan991, 2018-10-23:
@chegewara I've some spare time now also. I've experience with BLE/WiFi and ESP32 using ESP-IDF, so if you can think of something that I can do, tell me. Maybe some BLE example with custom services? Read, Write, Notify example?

chegewara, 2018-10-24:
@Milan991 If you have spare time and willing to help, there is few examples that require to be checked because few of them can be outdated. Thanks for help.

Milan991, 2018-10-24:
@chegewara Please tell me where I could find that examples. I'm gona test them, no problem.

chegewara, 2018-10-24:
esp-idf examples:
https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests
arduino examples:
https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils/tests/BLETests/Arduino

If you wish to do some more tests i am doing upgrades and adding new features here (will be merged later):
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

There is few more example in that repo:
- ble + wifi and few more

Milan991, 2018-10-24:
Thank you. I will notify you about progress.

Milan991, 2018-10-25:
@chegewara I just tested Sample-MLE-15.cpp, and I keep getting this error and rebooting all the time  . It is the same with Sample1.cpp, just without reboot.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6088
load:0x40078000,len:10084
load:0x40080400,len:6552
entry 0x40080764
I (29) boot: ESP-IDF v3.2-dev-1385-g129d32772-dirty 2nd stage bootloader
I (29) boot: compile time 21:01:46
I (30) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed      : 40MHz
I (40) boot: SPI Mode       : DIO
I (44) boot: SPI Flash Size : 4MB
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (74) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3c3e8 (246
760) map
I (181) esp_image: segment 1: paddr=0x0004c410 vaddr=0x3ff80000 size=0x00000 (
   0) load
I (181) esp_image: segment 2: paddr=0x0004c418 vaddr=0x3ff80000 size=0x00000 (
   0) load
I (187) esp_image: segment 3: paddr=0x0004c420 vaddr=0x3ffbdb60 size=0x02dbc ( 1
1708) load
I (201) esp_image: segment 4: paddr=0x0004f1e4 vaddr=0x3ffc091c size=0x00000 (
   0) load
I (205) esp_image: segment 5: paddr=0x0004f1ec vaddr=0x40080000 size=0x00400 (
1024) load
0x40080000: _WindowOverflow4 at D:/Milan/Programi/esp-idf/components/freertos/xt
ensa_vectors.S:1685

I (215) esp_image: segment 6: paddr=0x0004f5f4 vaddr=0x40080400 size=0x00a1c (
2588) load
I (224) esp_image: segment 7: paddr=0x00050018 vaddr=0x400d0018 size=0xa3734 (66
9492) map
0x400d0018: _flash_cache_start at ??:?

I (466) esp_image: segment 8: paddr=0x000f3754 vaddr=0x40080e1c size=0x117bc ( 7
1612) load
I (497) esp_image: segment 9: paddr=0x00104f18 vaddr=0x400c0000 size=0x00000 (
   0) load
I (497) esp_image: segment 10: paddr=0x00104f20 vaddr=0x50000000 size=0x00000 (
    0) load
I (503) esp_image: segment 11: paddr=0x00104f28 vaddr=0x50000000 size=0x00000 (
    0) load
I (523) boot: Loaded app from partition at offset 0x10000
I (523) boot: Disabling RNG early entropy source...
I (524) cpu_start: Pro cpu up.
I (528) cpu_start: Starting app cpu, entry point is 0x4008104c
0x4008104c: call_start_cpu1 at D:/Milan/Programi/esp-idf/components/esp32/cpu_st
art.c:245

I (520) cpu_start: App cpu up.
I (538) heap_init: Initializing. RAM available for dynamic allocation:
I (545) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (551) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (557) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (563) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (569) heap_init: At 3FFCAAE8 len 00015518 (85 KiB): DRAM
I (576) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (582) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (588) heap_init: At 400925D8 len 0000DA28 (54 KiB): IRAM
I (595) cpu_start: Pro cpu start user code
I (278) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (279) Sample-MLE-15: MLE-15 sample starting
I (359) BTDM_INIT: BT controller compile version [b4633da]
I (359) system_api: Base MAC address is not set, read default base MAC address f
rom BLK0 of EFUSE
**E (369) BT_LOG: Conroller not initialised

E (369) BLEDevice: esp_bluedroid_init: rc=259 ESP_ERR_INVALID_STATE
E (379) BLEClient: esp_ble_gattc_app_register: rc=259 ESP_ERR_INVALID_STATE
E (389) BLEClient: esp_ble_gattc_search_service: rc=259 ESP_ERR_INVALID_STATE**
abort() was called at PC 0x400e6487 on core 0
0x400e6487: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.bui
ld/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112

Backtrace: 0x4008fc60:0x3ffbb300 0x4008fe8d:0x3ffbb320 0x400e6487:0x3ffbb340 0x4
00e64ce:0x3ffbb360 0x400e4fef:0x3ffbb380 0x400fef5d:0x3ffbb3a0 0x400d3eab:0x3ffb
b400 0x400d3fa3:0x3ffbb490 0x400d3d77:0x3ffbb4b0 0x400d213a:0x3ffbb4d0 0x4008dba
5:0x3ffbb4f0
0x4008fc60: invoke_abort at D:/Milan/Programi/esp-idf/components/esp32/panic.c:6
80

0x4008fe8d: abort at D:/Milan/Programi/esp-idf/components/esp32/panic.c:680

0x400e6487: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.bui
ld/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112

0x400e64ce: std::terminate() at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/li
bstdc++-v3/libsupc++/eh_terminate.cc:112

0x400e4fef: __cxa_throw at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc
++-v3/libsupc++/eh_throw.cc:87

0x400fef5d: BLEClient::getService(BLEUUID) at D:/Milan/Files/Projects/test/compo
nents/BLEClient.cpp:389

0x400d3eab: run() at D:/Milan/Files/Projects/test/main/Sample-MLE-15.cpp:65

0x400d3fa3: Sample_MLE_15() at D:/Milan/Files/Projects/test/main/Sample-MLE-15.c
pp:87

0x400d3d77: app_main at D:/Milan/Files/Projects/test/main/main.cpp:29

0x400d213a: main_task at D:/Milan/Programi/esp-idf/components/esp32/cpu_start.c:
493

0x4008dba5: vPortTaskWrapper at D:/Milan/Programi/esp-idf/components/freertos/po
rt.c:403

Rebooting...

chegewara, 2018-10-25:
There is a lo issues reporting this error:
`E (369) BT_LOG: Conroller not initialised`

Looks like something changed in esp-idf. Thanks, will investigate it later.

Milan991, 2018-10-25:
I'm glad I can help. If I can do anything else, say it, I will look what can be the problem also.

chegewara, 2018-10-25:
@Milan991 Yes, it would be nice and helpful if you could investigate this issue.

Milan991, 2018-10-25:
@chegewara I found out what issue was, there was no `esp_bt_controller_enable(ESP_BT_MODE_BLE);` inside `#ifndef CLASSIC_BT_ENABLED`. Updated BLEDevice.cpp is pasted below, there is no more error from above. I'll continue with testing.

`/*
 * BLE.cpp
 *
 *  Created on: Mar 16, 2017
 *      Author: kolban
 */
#include "sdkconfig.h"
#if defined(CONFIG_BT_ENABLED)
#include <freertos/FreeRTOS.h>
#include <freertos/event_groups.h>
#include <freertos/task.h>
#include <esp_err.h>
#include <nvs_flash.h>
#include <esp_bt.h>            // ESP32 BLE
#include <esp_bt_device.h>     // ESP32 BLE
#include <esp_bt_main.h>       // ESP32 BLE
#include <esp_gap_ble_api.h>   // ESP32 BLE
#include <esp_gatts_api.h>     // ESP32 BLE
#include <esp_gattc_api.h>     // ESP32 BLE
#include <esp_gatt_common_api.h>// ESP32 BLE
#include <esp_err.h>           // ESP32 ESP-IDF
#include <esp_log.h>           // ESP32 ESP-IDF
#include <map>                 // Part of C++ Standard library
#include <sstream>             // Part of C++ Standard library
#include <iomanip>             // Part of C++ Standard library

#include "BLEDevice.h"
#include "BLEClient.h"
#include "BLEUtils.h"
#include "GeneralUtils.h"
#ifdef ARDUINO_ARCH_ESP32
#include "esp32-hal-log.h"
#include "esp32-hal-bt.h"
#endif

static const char* LOG_TAG = "BLEDevice";

/**
 * Singletons for the BLEDevice.
 */
BLEServer* BLEDevice::m_pServer = nullptr;
BLEScan*   BLEDevice::m_pScan   = nullptr;
BLEClient* BLEDevice::m_pClient = nullptr;
bool       initialized          = false;   // Have we been initialized?
esp_ble_sec_act_t	 BLEDevice::m_securityLevel = (esp_ble_sec_act_t) 0;
BLESecurityCallbacks* BLEDevice::m_securityCallbacks = nullptr;
uint16_t   BLEDevice::m_localMTU = 23;
BLEAdvertising* BLEDevice::m_bleAdvertising = nullptr;

/**
 * @brief Create a new instance of a client.
 * @return A new instance of the client.
 */
/* STATIC */ BLEClient* BLEDevice::createClient() {
	ESP_LOGD(LOG_TAG, ">> createClient");
#ifndef CONFIG_GATTC_ENABLE  // Check that BLE GATTC is enabled in make menuconfig
	ESP_LOGE(LOG_TAG, "BLE GATTC is not enabled - CONFIG_GATTC_ENABLE not defined");
	abort();
#endif  // CONFIG_GATTC_ENABLE
	m_pClient = new BLEClient();
	ESP_LOGD(LOG_TAG, "<< createClient");
	return m_pClient;
} // createClient

/**
 * @brief Create a new instance of a server.
 * @return A new instance of the server.
 */
[... 452 lines cut ...]
uint16_t BLEDevice::getMTU() {
	return m_localMTU;
}

bool BLEDevice::getInitialized() {
	return initialized;
}

BLEAdvertising* BLEDevice::getAdvertising() {
	if (m_bleAdvertising == nullptr) {
		m_bleAdvertising = new BLEAdvertising();
		ESP_LOGI(LOG_TAG, "create advertising");
	}
	ESP_LOGD(LOG_TAG, "get advertising");
	return m_bleAdvertising;
}

void BLEDevice::startAdvertising() {
	ESP_LOGD(LOG_TAG, ">> startAdvertising");
	getAdvertising()->start();
	ESP_LOGD(LOG_TAG, "<< startAdvertising");
} // startAdvertising

#endif // CONFIG_BT_ENABLED
`

closed 2019-06-21 by chegewara