nkolban/esp32-snippets #681: BLE semaphores - a broader discussion

mws-rmain, 2018-10-12:
I have some issues with semaphore use in the BLE classes, and thought I'd open a discussion here (or a Neil might say flesh out 'the story'). I fully expect that many of the defined semaphores are required for reasons that I've not thought of or encountered, and if so, I'd like to expand my understanding. 

Broadly speaking, it appears that calls to functions of the esp-idf Bluetooth stack create internal 'events' that are posted to handler queues to be serviced by internal tasks. I can see the need for semaphore use to synchronize 'app code' with the underlying BT stack tasks (for instance to ensure a 'service' has actually been created, before allowing the 'application code' to continue with adding characteristics to it, etc.).

However, using a semaphore in BLECharacteristic to 'stall' application code that calls 'indicate()' until an  ESP_GATTS_CONF_EVT shows a connected client confirmed the change was successfully received (via 'm_semaphoreConfEvt.wait()' seems pointless to me, unless you plan on spawning a new thread to update attribute changes for each & every attribute (?). If the indicate was not yet acknowledged, and a subsequent indicate was called on the same attribute, calling 'm_semaphoreConfEvt.wait()' just before 'm_semaphoreConfEvt.take() would stall the calling thread just as easily, and this way would allow a single 'update thread' to update multiple changed attributes (possibly on a timer), with a simple loop:
```
while(1) {
  if (attribute_a_changed == true) {
    pAttribA->setValue( (uint8_t *)&attribA, sizeof(attribA) );
    pAttribA->indicate();  // send indicate to any clients that have enabled it
    pAttribA->notify();   // send notify to any clients that have enabled it
  }
  if (attribute_b_changed == true) {
    pAttribB->setValue( (uint8_t *)&attribB, sizeof(attribB) );
    pAttribB->indicate();  // send indicate to any clients that have enabled it
    pAttribB->notify();   // send notify to any clients that have enabled it
  }
  ...
  vTaskDelay( pdMS_TO_TICKS(200) );
}
```

I'm aware that Neil raised the issue of ESP_GATTS_CONF_EVT being raised on notify() [#749](https://github.com/espressif/esp-idf/issues/749), and the response that "it's a feature of our system". Regardless, since BLE defines notify() as equivalent to UDP (vs TCP), I'd make the case that there is certainly NO valid reason for a semaphore against it.

Note that  @chegewara is currently working on multi-device connections, which will make all of this semaphore handling much more complicated, when it _IS_ legitimate, so taking some time to do some thoughtful cleanup now would be very helpful.

mws-rmain, 2018-10-12:
I think the first issue is one I also encountered, and determined a fix for - that I have not yet posted.
The issue must be checked for ALL places where semaphores are used. Back when I did, I found there were numerous places with various semaphores that the same issue occurs. Surprising, but I found some locations where the issue _was_ properly addressed.

All ERROR PATHS that exit from functions that allocate a semaphore MUST invoke '.give()' on the semaphore before exiting. Otherwise, the semaphore remain blocked. This can cause all manner of problems, but there are numorous cases where an error exit path exits that does not release a held semaphore. For example, in BLECharacteristic::indicate():
```
	m_semaphoreConfEvt.take("indicate");

	esp_err_t errRc = ::esp_ble_gatts_send_indicate(
			getService()->getServer()->getGattsIf(),
			getService()->getServer()->getConnId(),
			getHandle(), length, (uint8_t*)m_value.getValue().data(), true); // The need_confirm = true makes this an indication.

	if (errRc != ESP_OK) {
		m_semaphoreConfEvt.give();  // RELEASE semaphore before exiting...
		ESP_LOGE(LOG_TAG, "<< esp_ble_gatts_send_indicate: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
		return;
	}
	m_semaphoreConfEvt.wait("indicate");
```

@chegewara the [lines](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L561-L570) you pointed to are in notify(), and I don't believe the semaphore should even exist in notify().

In my case, requests to indicate() &/or notify() would sometimes fail - generally due to 'congested channel'. I believe a recent esp-idf update has addressed allot of these problems by updating their (closed source) BT Stack. The semaphore would still get cleared eventually, because of the indescriminate way ESP_GATTS_CONF_EVT cleared m_semaphoreConfEvt of ALL attributes when ANY confirmation event was triggered.

chegewara, 2018-10-12:
My lines with notify code it was just example, but i agree with all. 

I think ive done with multi connect code for some time, now it will require more tests so i could fix introduced bugs. I meantime we can coop and try to do something with semaphores. I am available here and on gitter.

chegewara, 2018-10-14:
@mws-rmain 
Could someone confirm my observations? Because we are working with bluetooth, and by default all bt tasks are pinned to core 0. This ble library is also build on tasks, but we by default are using `tskNO_AFFINITY`. I have strange feelings that we are having so many troubles with semaphores because of that.
It can be tested in two ways:
- change in menuconfig to run on one core only,
- change value in Task.cpp to use core 0 instead `tskNO_AFFINITY`.

mws-rmain, 2018-11-05:
Thanks @chegewara. I haven't had much time to work on this recently. I had taken a different approach, grouping instances of semaphores into three classes, and ensuring appropriate handling for each class. Your approach (getting rid of them entirely) would be greatly preferred - Well done!

There are other semaphores that need a bit of cleanup (from my investigation). I'll post details when I can.

RE: the final remaining semaphore for indicate; I had raised an [issue](https://github.com/espressif/esp-idf/issues/2552) against esp-idf to include the associated attribute handle in the structure passed with ESP_GATTS_CONF_EVT. This should allow for much more efficient/intelligent handling of the event rather than the current 'release everything' shotgun. I'm unsure if this change is implemented in the latest build of esp-idf yet.

chegewara, 2018-11-05:
> This should allow for much more efficient/intelligent handling of the event rather than the current 'release everything' shotgun

I already did some cleanup in this matter. I had to do it because of multi connection feature, and now it looks very promising. I have some issues with BLEAdvertising and i have no idea what should i do, besides a lot changes in code that should make it run faster and without weird crashes because of uninitialized pointer or something like that. I will try to work in next few days at least few hours a day so should be better.

I received iPhone from our community member and i will try to find issues with HID library and next step will be to rewrite it to make it even easier to use.

mws-rmain, 2018-11-05:
Verified the attribute handle was added to ESP_GATTS_CONF_EVT 21 days ago ( [44827bb](https://github.com/espressif/esp-idf/commit/44827bb681d9c3923dd33a9ad5b8459613e017bd) ).

@chegewara have you commited your changes to your github 'esp-snippets-enhancements-test' repo, or are they still too raw?

chegewara, 2018-11-05:
@mws-rmain Not yet, i have few tests to do and i can add now this handle to conf_evt too. For sure there still be a lot to do, but at least more eyes can find bugs and issues faster.

chegewara, 2018-11-05:
Ok, i dont know if i didnt mess something with git push thing, but here is new PR:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

chegewara, 2018-11-05:
@mws-rmain i just pulled master commit and still have thise error:
`error: 'struct esp_ble_gatts_cb_param_t::gatts_conf_evt_param' has no member named 'handle'`, so your esp-idf issue is not merged yet to master.

mws-rmain, 2018-11-06:
very odd.
I re-sync'd to master & physically verified  in
esp-idf\components\bt\bluedroid\api\include\api\esp_gatts_api.h, lines
117-127:
-------------
    /**
     * @brief ESP_GATTS_CONF_EVT
     */
    struct gatts_conf_evt_param {
        esp_gatt_status_t status;       /*!< Operation status */
        uint16_t conn_id;               /*!< Connection id */
        uint16_t handle;                /*!< attribute handle */
        uint16_t len;                   /*!< The indication or notification
value length, len is valid when send notification or indication failed */
        uint8_t *value;                 /*!< The indication or notification
value , value is valid when send notification or indication failed */
    } conf;                             /*!< Gatt server callback param of
ESP_GATTS_CONF_EVT (confirm) */
-------------

maybe do a 'make clean' then rebuild?

chegewara, 2018-11-06:
Ok, my bad. I did git pull but it failed.

chegewara, 2018-11-08:
@mws-rmain We have handle in conf event, but it seems to be bugged. 
`W (22205) handleGATTServerEvent: m_handle = 42, conf->handle = 0`