nkolban/esp32-snippets #684: Size of BLECharacteristic

Bascy, 2018-10-14:
Hi

I've noticed that adding a new BLECharacteristic takes at least 600 bytes of memory.
As we are planning to create a lot of those (around 80), including the Descriptors that will add up to over 100K of heap space needed. We are afraid that this will cause memory problems ..

Is there a way to reduce the footprint of the Characteristic? Are there any parts that are "optional"?
Does anybody have any other suggestions ?

chegewara, 2018-10-14:
80 characteristics? O.O

First suggestion is to use wrover with 4MB psram. Building such huge project, i am assuming that you gonna work with esp-idf, not arduino-ide. In menuconfig you will find options to allocate memory for ble from psram. 
Turning off all debugging options also can help. In meantime i can see what i can do with memory usage, but i cant promise anything.

Because you are just starting with project i am obligated to say that library still have some bugs and memory leaks. Last few days i am working hard on bug fixing and adding few features.

Bascy, 2018-10-14:
Its always very good to hear that improvements are being made on a library we're using :)
We are already using  esp32's with 4Mb of memory, but as you already realized it is a huge project.
We've upgraded from ESP2688 to ESP32 now, using Eclipse (with Sloeber plugin) and the arduino-esp32 core because that is the easiest way to transform from ESP2688 to ESP32 we think. And we also have a Arduino Mega in the same project which is also coded though Eclipse and an Arduino core.

I'm not quite sure what you mean by "using esp-idf". Can you elaborate and enlighten us on the advantages of that?

chegewara, 2018-10-14:
Arduino-esp32 is precompiled set of libraries (components) on top of this primary library.
https://github.com/espressif/esp-idf

esp-idf is (i dont know how to say) adaptation? of FreeRTOS operating system.
Its worth to know at least basics about freertos if you plan to do big projects:
http://www.freertos.org/FreeRTOS-Plus/index.shtml
https://www.freertos.org/Documentation/161204_Mastering_the_FreeRTOS_Real_Time_Kernel-A_Hands-On_Tutorial_Guide.pdf

Also if you dont mind to use this library with updates, but that are not widely tested yet, you can get it from here:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

Bascy, 2018-10-14:
We are actually using the multitasking and mutlithreading functionality of the ESP32, while using arduino-esp32 core. The core already exposes a lot functions to handle that. 

Whats is the difference with programming the ESP without using the arduino-esp32 core? Can i still do that in Eclipse? is it still C(++)? Can I keep using all the other libraries we use?

chegewara, 2018-10-14:
Yes, you can use eclipse.
core library is plain C, so yes, you can write C++. As you can see this library is C++.

About libraries, i dont have knowledge to tell you yes you can. But you can try to search about `esp32 arduino as compnent`. Here is top link from google:
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

Bascy, 2018-10-14:
> Problem with arduino-esp32 is that some parameters are set and you dont have control over them.
Can you tell me which parameters?

Bascy, 2018-10-17:
I've browsed through the code of a BLECharacteristics and it strikes me that you regularly use std::map structures. As I understand it these are very convenient in terms of functionality, but also notorious memory hogs ;) Maybe they can be replaced by more economical structures like std::set or list?

chegewara, 2018-10-17:
> As we are aiming for this project to be commercially successfull i'm a bit hesitant to provide any more details on a public thread

No problem, just curiosity what project is requiring 80 characteristics.

I am not C++ specialist, i just found this project about year ago and since then i learned some about coding in C++ and a lot about ble. If you think that std::map is not efficient or other data type would be better you can suggest something and we will see.

Bascy, 2018-10-18:
Found this, might be a bit dated but the differences are huge!
http://jsteemann.github.io/blog/2016/06/14/how-much-memory-does-an-stl-container-use/

chegewara, 2018-10-18:
I did research about map vs set. We are using map because we need to have access to element like service or characteristic with specific key like uuid or handle. This is main reason we cant use set. Of course we could use set with only service or characteristic in it, but we would need to loop thru set to find exactly the one we need. 
In your case our goal is n = 80, this is close enough:
```
n =    64
---------
vector         =>   512 bytes allocd at end,  total:  1016 bytes mallocd,    7 malloc(s), 6 free(s)
map            =>  3072 bytes allocd at end,  total:  3072 bytes mallocd,   64 malloc(s), 0 free(s)
set            =>  2560 bytes allocd at end,  total:  2560 bytes mallocd,   64 malloc(s), 0 free(s)
unordered_map  =>  2312 bytes allocd at end,  total:  3016 bytes mallocd,   70 malloc(s), 5 free(s)
unordered_set  =>  1800 bytes allocd at end,  total:  2504 bytes mallocd,   70 malloc(s), 5 free(s)
deque          =>  1088 bytes allocd at end,  total:  1088 bytes mallocd,    3 malloc(s), 0 free(s)
```
If it makes any difference (700 bytes for 64 elements, very rare case) i would say we could switch to unordered_map.

Bascy, 2018-10-18:
> I did research about map vs set. We are using map because we need to have access to element like service or characteristic with specific key like uuid or handle. This is main reason we cant use set. [...]

Why not go for the simple vector? It has almost no overhead to the storage of pointers to Characteristics

The question is what the impact is of locating a specific characteristic in a vector. Of course this means iterating through it and comparing either UUID or handle, but (even for 80 characteristics) how much extra cpu-time would it actually cost? 
I assume that the number of Descriptors per characteristic is low (< 5) so replacing the map with Descirptors in a Charactersitic with a vector would also free memory at very little cpu-cost.

chegewara, 2018-10-18:
Im not saying no, but if not counting that i did not do research in this direction before, it was also too much effort to change code and i could spend it on fixing bugs or implementing new stuff. I understand your point of view, but understand me it is first and probably only case when we have so many characteristics so we need to consider such changes. 

For some time i will be occupied by paid project nor related to ble so i cant make any changes in code, but this is open source project so feel free to fix it and PR.

chegewara, 2018-11-04:
Hi, i am facing completely opposite situation right now. I need size of characteristic value to be much bigger, 30-300kB. The issue with characteristic size is that we can do nothing to change it, it is hardcoded into esp-idf bt stack. Unlucky for us it is array with constant length:
https://github.com/espressif/esp-idf/blob/12b2268ee56dac79988b75dbb49a453629ed8022/components/bt/bluedroid/api/include/api/esp_gatt_defs.h#L369

chegewara, 2018-11-04:
@Bascy I think that best option for you would be to change this value to something more suitable, but im not sure if this change anything:
https://github.com/espressif/esp-idf/blob/12b2268ee56dac79988b75dbb49a453629ed8022/components/bt/bluedroid/api/include/api/esp_gatt_defs.h#L290

chegewara, 2018-11-06:
@Bascy https://github.com/espressif/esp-idf/issues/2664

chegewara, 2018-11-19:
I am working on library and i will add parameter to set characteristic and descriptor size during instantiating. With descriptor there is not much work, but with characteristic its a lot, because it require to change characteristic value from BLEValue type to standard esp_attr_value_t.

akotowski, 2019-02-25:
I find that having more than 5 characteristics on 1 service does not work.  I this th e limit.  How can I change this?

closed 2022-11-05 by Bascy