nkolban/esp32-snippets #685: Wifi +BLE

mangooose, 2018-10-14:
Currently when i try to connect to one ble server while wifi is connected it hangs after "<<clear services". Any idea why?

chegewara, 2018-10-14:
You have to be more precise what currently means. I am currently living this version:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test ;)

I have to admit i didnt write as much code as i wish with wifi + ble. My first application was ESP32_explorer from Kolbans repository. I had no issue back then, i was able to scan for ble devices and send results to browser (esp32 working as http server). Next app was on arduino aws IoT with mqtt + ble client connecting to POS printer. Only issue here was short of heap. Last "my" app is this example:
https://github.com/chegewara/esp32-snippets-enchancements-test/blob/multi_connect_test/main/wifi_server_with_ble_scan.c
Basically its copy/paste esp-idf http server with BLEScaner to display how many unique ble devices has been discovered.

In last few days ive stuck on these message many times. In my case it was caused by esp-idf bt stack. Its very important to stop scan before you start connecting to peripheral, this procedure is also in esp-idf multi connect example.

mangooose, 2018-10-14:
Currently i'm using arduino for that with no tasks. Can it be the problem?

chegewara, 2018-10-14:
It all depends on how you are using wifi and how ble. Few months ago i would say its easy to work wifi + ble, because i did simple test and got no issues. After struggling with POS app i can understand how hard, frustrating  and time consuming it can be to make it works. It can be as easy as `Serial.print` or as hard as surgery on battle field.

Yes, having tasks can help you, but every app is different and i dont know what to advise you. In my app what i did was to make just a few init calls, then i created whole bunch of tasks and deleted default arduino looper task with `vTaskDelete`. From this moment loop is not usable, but you have few kB of heap recovered because looper task have about 8 or 10kB stack.

Having 2 separate tasks can be good if you are pooling wifi like in mqtt library, or gives you nothing if you do infinite ble scan, because you cant manage task switching. In such case you can only start and stop scanning or try to find good interval and window settings. I can elaborate all night about it but not knowing what you are trying to do there is too many options.

chegewara, 2018-10-16:
hi @mangooose 
i am doing some tests, because i want to write wifi + ble example app and here is what i found:
- if you want to have more responsive http server/client and less accurate ble scanner its good to set long scan interval with short scan window, interval >1000ms and window <200ms 
- if you set scan interval short then http server/client will have less radio time and become lagged

chegewara, 2018-11-16:
Hi @mangooose 
I think i found good solution how to build ble + wifi coexist app, but this requires to change one setting in sdkconfig and of course rebuild arduino-esp32 library. I have test app in esp-idf that is running http server and in the same time is performing ble scan. Ble scan never stops. Ive been trying without option wifi + bluetooth coex in menuconfig and i had issue with wifi, radio went in sleep very often and get disconnected from AP. I had solution on this by stopping ble scan when disconnect event occurs and start ble scan when wifi got IP event. With arduino im having problem to open web page server by esp32 but in esp-idf it works like a charm in latest master, so maybe its been fixed and will be merged to arduino-esp32 soon.

mangooose, 2018-11-21:
Hi @chegewara 
Thanks for letting me know :)

chegewara, 2018-11-21:
I have test app wifi + ble. My wifi part is a simple http server with very simple page to report how many ble devices has been found. BLE part is simple scanner. Both work over 110 hours now and i had not crash, lag or something to report. I can see how many devices has been found and i can also clear my ble scanned devices buffer to avoid heap shortage. At the moment each scanned device is taking about 250 bytes so i can scan for about 300 devices. I didnt expected that simply code can make me cheer so much, but well.

WJohnnyW, 2019-01-14:
@chegewara your link above can't find,I need the example  you mentioned ,thanks!

chegewara, 2019-01-14:
```
/*
   Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleScan.cpp
   Ported to Arduino ESP32 by Evandro Copercini
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>
#include <WiFi.h>

const char* ssid     = "chege24";
const char* password = "1234567890";

WiFiServer server(80);

int scanTime = 50; //In seconds

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster

  pinMode(5, OUTPUT);      // set the LED pin mode

  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  
  server.begin();
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
}

void loop() {
 WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn the LED on pin 5 on.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn the LED on pin 5 off.<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(5, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(5, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}
```

chegewara, 2019-01-14:
Here is another example (esp-idf):
```
/* Persistent Sockets Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

#include <esp_wifi.h>
#include <esp_event_loop.h>
#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>

#include <http_server.h>

/* An example to demonstrate persistent sockets, with context maintained across
 * multiple requests on that socket.
 * The examples use simple WiFi configuration that you can set via 'make menuconfig'.
 * If you'd rather not, just change the below entries to strings with
 * the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
 */
#define EXAMPLE_WIFI_SSID CONFIG_WIFI_SSID
#define EXAMPLE_WIFI_PASS CONFIG_WIFI_PASSWORD

static const char *TAG="APP";
extern uint16_t get_count();
/* Function to free context */
void adder_free_func(void *ctx)
{
    ESP_LOGI(TAG, "/adder Free Context function called");
    free(ctx);
}

/* This handler keeps accumulating data that is posted to it into a per
 * socket/session context. And returns the result.
 */
esp_err_t adder_post_handler(httpd_req_t *req)
{
    /* Log total visitors */
    unsigned *visitors = (unsigned *)req->user_ctx;
    ESP_LOGI(TAG, "/adder visitor count = %d", ++(*visitors));

    char buf[10];
    char outbuf[50];
    int  ret;

    /* Read data received in the request */
    ret = httpd_req_recv(req, buf, sizeof(buf));
    if (ret < 0) {
        return ESP_FAIL;
    }

    buf[ret] = '\0';
    int val = atoi(buf);
    ESP_LOGI(TAG, "/adder handler read %d", val);

    /* Create session's context if not already available */
    if (! req->sess_ctx) {
        ESP_LOGI(TAG, "/adder allocating new session");
        req->sess_ctx = malloc(sizeof(int));
        req->free_ctx = adder_free_func;
        *(int *)req->sess_ctx = 0;
    }

[... 176 lines cut ...]

static const char* LOG_TAG = "SampleScan";
static uint16_t count = 0;
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
	void onResult(BLEAdvertisedDevice advertisedDevice) {
        ESP_LOGI(LOG_TAG, "Found BLE new device!  address: %s", advertisedDevice.getAddress().toString().c_str()); 
        count++;
	} 
}; 

extern "C" void start_ble_scan(void*) {
	ESP_LOGI(LOG_TAG, "Scanning sample starting");
	BLEDevice::init("esp32");
	BLEScan *pBLEScan = BLEDevice::getScan();
	pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), false);
    pBLEScan->setInterval(189);
    pBLEScan->setWindow(49);
	pBLEScan->setActiveScan(true);
	pBLEScan->start(0);
} 

extern "C" uint16_t get_count(){
    return count;
}
```

mkardan, 2020-07-12:
Are you using external flash? It gives me the size error. I am using ESP32_Devkitc_V4. 4MB flash.

chegewara, 2020-07-12:
Regular devKitC.
What do you mean size error?

mkardan, 2020-07-12:
[image]

chegewara, 2020-07-12:
Tools->Partition scheme

mkardan, 2020-07-12:
Thanks a lot! :)

Eiyvor, 2020-08-25:
In the menuconfig menu, there is a special option called “Software controls WiFi/
Bluetooth coexistence”, which is used to control the ESP32's Bluetooth and Wi-Fi
coexistence using software, thus balancing the coexistence requirement for controlling the
RF module by both the Wi-Fi and Bluetooth modules. Please note that if Option “Software
controls WiFi/Bluetooth coexistence” is enabled, the BLE scan interval shall not exceed
0x100 slots (about 160 ms).
• If only the BLE and Wi-Fi coexistence is required, this option can be enabled or
disabled. However, if this option is not enabled, please note that the “BLE scan
interval - BLE scan window” should be larger than 150 ms, and the BLE scan interval
should be less than 500 ms.
• If the Classic Bluetooth and Wi-Fi coexistence is required, it is recommended that you
enable this option. 

I found this in a esp32 document from espressif, this only works in esp-idf.

Shaivald16, 2021-06-23:
I have been trying to use ESP32 as a gateway for BLE devices using Blynk server.

I notice that BLE scanning is not much of an issue while Blynk is connected.

However, when I try to connect with a BLE device to exchange data it freezes or crashes.

@chegewara , Have you had a chance to try this use case? Also, Are you planning to add the “Software controls WiFi/
Bluetooth coexistence” into the Arduino library?

chegewara, 2021-06-23:
Im not planing to add anything to arduino library, unless i will be forced.

I have plenty of projects with BLE + wifi, including mqtt, http etc. I cant remember any case i had problems with it, so i am guessing i am very lucky.

Shaivald16, 2021-06-23:
> Im not planing to add anything to arduino library, unless i will be forced.
> 
> I have plenty of projects with BLE + wifi, including mqtt, http etc. I cant remember any case i had problems with it, so i am guessing i am very lucky.

Thanks for your quick reply! apart from scan interval and scan window tuning, is there any other timing/sequencing you can advise on? I have tried alternatively turning BLE and WiFi on or off but it also does not show stable performance.

I have learned a lot from your examples and contributions! Very grateful for your work.

chegewara, 2021-06-23:
Actually i dont think i had to do any adjustments, except window/interval.
Maybe all my apps were too easy and simplified so i didnt have to worry about it.

I would like to help more, but i think it is just a matter of spending few hours testing different settings.