nkolban/esp32-snippets #704: Questions: BT/BLE congestion when notifications/indications are sent

capablegh, 2018-10-18:
Hello.

A BT/BLE and ESP32 newbie here with following questions.

In the [BLE_uart.ino example](https://github.com/nkolban/ESP32_BLE_Arduino/blob/7951347ed68313d75c367e1f2cce763cb56d1eb2/examples/BLE_uart/BLE_uart.ino#L115) there is a delay inserted after a notify to handle ESP32 BT stack congestion. 

Is it a reasonable understanding that this congestion could occur because the BT stack is still transmitting a previous request, that may be because of a congested BT RF environment? If correct, would the 20ms delay used in the example be sufficient in general?

As opposed to BLE notifications, any sent BLE indications are expected to receive a Handle Value Indication from remote client. How does a ESP32 client program receive this confirmation? I mean to ask,  is the BLECharacteristic::indicate call synchronous? If yes, how is an error response given to the API user for a failure to successfully send the indication?  On the other hand, if the indicate method asynchronous, then how does the API user register a callback?

The attached capture was taken on a Linux/BlueZ stack using hcidump, then opened and PDF exported from Wireshark. It shows BlueZ receiving the indication from an ESP32, and BlueZ sending a response. I have un-collapsed only the attributes in context of these questions.

[ble-indicate-msgs.pdf](https://github.com/nkolban/esp32-snippets/files/2491414/ble-indicate-msgs.pdf)

Thanks you.

chegewara, 2018-10-18:
This library is pretty good developed but is not finished yet and have few bugs. It is because we, who are developing it, are learning ble from basics. I can say honestly that year ago i knew about bluetooth only one thing: it is some sort of wireless communication, period. 
What i want to say is that there is missing congestion and possibly few other important events in callback. At the moment only CONF_EVT is handled. There is still a lot i have to learn and thanks users like you i can learn what i am still missing.

Now answering you question about delay() in example. It is possible this delay was required when esp-idf bt stack was still in development stage. I would say it is save to delete it now (not tested).

capablegh, 2018-10-18:
@chegewara Thank you for your prompt and helpful response. Without your C++ API in the Arduino framework for the ESP32 (BLE and others), it would be much harder to read and work with the plain C Espressif APIs. These C++ references have helped me expedite my learning. Thank you and other contributors for these outstanding contributions!

About the delay: From a perspective the esp-idf stack may have become more efficient in that it does not lead to congestion as it did before, but depending on other load on the processors, could the BT stack not be still susceptible to congestion under some conditions? Small delays as you put could still provide for a workaround. 

External noisy/congested RF environment could be still be relevant, in which case callbacks could be better to deal with that situation than the delay.

I agree with you that every reasonably complex libraries and APIs go through iterations of improvements, it is anticipated.

chegewara, 2018-10-24:
At the moment its hard to test congesting in this library. I am now very busy at my daily job, i dont even have time to do esp32 payment job, but i have idea how to add functionality to library.

My idea is to add 2 callback functions. User/programmer will have option to register callback and receive all GAP or GATT events and take responsibility to add functionality that is not implemented in this generic library. It is not hard to implement, just a few lines of code.

wanqbabay, 2018-11-02:
hello, the question is that in the BLE_uart.ino example there is a delay (delay(10); // bluetooth stack will go into congestion, if too many packets are sent)inserted after every notify .

when i set the delay(100 or less), it will disconnect with my iphone through NRF in 1 min. After many tests, i think the BLE stack is congestiong, but i dont know how to do?
anyone can help me?  
here is the code:

#include "sdkconfig.h"

#include <esp_log.h>
#include <string>
#include <sstream>
#include <sys/time.h>
#include "BLEDevice.h"

#include "BLEServer.h"
#include "BLEUtils.h"
#include "BLE2902.h"
#include "Task.h"

static char LOG_TAG[] = "SampleNotify";
 BLEServer *pServer=NULL;
// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

BLECharacteristic *pCharacteristic;

class MyNotifyTask: public Task {
  void run(void *data) {
    uint8_t value = 0;
    while(1) {
      delay(30);
      ESP_LOGD(LOG_TAG, "*** NOTIFY: %d ***", value);
      pCharacteristic->setValue(&value, 1);
      Serial.println(value);
      pCharacteristic->setNotifyProperty(true);
      pCharacteristic->notify();
      ESP_LOGD(TAG,"%d:-RAM left %d",__LINE__,esp_get_free_heap_size());
      pCharacteristic->indicate();
//      ESP_LOGD(LOG_TAG, ">> setPower: %d", powerLevel);
     
      value++;
    } // While 1
  } // run
}; // MyNotifyTask

MyNotifyTask *pMyNotifyTask;

class MyServerCallbacks: public BLEServerCallbacks {
  void onConnect(BLEServer* pServer) {
    pMyNotifyTask->start();
    BLE2902* desc = (BLE2902*)  pCharacteristic->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
     desc->setNotifications(true);
  };

  void onDisconnect(BLEServer* pServer) {
    pMyNotifyTask->stop();
     delay(500); // give the bluetooth stack the chance to get things ready
     pServer->startAdvertising(); // restart advertising
     Serial.println("start advertising");
  }
};

static void run() {
  pMyNotifyTask = new MyNotifyTask();
  pMyNotifyTask->setStackSize(10000);
  // Create the BLE Device
  BLEDevice::init("MyESP32");
[... 2580 lines cut ...]
[D][BLEUtils.cpp:1692] dumpGattServerEvent(): [status: ESP_GATT_OK, conn_id: 0x00]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
[D][BLECharacteristic.cpp:566] notify(): << notify
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLE_notify_task.ino:35] run(): 35:-RAM left 79748
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:473] indicate(): >> indicate: length: 1
[D][BLECharacteristic.cpp:490] indicate(): << indications disabled; ignoring
[D][BLE_notify_task.ino:30] run(): *** NOTIFY: 80 ***
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=1, data=50, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:671] setValue(): << setValue
80
[D][BLECharacteristic.cpp:524] notify(): >> notify: length: 1
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONF_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONF_EVT
[D][BLEUtils.cpp:1692] dumpGattServerEvent(): [status: ESP_GATT_OK, conn_id: 0x00]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
[D][BLECharacteristic.cpp:566] notify(): << notify
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLE_notify_task.ino:35] run(): 35:-RAM left 79748
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:473] indicate(): >> indicate: length: 1
[D][BLECharacteristic.cpp:490] indicate(): << indications dis

wanqbabay, 2018-11-02:
when i disenable the notify and enable the indicate,the link between esp32 and my iphone is stable;
Everything is ok!

chegewara, 2018-11-02:
> when i disenable the notify and enable the indicate,the link between esp32 and my iphone is stable;
> Everything is ok!

Thats good clue, probably issues with semaphore.

wanqbabay, 2018-11-02:
thanks for the reply; i also test the  example "BLE_uart_into" with my iphone; the result is the same"disconnect"; if there is  something wrong with  semaphore, the result should be the same when i disenable the notify and enable the indicate,the link between esp32 and my iphone is stable;

could you give some advice to test!
PS  i also test the Android with NRF, the link bteween them can last longer,but the result is also the same

chegewara, 2018-11-02:
If you could try library from this repo:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

Ive been doing some bug fixes, notification semaphore too. Thanks.

wanqbabay, 2018-11-06:
hello! sorry for the delay; i did also tests,and i find something different by android phone not Iphone like this: 
use these code:
class MyNotifyTask: public Task {
  void run(void *data) {
    uint8_t value = 0;
   while(1){
      delay(20);
      ESP_LOGD(LOG_TAG, "*** NOTIFY: %d ***", value);
      pCharacteristic->setValue(&value, 1);
      Serial.println(value);
      pCharacteristic->setNotifyProperty(true);
      pCharacteristic->notify();
      ESP_LOGD(TAG,"%d:-RAM left %d",__LINE__,esp_get_free_heap_size());

here is the log:
[D][BLECharacteristic.cpp:524] notify(): >> notify: length: 1
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONGEST_EVT
**[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONGEST_EVT**
**[D][BLEUtils.cpp:1700] dumpGattServerEvent(): [conn_id: 0, congested: 1]**
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONGEST_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONGEST_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONF_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONF_EVT
[D][BLEUtils.cpp:1692] dumpGattServerEvent(): [status: ESP_GATT_CONGESTED, conn_id: 0x00]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
[D][BLECharacteristic.cpp:566] notify(): << notify
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLE_notify_task.ino:43] run(): 43:-RAM left 61956
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLE_notify_task.ino:38] run(): *** NOTIFY: 163 ***
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=1, data=a3, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:671] setValue(): << setValue

 

from the log,i think the reason for ble_notify problem is the congest not  semaphore; 
What can i do to solve this?
thank you

chegewara, 2018-11-06:
@wanqbabay You can increase delay or option 2 is harder to do. In BLEServer is additional onConnect callback function:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEServer.h#L112
you could add something like that in your code:
```
class MyCallbacks : public BLEServerCallbacks{
	void onConnect(BLEServer *pServer, esp_ble_gatts_cb_param_t *param)
	{
		// testing how it works, before it will be implemented as part of BLEServer
		esp_ble_conn_update_params_t conn_params;
        memcpy(conn_params.bda, param->connect.remote_bda, sizeof(esp_bd_addr_t));
		conn_params.latency = 0;
        conn_params.max_int = 0x0c;    // max_int = 0x0c*1.25ms = 15ms
        conn_params.min_int = 0x06;    // min_int = 0x06*1.25ms = 7.5ms
        conn_params.timeout = 1000;    // timeout = 1000*10ms = 10000ms  <-- just example
		esp_ble_gap_update_conn_params(&conn_params); 

		// BLEDevice::getAdvertising()->stop();
		// BLEDevice::startAdvertising();
		ESP_LOGI(LOG_TAG, "onConnect");
	}

	void onDisconnect(BLEServer *pServer)
	{
		ESP_LOGI(LOG_TAG, "onDisconnect");
	}

};
```
There is no warrant it will work, because it depends on peer device settings, but its worth to try. 

To use this code you have to delete BLE library from arduino and replace it with the one from this repository or from mine repository where you have more bugfixes.
Instruction:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#replacing-the-version-that-comes-with-arduino-esp32

My repository:
https://github.com/chegewara/esp32-snippets-enchancements-test/tree/multi_connect_test

chegewara, 2018-11-06:
@wanqbabay This is not about BLEServer.cpp, there is a lot more than that. Even more important is BLECharacteristic, because this is all about notify. I think i am able to send notifications with rate around 60-100Hz in my HID keyboard example even without that special callback modification (library from this repo). This is my test code and as you can see i have only 10ms delay between 2 notifications and i can confirm that all notifications can be received:
```
    	while(1){
			char* hello = (char*)world;
			if(!gpio_get_level(GPIO_NUM_0)){
				while(*hello){
					KEYMAP map = keymap[(uint8_t)*hello];
					/*
					* simulate keydown, we can send up to 6 keys
					*/
					uint8_t a[] = {map.modifier, 0x0, map.usage, 0x0,0x0,0x0,0x0,0x0};
					input->setValue(a,8);
					input->notify();
					vTaskDelay(10/portTICK_PERIOD_MS);

					/*
					* simulate keyup
					*/
					uint8_t v[] = {0x0, 0x0, 0x0, 0x0,0x0,0x0,0x0,0x0};
					input->setValue(v, 8);
					input->notify();
					hello++;

					vTaskDelay(10/portTICK_PERIOD_MS);
				}
				// vTaskDelay(10/portTICK_PERIOD_MS); // simulate write message every 2 seconds
			}
			vTaskDelay(50/portTICK_PERIOD_MS);
    	}
```

chegewara, 2018-11-06:
Logs from my hid example:
```
W (35258) SampleHIDDevice: custom gatts event handler, event: 5
W (35268) SampleHIDDevice: custom gatts event handler, event: 5
W (35278) SampleHIDDevice: custom gatts event handler, event: 5
W (35288) SampleHIDDevice: custom gatts event handler, event: 5
W (35298) SampleHIDDevice: custom gatts event handler, event: 5
W (35308) SampleHIDDevice: custom gatts event handler, event: 5
W (35318) SampleHIDDevice: custom gatts event handler, event: 5
W (35328) SampleHIDDevice: custom gatts event handler, event: 5
W (35338) SampleHIDDevice: custom gatts event handler, event: 5
W (35348) SampleHIDDevice: custom gatts event handler, event: 5
W (35358) SampleHIDDevice: custom gatts event handler, event: 5
W (35368) SampleHIDDevice: custom gatts event handler, event: 5
W (35378) SampleHIDDevice: custom gatts event handler, event: 5
W (35388) SampleHIDDevice: custom gatts event handler, event: 5
W (35398) SampleHIDDevice: custom gatts event handler, event: 5
W (35408) SampleHIDDevice: custom gatts event handler, event: 5
W (35418) SampleHIDDevice: custom gatts event handler, event: 5
W (35428) SampleHIDDevice: custom gatts event handler, event: 5
W (35438) SampleHIDDevice: custom gatts event handler, event: 5
W (35448) SampleHIDDevice: custom gatts event handler, event: 5
W (35458) SampleHIDDevice: custom gatts event handler, event: 5
W (35468) SampleHIDDevice: custom gatts event handler, event: 5
W (35478) SampleHIDDevice: custom gatts event handler, event: 5
W (35488) SampleHIDDevice: custom gatts event handler, event: 5
W (35498) SampleHIDDevice: custom gatts event handler, event: 5
W (35508) SampleHIDDevice: custom gatts event handler, event: 5
W (35518) SampleHIDDevice: custom gatts event handler, event: 5
W (35528) SampleHIDDevice: custom gatts event handler, event: 5
W (35538) SampleHIDDevice: custom gatts event handler, event: 5
W (35548) SampleHIDDevice: custom gatts event handler, event: 5
W (35558) SampleHIDDevice: custom gatts event handler, event: 5
W (35568) SampleHIDDevice: custom gatts event handler, event: 5
W (35578) SampleHIDDevice: custom gatts event handler, event: 5
W (35588) SampleHIDDevice: custom gatts event handler, event: 5
W (35598) SampleHIDDevice: custom gatts event handler, event: 5
W (35608) SampleHIDDevice: custom gatts event handler, event: 5
W (35618) SampleHIDDevice: custom gatts event handler, event: 5
W (35628) SampleHIDDevice: custom gatts event handler, event: 5
W (35638) SampleHIDDevice: custom gatts event handler, event: 5
W (35648) SampleHIDDevice: custom gatts event handler, event: 5
W (35658) SampleHIDDevice: custom gatts event handler, event: 5
W (35668) SampleHIDDevice: custom gatts event handler, event: 5
W (35678) SampleHIDDevice: custom gatts event handler, event: 5
W (35688) SampleHIDDevice: custom gatts event handler, event: 5
W (35698) SampleHIDDevice: custom gatts event handler, event: 5
W (35708) SampleHIDDevice: custom gatts event handler, event: 5
W (35718) SampleHIDDevice: custom gatts event handler, event: 5
W (35728) SampleHIDDevice: custom gatts event handler, event: 5
W (35738) SampleHIDDevice: custom gatts event handler, event: 5
W (35748) SampleHIDDevice: custom gatts event handler, event: 5
W (35758) SampleHIDDevice: custom gatts event handler, event: 5
W (35768) SampleHIDDevice: custom gatts event handler, event: 5
W (35778) SampleHIDDevice: custom gatts event handler, event: 5
W (35788) SampleHIDDevice: custom gatts event handler, event: 5
W (35798) SampleHIDDevice: custom gatts event handler, event: 5
W (35808) SampleHIDDevice: custom gatts event handler, event: 5
W (35818) SampleHIDDevice: custom gatts event handler, event: 5
W (35828) SampleHIDDevice: custom gatts event handler, event: 5
W (35838) SampleHIDDevice: custom gatts event handler, event: 5
W (35848) SampleHIDDevice: custom gatts event handler, event: 5
W (35858) SampleHIDDevice: custom gatts event handler, event: 5
W (35868) SampleHIDDevice: custom gatts event handler, event: 5
W (35878) SampleHIDDevice: custom gatts event handler, event: 5
W (35888) SampleHIDDevice: custom gatts event handler, event: 5
W (35898) SampleHIDDevice: custom gatts event handler, event: 5
W (35908) SampleHIDDevice: custom gatts event handler, event: 5
W (35918) SampleHIDDevice: custom gatts event handler, event: 5
W (35928) SampleHIDDevice: custom gatts event handler, event: 5
W (35938) SampleHIDDevice: custom gatts event handler, event: 5
W (35948) SampleHIDDevice: custom gatts event handler, event: 5
W (35958) SampleHIDDevice: custom gatts event handler, event: 5
W (35968) SampleHIDDevice: custom gatts event handler, event: 5
W (35978) SampleHIDDevice: custom gatts event handler, event: 5
W (35988) SampleHIDDevice: custom gatts event handler, event: 5
W (35998) SampleHIDDevice: custom gatts event handler, event: 5
W (36008) SampleHIDDevice: custom gatts event handler, event: 5
W (36018) SampleHIDDevice: custom gatts event handler, event: 5
W (36028) SampleHIDDevice: custom gatts event handler, event: 5
```
`    ESP_GATTS_CONF_EVT                = 5,       /*!< When receive confirm, the event comes */`
As you cans see notify is sent every 10ms and no congest event present.

wanqbabay, 2018-11-07:
@chegewara  thanks for your help.
From the code you provide,there is a "Big Delay" (vTaskDelay(50/portTICK_PERIOD_MS);) that i need not,which means more tha 10ms delay between 2 notifications average ;
besides,what i need is that the BLE_Notify() can last for 1h at some speed(30Hz or more);

i test the code; PS i didnot replace any source files, and got the same outcome.
Here is the log:
[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLECharacteristic.cpp:524] notify(): >> notify: length: 8
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONGEST_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONGEST_EVT
[D][BLEUtils.cpp:1700] dumpGattServerEvent(): [conn_id: 0, congested: 1]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONGEST_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONGEST_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONF_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONF_EVT
[D][BLEUtils.cpp:1692] dumpGattServerEvent(): [status: ESP_GATT_CONGESTED, conn_id: 0x00]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONF_EVT
[D][BLECharacteristic.cpp:566] notify(): << notify
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=6, data=000000000000, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLECharacteristic.cpp:524] notify(): >> notify: length: 6
[E][BLECharacteristic.cpp:560] notify(): << esp_ble_gatts_send_indicate: rc=-1 Fail
[D][BLECharacteristic.cpp:664] setValue(): >> setValue: length=8, data=0000000000000000, characteristic UUID=beb5483e-36e1-4688-b7f5-ea07361b26a8
[D][BLECharacteristic.cpp:671] setValue(): << setValue
[D][BLECharacteristic.cpp:524] notify(): >> notify: length: 8
[D][BLEDevice.cpp:96] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_CONGEST_EVT
[D][BLEUtils.cpp:1647] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_CONGEST_EVT
[D][BLEUtils.cpp:1700] dumpGattServerEvent(): [conn_id: 0, congested: 0]
[D][BLEServer.cpp:177] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONGEST_EVT
[D][BLECharacteristic.cpp:209] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_CONGEST_EVT
[D][BLECharacteristic.cpp:461] handleGATTServerEvent(): << handleGATTServerEvent
[D][BLEServer.cpp:295] handleGATTServerEvent(): << handleGATTServerEvent

ESP_GATTS_CONGEST_EVT,are there any other ideas except increasing the delay or the packages ?

thanks

wanqbabay, 2018-11-07:
@chegewara 
thanks for you help;
All i need:
notify rate :30Hz~70Hz
notify length: 16Bytes  like this: 1000000010000000
in fact ,i use the arduino library; and i test the example about ble (BLE_Notify and BLE_Uart);but it is not stable, Android can last longer than IOS by NRF. Then i creat the task,but the same outcome.
the reason for Android  is the  **CONGEST_EVT**
the reason for IOS  is the  **DISCONNECT_EVT**
i think   the **CONGEST_EVT**  is the reason.
Anything what  i can to  help? please let me konw.
Thanks again.

chegewara, 2018-11-07:
The problem with arduino ble library is that it has not been updated for months (over 6 months i think). Even this repository has few bugs, memory leaks etc. I will create big PR when i am ready with all features and bugfixes i am doing now, but it takes time to test all.

chegewara, 2018-11-07:
You can get this repository or from this repository --> https://github.com/chegewara/esp32-snippets-enchancements-test and use instruction from link to replace library:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#replacing-the-version-that-comes-with-arduino-esp32

wanqbabay, 2018-11-08:
@chegewara  
Thanks again.
BLE_notify  is more stable than before after replacing the library. But it happens unexpectedly.
I test the **BLE_uart example**,when i declare two Characteristics,the Characteristic->notify is disabled
the code:
 BLECharacteristic * pRxCharacteristic = pService->createCharacteristic(
		CHARACTERISTIC_UUID_RX,
		BLECharacteristic::PROPERTY_WRITE
		);
  pRxCharacteristic->setCallbacks(new MyCallbacks());
 BLECharacteristic  * pTxCharacteristic = pService->createCharacteristic(
                CHARACTERISTIC_UUID_TX,
                BLECharacteristic::PROPERTY_NOTIFY
               );                   
  pTxCharacteristic->addDescriptor(new BLE2902());
the log:
[D][BLECharacteristic.cpp:640] setValue(): << setValue
[D][BLECharacteristic.cpp:484] notify(): >> notify: length: 1
[D][BLECharacteristic.cpp:504] notify(): << notifications disabled; ignoring
[D][BLECharacteristic.cpp:633] setValue(): >> setValue: length=1, data=66, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e
[D][BLECharacteristic.cpp:640] setValue(): << setValue
[D][BLECharacteristic.cpp:484] notify(): >> notify: length: 1
[D][BLECharacteristic.cpp:504] notify(): << notifications disabled; ignoring
[D][BLECharacteristic.cpp:633] setValue(): >> setValue: length=1, data=67, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e
[D][BLECharacteristic.cpp:640] setValue(): << setValue
[D][BLECharacteristic.cpp:484] notify(): >> notify: length: 1
[D][BLECharacteristic.cpp:504] notify(): << notifications disabled; ignoring
[D][BLECharacteristic.cpp:633] setValue(): >> setValue: length=1, data=68, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e
[D][BLECharacteristic.cpp:640] setValue(): << setValue
[D][BLECharacteristic.cpp:484] notify(): >> notify: length: 1
[D][BLECharacteristic.cpp:504] notify(): << notifications disabled; ignoring
[D][BLECharacteristic.cpp:633] setValue(): >> setValue: length=1, data=69, characteristic UUID=6e400003-b5a3-f393-e0a9-e50e24dcca9e
[D][BLECharacteristic.cpp:640] setValue(): << setValue
[D][BLECharacteristic.cpp:484] notify(): >> notify: length: 1
[D][BLECharacteristic.cpp:504] notify(): << notifications disabled; ignoring

But when i declare one Characteristic,it is normal
code:
BLECharacteristic  * pTxCharacteristic = pService->createCharacteristic(
                CHARACTERISTIC_UUID_TX,
                BLECharacteristic::PROPERTY_NOTIFY
               );                   
  pTxCharacteristic->addDescriptor(new BLE2902());

Strangely  and deceptively.

thanks again

chegewara, 2018-11-08:
`[D][BLECharacteristic.cpp:504] notify(): << notifications disabled; ignoring` This means that client app like nRF connect didnt register for notifications. If you do register for notifications and app still logs that then its bug and i will investigate it.

wanqbabay, 2018-11-08:
i did  many tests,the notify was disabled.
and there is a bug in BLECharacteristic.cpp
the code:
case ESP_GATTS_CONF_EVT: {
			if(param->conf.conn_id == getService()->getServer()->getConnId() && param->**conf.handle** == m_handle)
				m_semaphoreConfEvt.give(param->conf.status);
			break;
		}
i uesed **conf.len** replace the **conf.handle**. then everything is ok but ble_notify i found ;i dont know whether it  matters.
thanks  again

chegewara, 2018-11-08:
Yes, you're right. This has been added in esp-idf master and i forgot that arduino-esp32 is a little behind:
`conf.handle`

chegewara, 2018-11-08:
I am running tests right now. I will give it a try for about hour. For now i am getting such logs:
```
W (172253) SampleServer: custom gatts event handler, GAP event: 5
W (172333) SampleServer: custom gatts event handler, GAP event: 5
W (172413) SampleServer: custom gatts event handler, GAP event: 5
W (172493) SampleServer: custom gatts event handler, GAP event: 5
W (172573) SampleServer: custom gatts event handler, GAP event: 5
W (172653) SampleServer: custom gatts event handler, GAP event: 5
W (172733) SampleServer: custom gatts event handler, GAP event: 5
W (172813) SampleServer: custom gatts event handler, GAP event: 5
W (172893) SampleServer: custom gatts event handler, GAP event: 5
W (172973) SampleServer: custom gatts event handler, GAP event: 5
W (173053) SampleServer: custom gatts event handler, GAP event: 5
W (173133) SampleServer: custom gatts event handler, GAP event: 5
W (173213) SampleServer: custom gatts event handler, GAP event: 5
W (173293) SampleServer: custom gatts event handler, GAP event: 5
W (173373) SampleServer: custom gatts event handler, GAP event: 5
W (173453) SampleServer: custom gatts event handler, GAP event: 5
W (173533) SampleServer: custom gatts event handler, GAP event: 5
W (173613) SampleServer: custom gatts event handler, GAP event: 5
W (173693) SampleServer: custom gatts event handler, GAP event: 5
W (173773) SampleServer: custom gatts event handler, GAP event: 5
W (173853) SampleServer: custom gatts event handler, GAP event: 5
W (173933) SampleServer: custom gatts event handler, GAP event: 5
W (174013) SampleServer: custom gatts event handler, GAP event: 5
W (174093) SampleServer: custom gatts event handler, GAP event: 5
W (174173) SampleServer: custom gatts event handler, GAP event: 5
W (174253) SampleServer: custom gatts event handler, GAP event: 5
W (174333) SampleServer: custom gatts event handler, GAP event: 5
W (174413) SampleServer: custom gatts event handler, GAP event: 5
W (174493) SampleServer: custom gatts event handler, GAP event: 5
W (174573) SampleServer: custom gatts event handler, GAP event: 5
W (174653) SampleServer: custom gatts event handler, GAP event: 5
W (174733) SampleServer: custom gatts event handler, GAP event: 5
W (174813) SampleServer: custom gatts event handler, GAP event: 5
W (174893) SampleServer: custom gatts event handler, GAP event: 5
W (174973) SampleServer: custom gatts event handler, GAP event: 5
W (175053) SampleServer: custom gatts event handler, GAP event: 5
W (175133) SampleServer: custom gatts event handler, GAP event: 5
W (175213) SampleServer: custom gatts event handler, GAP event: 5
W (175293) SampleServer: custom gatts event handler, GAP event: 5
W (175373) SampleServer: custom gatts event handler, GAP event: 5
```
`event 5` means confirm from bt stack that notify has been sent. My nRF connect in other hand is receiving, or at least is logging, notifications every 70-80ms.

chegewara, 2018-11-08:
```
W (638427) SampleServer: custom gatts event handler, GAP event: 5
W (638436) SampleServer: custom gatts event handler, GAP event: 5
W (638445) SampleServer: custom gatts event handler, GAP event: 5
W (638454) SampleServer: custom gatts event handler, GAP event: 5
W (638463) SampleServer: custom gatts event handler, GAP event: 5
W (638472) SampleServer: custom gatts event handler, GAP event: 5
W (638481) SampleServer: custom gatts event handler, GAP event: 5
W (638490) SampleServer: custom gatts event handler, GAP event: 5
W (638499) SampleServer: custom gatts event handler, GAP event: 5
W (638508) SampleServer: custom gatts event handler, GAP event: 5
W (638517) SampleServer: custom gatts event handler, GAP event: 5
W (638526) SampleServer: custom gatts event handler, GAP event: 5
W (638535) SampleServer: custom gatts event handler, GAP event: 5
W (638544) SampleServer: custom gatts event handler, GAP event: 5
W (638553) SampleServer: custom gatts event handler, GAP event: 5
W (638562) SampleServer: custom gatts event handler, GAP event: 5
W (638571) SampleServer: custom gatts event handler, GAP event: 5
W (638580) SampleServer: custom gatts event handler, GAP event: 5
W (638589) SampleServer: custom gatts event handler, GAP event: 5
W (638598) SampleServer: custom gatts event handler, GAP event: 5
W (638607) SampleServer: custom gatts event handler, GAP event: 5
W (638616) SampleServer: custom gatts event handler, GAP event: 5
W (638625) SampleServer: custom gatts event handler, GAP event: 5
W (638634) SampleServer: custom gatts event handler, GAP event: 5
W (638643) SampleServer: custom gatts event handler, GAP event: 5
W (638652) SampleServer: custom gatts event handler, GAP event: 5
W (638661) SampleServer: custom gatts event handler, GAP event: 5
W (638670) SampleServer: custom gatts event handler, GAP event: 5
W (638679) SampleServer: custom gatts event handler, GAP event: 5
W (638688) SampleServer: custom gatts event handler, GAP event: 5
W (638697) SampleServer: custom gatts event handler, GAP event: 5
W (638706) SampleServer: custom gatts event handler, GAP event: 5
W (638715) SampleServer: custom gatts event handler, GAP event: 5
W (638724) SampleServer: custom gatts event handler, GAP event: 5
```

chegewara, 2018-11-08:
I can say that after about 45-50 minutes my console went down (happened sometimes), but esp32 is still sending notifications with about 100-110Hz rate with no congestion. I know it is only simple raw test and does not prove that application also can send notifications with 100Hz rate, also this is esp-idf app and not arduino, but still i think its good stress test.
```
W (2862212) SampleServer: custom gatts event handler, GAP event: 5
W (2862221) SampleServer: custom gatts event handler, GAP event: 5
W (2862230) SampleServer: custom gatts event handler, GAP event: 5
W (2862238) SampleServer: custom gatts event handler, GAP event: 5
W (2862248) SampleServer: custom gatts event handler, GAP event: 5
W (2862257) SampleServer: custom gatts event handler, GAP event: 5
W (2862266) SampleServer: custom gatts event handler, GAP event: 5
W (2862275) SampleServer: custom gatts event handler, GAP event: 5
W (2862284) SampleServer: custom gatts event handler, GAP event: 5
W (2862293) SampleServer: custom gatts event handler, GAP event: 5
W (2862302) SampleServer: custom gatts event handler, GAP event: 5
W (2862311) SampleServer: custom gatts event handler, GAP event: 5
W (2862320) SampleServer: custom gatts event handler, GAP event: 5
W (2862329) SampleServer: custom gatts event handler, GAP event: 5
W (2862338) SampleServer: custom gatts event handler, GAP event: 5
W (2862347) SampleServer: custom gatts event handler, GAP event: 5
W (2862356) SampleServer: custom gatts event handler, GAP event: 5
W (2862365) SampleServer: custom gatts event handler, GAP event: 5
W (2862374) SampleServer: custom gatts event handler, GAP event: 5
W (2862383) SampleServer: custom gatts event handler, GAP event: 5
W (2862392) SampleServer: custom gatts event handler, GAP event: 5
W (2862401) SampleServer: custom gatts event handler, GAP event: 5
W (2862410) SampleServer: custom gatts event handler, GAP event: 5
W (2862419) SampleServer: custom gatts event handler, GAP event: 5
W (2862428) SampleServer: custom gatts event handler, GAP event: 5
W (2862437) SampleServer: custom gatts event handler, GAP event: 5
W (2862446) SampleServer: custom gatts event handler, GAP event: 5
W (2862455) SampleServer: custom gatts event handler, GAP event: 5
W (2862464) SampleServer: custom gatts event handler, GAP event: 5
W (2862473) SampleServer: custom gatts event handler, GAP event: 5
W (2862482) SampleServer: custom gatts event handler, GAP event: 5
W (2862491) SampleServer: custom gatts event handler, GAP event: 5
W (2862500) SampleServer: custom gatts event handler, GAP event: 5
W (2862509) SampleServer: custom gatts event handler, GAP event: 5
W (2862518) SampleServer: custom gatts event handler, GAP event: 5
W (2862527) SampleServer: custom gatts event handler, GAP event: 5
W (2862536) SampleServer: custom gatts event handler, GAP event: 5
W (2862545) SampleServer: custom gatts event handler, GAP event: 5
W (2862554) SampleServer: custom gatts event handler, GAP event: 5
W (2862563) SampleServer: custom gatts event handler, GAP event: 5
W (2862572) SampleServer: custom gatts event handler, GAP event: 5
W (2862581) SampleServer: custom gatts event handler, GAP event: 5
W (2862590) SampleServer: custom gatts event handler, GAP event: 5
W (2862599) SampleServer: custom gatts event handler, GAP event: 5
W (2862608) SampleServer: custom gatts event handler, GAP event: 5
W (2862617) SampleServ
```

chegewara, 2018-11-08:
There is more. I am trying to trigger congestion event by setting loop delay to 3ms but instead i have btuT watchdog:
```
W (42222) SampleServer: custom gatts event handler, GAP event: 5
W (42229) SampleServer: custom gatts event handler, GAP event: 5
W (42235) SampleServer: custom gatts event handler, GAP event: 5
W (42242) SampleServer: custom gatts event handler, GAP event: 5
W (42250) SampleServer: custom gatts event handler, GAP event: 5
W (42255) SampleServer: custom gatts event handler, GAP event: 5
W (42262) SampleServer: custom gatts event handler, GAP event: 5
E (42269) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (42269) task_wdt:  - IDLE0 (CPU 0)
E (42269) task_wdt: Tasks currently running:
E (42269) task_wdt: CPU 0: btuT
E (42269) task_wdt: CPU 1: IDLE1
W (42297) SampleServer: custom gatts event handler, GAP event: 5
W (42303) SampleServer: custom gatts event handler, GAP event: 5
W (42309) SampleServer: custom gatts event handler, GAP event: 5
W (42317) SampleServer: custom gatts event handler, GAP event: 5
W (42323) SampleServer: custom gatts event handler, GAP event: 5
W (42329) SampleServer: custom gatts event handler, GAP event: 5
W (42338) SampleServer: custom gatts event handler, GAP event: 5
W (42343) SampleServer: custom gatts event handler, GAP event: 5
```

chegewara, 2018-11-08:
Hi @wanqbabay 
ive been doing some performance test and i can say few thing that you may be interested in:
- first of all that new parameter which should help us (param->conf.handle) is bugged in esp-idf, but this is no big issue, we can just ignore it for now,
- second of all i cant with my simple test app trigger congestion even if i am trying to do it on purpose (esp-idf app),
- third of all with proper parameters setting esp32 is connecting with my android smartphone with 11.5ms interval, with this setup and using indications instead you should be able to have very stable average indication rate about 23ms,
- if you have more question i will be happy to help

wanqbabay, 2018-11-12:
Thanks a lot @chegewara 
Now i have two other questions:

1. If i creat two BLE Characteristics 
 the code like this:
 pRXCharacteristic = pService->createCharacteristic(
    BLEUUID(CHARACTERISTIC_UUID_RX),
    BLECharacteristic::PROPERTY_WRITE 
  );
 pRXCharacteristic->setCallbacks(new MyCallbacks()); 
    pTXCharacteristic = pService->createCharacteristic(
    BLEUUID(CHARACTERISTIC_UUID_TX),
    BLECharacteristic::PROPERTY_NOTIFY 
  ); 
 pTXCharacteristic->addDescriptor(new BLE2902());
then the **PROPERTY_NOTIFY**  is not enable and  can not  register  notifications by NRF;if i  creat only one BLE Characteristic,i can  register  notifications by NRF.
the code like this:
 pRXCharacteristic->setCallbacks(new MyCallbacks()); 
    pTXCharacteristic = pService->createCharacteristic(
    BLEUUID(CHARACTERISTIC_UUID_TX),
    BLECharacteristic::PROPERTY_NOTIFY 
  ); 
 pTXCharacteristic->addDescriptor(new BLE2902());
All above is based on the test Repo. if i use the latest arduino library, the question is gone.

2. BLE link
I want to link the ESP32 BLE and Phone through the MAC address and failed. the Phone can make it to link other device BLE through MAC address.  

Thanks again

chegewara, 2018-11-15:
@wanqbabay Do you have example i can use for tests?

jubueche, 2019-01-14:
Hey great work @chegewara . When will you do a PR? I am currently using arduino-esp with the esp-idf and setting the delay between notifications to lower than 20ms causes congestion. Now I will try your version.

jubueche, 2019-01-14:
Hi,
I am also trying to reduce the delay time between notify calls. It currently works for me with >30ms delay. I have tried nkolban's version of the BLE directory, but nothing changed (I tried 15ms). Then I used your files in cpp_utils (I assume this is correct?) and got the error:

E (1157) BT_BTC: Invalid advertisting peer address type parameters.

Any idea how I should fix it?
And why doesn't nkolban's version give me lower delay time?

Thanks!

chegewara, 2019-08-14:
To all who may find it useful.
BLE congestion is to inform that peer device that we are connected to cant process messages we are sending, which means we shall wait between sending notification/indication.