nkolban/esp32-snippets #705: BT_LOG: Conroller not initialised

dkselw, 2018-10-19:
I have been using the BLE C++ classes (downloaded 9 Aug) from esp32-snippets for a project I am working on. I have been having some stability issues, so I tried downloading the current  master branch today, and I also updated to the latest version of esp-idf v3.2-dev-1250-g221eced06.

Now BLE does not run at all. I get the same result if I compile and run SampleServer.cpp from BLETests.

The following errors are reported when I run _make monitor_:
E (285) BT_LOG: Conroller not initialised
E (288) BLEDevice: esp_bluedroid_init: rc=259 ESP_ERR_INVALID_STATE

Complete log file
[log.txt](https://github.com/nkolban/esp32-snippets/files/2494900/log.txt)

chegewara, 2018-10-19:
Hi,
Please check Bluetooth settings in menuconfig 
`make menuconfig `

dkselw, 2018-10-19:
These are my Bluetooth settings in menuconfig

[*] Bluetooth
Bluetooth controller  --->
[*] Bluedroid Enable  --->            

Bluetooth controller mode (BR/EDR/BLE/DUALMODE) (BLE Only)  --->
(3) BLE Max Connections
The cpu core which bluetooth controller run (Core 0 (PRO CPU))  --->
HCI mode (VHCI)  --->
MODEM SLEEP Options  --->
[*] BLE Scan Duplicate Options
Scan Duplicate Type (Scan Duplicate By Device Address)  --->
	(X) Scan Duplicate By Device Address
	( ) Scan Duplicate By Advertising Data
	( ) Scan Duplicate By Device Address And Advertising Data
(200) Maximum number of devices in scan duplicate filter
[ ]   Special duplicate scan mechanism for BLE Mesh scan

--- Bluedroid Enable
The cpu core which Bluedroid run (Core 0 (PRO CPU))  --->
(8000) Bluetooth event (callback to application) task stack size
[ ]   Bluedroid memory debug
[ ]   Classic Bluetooth
[*]   Include GATT server module(GATTS)
GATTS Service Change Mode (GATTS automatically send service change indication)  --->
	( ) GATTS manually send service change indication
	(X) GATTS automatically send service change indication
[*]   Include GATT client module(GATTC)
[ ]     Save gattc cache data to nvs flash
[*]   Include BLE security module(SMP)
[ ]   Disable BT debug logs (minimize bin size)
BT DEBUG LOG LEVEL  --->
(4)   BT/BLE MAX ACL CONNECTIONS(1~7)
[ ]   BT/BLE will first malloc the memory from the PSRAM
[ ]   Use dynamic memory allocation in BT/BLE stack
[*]   BLE queue congestion check           

I have also tried it without the BLE queue congestion check selected.

chegewara, 2018-10-19:
Everything seems to be fine. `BLEDevice::init()` should initialize controller.

chegewara, 2018-10-19:
@dkselw It is possible that somehow this line is causing this issue:
`https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L353`

It was uncommented just few days ago, and i think it suppose to be before controller init.

dkselw, 2018-10-25:
> @dkselw It is possible that somehow this line is causing this issue:
> https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L353
> 
> It was uncommented just few days ago, and i think it suppose to be before controller init.

I tried commenting out that line, and it did not help.

dkselw, 2018-10-25:
Testing SampleServer.cpp from BLETests with different commits.

f8d6a65b6a06a0a8a5778279842678ca610834e3 15 Oct, or anything newer does not work.
8f106d7c1a7a9684ce22c133949e2daeda7cdc59 11 Oct, does work.

chegewara, 2018-10-25:
Thanks guys, will investigate it when i have spare time. Looks like esp-idf is changed lately.

chegewara, 2018-10-25:
This is temporary solution, or just use my repository please.
https://github.com/nkolban/esp32-snippets/issues/680#issuecomment-433179431

Schekamatik, 2018-10-31:
Hey guys, do we have any additional info? Unfortunately the fix above do not work by me.

chegewara, 2018-10-31:
@Schekamatik Would be a problem if you use working commit from this post?
https://github.com/nkolban/esp32-snippets/issues/705#issuecomment-432974684

Sorry for inconvenience, but i cant fix it now. We can also ask @nkolban to revert all changes to confirmed working commit.

dkselw, 2018-10-31:
> Hey guys, do we have any additional info? Unfortunately the fix above do not work by me.

Making this change https://github.com/nkolban/esp32-snippets/issues/705#issuecomment-433247745 and https://github.com/nkolban/esp32-snippets/issues/707#issuecomment-433565926 gets the example code SampleClient.cpp and SampleServer.cpp working for me.

Schekamatik, 2018-11-01:
Hello 

> 
> 
> > Hey guys, do we have any additional info? Unfortunately the fix above do not work by me.
> 
> Making this change [#705 (comment)](https://github.com/nkolban/esp32-snippets/issues/705#issuecomment-433247745) and [#707 (comment)](https://github.com/nkolban/esp32-snippets/issues/707#issuecomment-433565926) gets the example code SampleClient.cpp and SampleServer.cpp working for me.

> 
> 
> > Hey guys, do we have any additional info? Unfortunately the fix above do not work by me.
> 
> Making this change [#705 (comment)](https://github.com/nkolban/esp32-snippets/issues/705#issuecomment-433247745) and [#707 (comment)](https://github.com/nkolban/esp32-snippets/issues/707#issuecomment-433565926) gets the example code SampleClient.cpp and SampleServer.cpp working for me.

I have tried it also many time, but unfortunately its didn't help. :(

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1324 
load:0x40078000,len:7788
ho 0 tail 12 room 4 
load:0x40080400,len:6448
entry 0x400806e8
Using ESP object:
v3.2-dev-1055-g3276a1316
Using lower level function:
v3.2-dev-1055-g3276a1316
**E (251) BLEDevice: esp_bt_controller_init: rc=259 ESP_ERR_INVALID_STATE**

Schekamatik, 2018-11-01:
> 
> 
> @Schekamatik It looks like arduino, right?

Yes

chegewara, 2018-11-01:
@Schekamatik 
Then you are not affected by any changes in past few months, there should not be any issue mentioned here.

My advice is to delete BLE from arduino libraries or even reinstall arduino-ide if this is possible.

Schekamatik, 2018-11-01:
> 
> 
> @Schekamatik
> Then you are not affected by any changes in past few months, there should not be any issue mentioned here.
> 
> My advice is to delete BLE from arduino libraries or even reinstall arduino-ide if this is possible.

Hello thank you very much for your help. I have found the issue. Somehow the ARDUINO_ARCH_ESP32 was not set inside eclipse IDE.
I have solved it by. => Workaroud.

#define ARDUINO_ARCH_ESP32
#ifdef ARDUINO_ARCH_ESP32
#include "esp32-hal-log.h"
#include "esp32-hal-bt.h"
#endif

chegewara, 2018-11-01:
@Schekamatik Im glad that you have found solution and it works for you now. Have a nice coding.

chegewara, 2018-11-05:
Ok guys, i have finally time to make quick test. The issue is missing code here:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L352-L360

```
#ifndef CLASSIC_BT_ENABLED
		// esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT); <-- commented because needs to be called before esp_bt_controller_init()
		errRc = esp_bt_controller_enable(ESP_BT_MODE_BLE);
		if (errRc != ESP_OK) {
			ESP_LOGE(LOG_TAG, "esp_bt_controller_enable: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}
#else
		errRc = esp_bt_controller_enable(ESP_BT_MODE_BTDM);
		if (errRc != ESP_OK) {
			ESP_LOGE(LOG_TAG, "esp_bt_controller_enable: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
			return;
		}
#endif
```

Sorry for late reply.

chegewara, 2018-11-13:
Shuld be fixed now with #716

closed 2018-11-27 by chegewara